micro-ROS/micro-ROS-Agent #9: client crashes after reconnecting to agent on pseudo terminal

christianrauch, 2018-12-19:
After terminating a serial connection over a pseudo terminal, a client cannot create a new connection to the agent.

Steps to reproduce:
1. create two pseudo terminal devices: `socat -d -d pty,raw,echo=0,link=$HOME/pts1 pty,raw,echo=0,link=$HOME/pts2`
2. start agent, connect to `pts1`: `ros2 run  uros_agent uros_agent serial $HOME/pts1`
3. recompile `rmw_microxrcedds` with `CONFIG_DEVICE` set to path of `pts2`
4. start client (connect to `pts2`): `./install/string_publisher_cpp/lib/string_publisher_cpp/string_publisher_cpp `
5. stop client (Ctrl+C) -> agent will print `Error client unknown.`
7. start client again -> client will crash with error:
```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: failed to create node session on Micro ROS Agent., at [...]/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_node.c:235, at [...]/ros2/rcl/rcl/src/rcl/node.c:336
```

qeyup, 2018-12-19:
Hi, I will try to reproduce and debug this scenario. As soon i have an answer I will respond you. Regards.

qeyup, 2019-01-08:
Hi, 

I have debugged the related error. I find out that this is a Micro XRCE-DDS agent issue. I have written an [issue ticket](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/48) on the [Micro XRCE-DDS agent repo](https://github.com/eProsima/Micro-XRCE-DDS-Agent). 

Regards.

closed 2019-11-04 by BorjaOuterelo