micro-ROS/micro-ROS-Agent #14: release for ROS dashing

christianrauch, 2019-06-20:
It would be nice to have the agent released for ROS dashing.

BorjaOuterelo, 2019-06-20:
We have Dashing support planned.
You can expect an update on this in the following weeks.
Micro XRCE-DDS Agent was recently updated to be compatible with Fast RTPS 1.8.0 which is the version in ROS2 Dashing and need to update a couple of components before releasing micro-ROS Dashing support.

BorjaOuterelo, 2019-07-11:
Hi @christianrauch. #15  Is the port for ROS dashing.
The rmw_microxrcedds port is still WIP, but micro-ROS-agent is ported, just need to update your .repos to use [Micro-XRCE-DDS Agent v1.1.0](https://github.com/eProsima/Micro-XRCE-DDS-Agent/releases/tag/v1.1.0).
I will notify you when the rest of micro-ROS stack is ported.

closed 2022-01-28 by pablogs9