micro-ROS/micro-ROS-Agent #21: Multiple client boards not performing well with single Agent through UDP

Viplav04, 2020-01-10:
Resources:
1 Olimex-stm32-e407 board with bmp180 pressure sensor - publisher
1 waveshare-stm32-open407i board with DP83848 ethernet peripheral - subscriber
1 TP-Link desktop switch for creating local network from Linux and 2 connections to the board
Boards NuttShell accessed through USB CDC/ACM profiles
1 USB pl2303 from waveshare board to Linux for serial connection

Case 1: Using built in publisher-subscriber of micro-ROS through UDP one agent
Data sending rate is really fast in this example, subscriber code changed to receive 500 messages. 
Result : Either one of the nodes work, either publisher or subscriber.

Case 2: Using built in publisher-subscriber of micro-ROS through UDP two agents with different port numbers
Result : same as above

Case 3 : Using bmp180 sensor and same structure of publisher-subscriber code as the built-in example through UDP with 1 agent and 2 agents with different ports
Data is sent every 1 second
Result: Both nodes are initialised at both the boards. Subscriber receives very less data, for eg, it receives 2 data messages while the publisher publishes about 10-15 messages in the same time.

Case 4 : Built in publisher-subscriber with 1 agent through UDP and 1 agent through serial
Data sending is very fast
Results: Nodes are initialised, but many messages are not received by the subscriber, i.e not reliable

Case 5 : Using BMP180 sensor with 1 agent through UDP and 1 agent through serial
Data is sent every 1 second
Results: Nodes are initialised and all values are received

iluetkeb, 2020-01-10:
Thank you for this report. Since it is agent related, I've moved it.

FranFin, 2020-07-20:
@Viplav04 Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

Viplav04, 2020-07-20:
Hi
Thank you for the invite. The topic was part of my master thesis which got
over in January 2020. I am currently not working with ROS programming. I
posted my results in the issue. Sorry, I cannot be of further assistance in
this moment.

Kind regards
Viplav

FranFin, 2020-07-20:
No problem! Thanks for taking the time to answer.
BR,
Francesca

closed 2022-01-28 by pablogs9