micro-ROS/micro-ROS-Agent #23: "Error written data on DDS entity" after restarting publisher

iluetkeb, 2020-01-20:
Using the pingpong publisher on the PC side, and the uros_pong_server on the MCU side, I can run the benchmark as long as I want.

*However* when I ran it for a couple of minutes and then *restart*, I get the "Error written data on DDS entity" from thee agent and communication does not work afterwards. I have to restart the agent.

iluetkeb, 2020-02-13:
Due to the various build system-related changes it has taken a while to test the new versions, but it looks as if this issue is solved now.

iluetkeb, 2020-02-13:
Sorry, a longer test revealed that the issue is still present.

Steps to reproduce:
 1. Set up an Olimex STM32-E407 with pingpong-eth configuration
 1. start agent on host and "uros_pong_server" on client
 1. Run micro_ros_rtt/pingpong for at least 30 seconds.
 1. Stop pingpong and restart it

-> agent prints "Error written data on DDS entity."

iluetkeb, 2020-02-13:
Versions used:
  - uros_agent version "dashing", commit id f377b7a72799d6eab6180fd77de3a2c13d2ddf0a
  - micro_xrce_agent tag "dashing", commit id 75febfeb7c89d7767ea98d38aa7cfcfda52dd9a7
  - micro-ros-rtt commit id "62fc547dd9009bb398f0fc1db5b306408e716119"
  - micro-xrce-dds-client tag dashing, commit id "99f7e84237b6808efa6a88bcb8f3b6c5c641104e"
  - rmw_microxrcedds branch dashing, commit id "e29e697aeac0b69f66cd4e33cb834d73d3ec0c37"

julibert, 2020-02-13:
Hi @iluetkeb, I'm going to take a look. Thanks for the description.

julibert, 2020-02-14:
Hi @iluetkeb, I was not able to reproduce this issue in my host using this [pingpong_server](https://github.com/micro-ROS/micro-ROS-rtt/pull/1) application. I will to reproduce it in NuttX the next week. Could you reproduce it in your host?

iluetkeb, 2020-02-14:
I always ran this on NuttX, but I can try again on the host.

julibert, 2020-02-20:
I was trying to reproduce this issue on NuttX but I was not able. Which network setup are you using? Are you using the Kobuki router?

iluetkeb, 2020-02-20:
The first time we noticed this issue has been with the Olimex board directly attached by Ethernet to my PC. In those cases, it occurred when we *restarted* the RTT test after it had been running for a while.

This week, I am using the Olimex board connected to the GL.Inet AP by Ethernet, and them I'm communicating with that using a USB Wifi adapter. In this setup, we also sometimes have the problem just during running, though rarely.

Moreover, with the second setup, we fairly see the session breaking down (no data being received) and the daemon then logs
```
[1582188574.700544] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1009760727, address: 192.168.8.133:516
```

Should I start the daemon with more verbose settings?

closed 2022-01-28 by pablogs9