micro-ROS/micro-ROS-Agent #26: Dashing Build Failure: uxrOnTopicFunc structure mismatch

Kelly-Coffey, 2020-04-13:
I cannot get the build to succeed. I have tried dockers ros:dashing + micro ros install and base:dashing.  
It looks like a structure mis-match issue. The uxrOnTopicFunc structure has an extra unit16_t "length" field. It looks like ePromisa code.
Below is the log for the base:dashing build failure.

ubu@ubu-VirtualBox:~$ sudo docker run -it microros/base:dashing
[sudo] password for ubu: 
root@1e804c6cccb2:/uros_ws# source /opt/ros/$ROS_DISTRO/setup.bash
root@1e804c6cccb2:/uros_ws# source install/local_setup.bash
root@1e804c6cccb2:/uros_ws# ros2 run micro_ros_setup create_agent_ws.sh
#All required rosdeps installed successfully
root@1e804c6cccb2:/uros_ws# colcon build --metas src
Starting >>> drive_base_msgs
Finished <<< drive_base_msgs [29.1s]                         
Starting >>> microxrcedds_agent
[Processing: microxrcedds_agent]                             
[Processing: microxrcedds_agent]                                      
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                        
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
[Processing: microxrcedds_agent]                                       
--- stderr: microxrcedds_agent                                        
Cloning into 'fastrtps'...
Note: checking out 'b257a33'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at b257a3350 XML Services [1.8.x] [7117] (#930)
Submodule 'thirdparty/android-ifaddrs' (https://github.com/michalsrb/android-ifaddrs.git) registered for path 'thirdparty/android-ifaddrs'
Submodule 'thirdparty/asio' (https://github.com/chriskohlhoff/asio.git) registered for path 'thirdparty/asio'
Submodule 'thirdparty/fastcdr' (https://github.com/eProsima/Fast-CDR.git) registered for path 'thirdparty/fastcdr'
Submodule 'thirdparty/idl' (https://github.com/eProsima/IDL-Parser.git) registered for path 'thirdparty/idl'
Submodule 'thirdparty/tinyxml2' (https://github.com/leethomason/tinyxml2.git) registered for path 'thirdparty/tinyxml2'
Cloning into '/uros_ws/build/microxrcedds_agent/fastrtps/src/fastrtps/thirdparty/android-ifaddrs'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastrtps/src/fastrtps/thirdparty/asio'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastrtps/src/fastrtps/thirdparty/fastcdr'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastrtps/src/fastrtps/thirdparty/idl'...
Cloning into '/uros_ws/build/microxrcedds_agent/fastrtps/src/fastrtps/thirdparty/tinyxml2'...
Cloning into 'microxrcedds_client'...
Switched to a new branch 'develop'
Cloning into 'microcdr'...
Switched to a new branch 'develop'
Cloning into 'cli11'...
Note: checking out 'v1.7.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 49ac989 Fixes to the readme
Submodule 'extern/googletest' (https://github.com/google/googletest.git) registered for path 'extern/googletest'
Submodule 'extern/json' (https://github.com/nlohmann/json.git) registered for path 'extern/json'
Submodule 'extern/sanitizers' (https://github.com/arsenm/sanitizers-cmake) registered for path 'extern/sanitizers'
Cloning into '/uros_ws/build/microxrcedds_agent/CLI11/src/cli11/extern/googletest'...
Cloning into '/uros_ws/build/microxrcedds_agent/CLI11/src/cli11/extern/json'...
Cloning into '/uros_ws/build/microxrcedds_agent/CLI11/src/cli11/extern/sanitizers'...
Cloning into 'sanitizers'...
Already on 'master'
Cloning into 'spdlog'...
Note: checking out 'v1.4.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 1549ff12 Replace STRING(PREPEND ..) in CMakeLists.txt to support older CMake versions
/uros_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp: In member function ‘void eprosima::uxr::InternalClient::set_callback()’:
/uros_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp:151:53: error: invalid conversion from ‘void (*)(uxrSession*, uxrObjectId, uint16_t, uxrStreamId, ucdrBuffer*, void*) {aka void (*)(uxrSession*, uxrObjectId, short unsigned int, uxrStreamId, ucdrBuffer*, void*)}’ to ‘uxrOnTopicFunc {aka void (*)(uxrSession*, uxrObjectId, short unsigned int, uxrStreamId, ucdrBuffer*, short unsigned int, void*)}’ [-fpermissive]
     uxr_set_topic_callback(&session_, on_topic, this);
                                                     ^
In file included from /uros_ws/build/microxrcedds_agent/temp_install/microxrcedds_client-1.2.1/include/uxr/client/client.h:24:0,
                 from /uros_ws/src/eProsima/Micro-XRCE-DDS-Agent/include/uxr/agent/p2p/InternalClient.hpp:18,
                 from /uros_ws/src/eProsima/Micro-XRCE-DDS-Agent/src/cpp/p2p/InternalClient.cpp:15:
/uros_ws/build/microxrcedds_agent/temp_install/microxrcedds_client-1.2.1/include/uxr/client/core/session/session.h:144:16: note:   initializing argument 2 of ‘void uxr_set_topic_callback(uxrSession*, uxrOnTopicFunc, void*)’
 UXRDLLAPI void uxr_set_topic_callback(
                ^~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [CMakeFiles/microxrcedds_agent.dir/src/cpp/p2p/InternalClient.cpp.o] Error 1
make[4]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [uagent-prefix/src/uagent-stamp/uagent-build] Error 2
make[1]: *** [CMakeFiles/uagent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< microxrcedds_agent	[ Exited with code 2 ]

Summary: 1 package finished [8min 35s]
  1 package failed: microxrcedds_agent
  1 package had stderr output: microxrcedds_agent
  1 package not processed
root@1e804c6cccb2:/uros_ws#

pablogs9, 2020-04-14:
Please check if https://github.com/micro-ROS/micro-ROS-Agent/pull/27 solves it. If it is solved, close the issue, thanks!

Kelly-Coffey, 2020-04-15:
I have verified with #27 the build is fixed. Thank you!

PR #27 (2020-04-14): Update main.cpp
This PR updates the *main.cpp* according to the API changes of Micro XRCE-DDS Agent. It fixes #26.
closed 2020-04-15 by Kelly-Coffey