micro-ROS/micro-ROS-Agent #29: Baud setting flag for serial devices are ignored

gramss, 2020-05-17:
I run the agent this way: 
```ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 230400 -v6```

I had the same baud setting running on my nucleo_f401re. I configured the baud setting via an overlay file of the dts. PR about this comes after it works..
```
<boardname>.overlay (at same level as prj.conf):
&usart2 {
        current-speed = <576000>;
        status = "okay";
};
```

I kept the agent running with the above settings and reflashed the firmware with 115200 baud.
The agent established the connection successfully while the -b setting should pipe in another baud rate..

Rechecked the API here: https://micro-xrce-dds.docs.eprosima.com/en/latest/agent.html

```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 --baudrate TEXT=576000 -v6
Press CTRL+C to exit
Warning: Custom baud rate set to: 0
[1589742529.260817] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1589742529.260958] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

Can somebody help me with this?

My goal:
Increase the transmission rate from ~100Hz at 115200 Baud to a higher on. The PLL Output is already set to 84MHz for my nucleo board (with the standard defconf for this board - this is the base KCONFIG file for my board from the zephyr/board/arm/nucleo_f401re folder inside zephyrproject - prj.conf overwrites/adds its setting to this). So I should get pretty high transmission rates regarding to this document: http://www.st.com/web/en/resource/technical/document/reference_manual/DM00031020.pdf (page 978 and following)

Thank you!

gramss, 2020-05-18:
@pablogs9 should be able to reproduces this with his iot discovery Board. It has the same serial interface as my nucleo_f401re. With the described overlay file it should be easily changed to my desired behavior of a faster baud rate.

The micro-ros extension for Zephyr only Checks the Baud rate in the usb-serial interface and not in the serial interface.
But, as it is not using this information or altering it, it should not be take any effect on the behavior.

The overlay file proves to be working as the communication cannot be established when I change the value to something different than 115200. With this default value set in the overlay file, everything works as expected.

Aiven92, 2020-05-18:
@gramss You can look at my PR with baudrate fix in https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/154.

gramss, 2020-05-18:
Hi @Aiven92 : works for me! :) 576000 is too fast for my board but 230400 already worked! I'm now playing with the baud to find the sweet spot.
At 576000 is gets a few messages but disconnects afterwards.. Maybe the MCU does not get the "hello" from the agent?

julibert, 2020-05-18:
Hi @gramss and @Aiven92,

I've already merged https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/154. Thank you so much for contributions!

pablogs9, 2020-05-18:
@gramss if you are trying to achieve the fastest publishing configuration consider using best effort publisher: check an example here https://github.com/micro-ROS/freertos_apps/blob/e0e857942449b786829c20e423a6fa0ab4600221/apps/crazyflie_position_publisher_reconnection/app.c#L89

According to our tests, best-effort publishers can dramatically improve the topic publishing period at the expense of not guaranteeing the reliability of the communication.

gramss, 2020-05-18:
@pablogs9 thanks for the hint. This is exactly what I was searching for. At least at 460800 this still fails:
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -b 460800 -v6
Press CTRL+C to exit
[1589797123.243930] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1589797123.244408] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1589797123.245093] info     | Root.cpp           | create_client            | create                 | client_key: 0xDEADBEEF, session_id: 0x81
[1589797123.245143] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x3735928559, address: 1
[1589797123.245256] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.246349] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.246583] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.246779] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.247298] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.247422] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.247605] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.248382] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.248538] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.248669] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.249612] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.249747] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.249884] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.250545] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.250721] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.250843] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.251676] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.251785] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.251885] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.252821] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.252930] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x3735928559, address: 1
[1589797123.253024] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xDEADBEEF, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1589797123.254176] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[1589797123.254196] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xDEADBEEF, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F DE AD BE EF 81 00 FC 01
[... cut ...]
[1589797166.899181] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0xDEADBEEF, address: 1
```

On 230400 Baud I get a really impressive rate of 1000 Hz..! Wow..
```
ros2 topic hz /zephyr_int32_publisher 
average rate: 1002.014
	min: 0.000s max: 0.003s std dev: 0.00057s window: 1003
```
On the defaultt 115200 Baud I get still 500Hz:
```
ros2 topic hz /zephyr_int32_publisher 
average rate: 500.946
	min: 0.001s max: 0.003s std dev: 0.00029s window: 503
```

Disclaimer: this is the int32_serial_publisher example code that just sends 4 Bytes (an int32 counter)

closed 2020-05-18 by gramss

pablogs9, 2020-05-18:
Maybe at these higher baudrates we should consider tweaking the serial transport for Zephyr... They have a couple of modes for serial drivers: sync and async. Check this: https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/drivers/uart

Maybe with the current implementation based on synchronous polling these rates are too high. If your require higher rates, consider modifying the serial transport to an async approach (like in the usb-serial transport).

gramss, 2020-05-18:
Thank you for the heads up. But I think this would be out of scope for me. At least for now.