micro-ROS/micro-ROS-Agent #35: Unintuitive cli flag handling for help flag

RobertWilbrandt, 2020-09-30:
Currently the cli flag handling for the -h/--help flag seems to be broken. When using

```console
> ros2 run micro_ros_agent micro_ros_agent
Usage: 'MicroXRCEAgent <udp4|udp6|tcp4|tpc6|serial|pseudoterminal> <<args>>'
For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.
```
This seems fair, but following the advice:

```console
> ros2 run micro_ros_agent micro_ros_agent -h
Error: chosen transport '--help' is invalid!
Usage: 'MicroXRCEAgent <udp4|udp6|tcp4|tpc6|serial|pseudoterminal> <<args>>'
For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.

> ros2 run micro_ros_agent micro_ros_agent --help
Error: chosen transport '-h' is invalid!
Usage: 'MicroXRCEAgent <udp4|udp6|tcp4|tpc6|serial|pseudoterminal> <<args>>'
For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.
```

Just a small qol issue, hopefully i'll have time to look at this and propose a pr for this soon.

jamoralp, 2020-10-01:
Thank you for the insight! See related pull request here :slightly_smiling_face: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/189

pablogs9, 2020-10-01:
Closing since merged. Please reopen if any other issue.

And thanks for the tip @RobertWilbrandt

closed 2020-10-01 by pablogs9