micro-ROS/micro-ROS-Agent #39: Build error: duplicate package names

anaelle-sw, 2020-11-30:
Hi! An error occurs when building micro_ros_msgs package.

## System
- Ubuntu 20.04
- ROS2 Foxy

## Steps to reproduce
I am following [this tutorial](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/), to install micro_ros_agent on my computer. I am building the micro_ros_msgs package when executing:
`ros2 run micro_ros_setup create_agent_ws.sh`.

## Actual behavior
The package micro_ros_msgs is present in both micro-ROS-Agent and rmw_microxrcedds repositories (which are both installed when following the tutorial above). Complete log is:
```
ERROR: Rosdep experienced an error: Multiple packages found with the same name "micro_ros_msgs":
- uros/micro-ROS-Agent/micro_ros_msgs
- uros/rmw_microxrcedds/micro_ros_msgs
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.20.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 144, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 430, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 485, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 96, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "micro_ros_msgs":
- uros/micro-ROS-Agent/micro_ros_msgs
- uros/rmw_microxrcedds/micro_ros_msgs
```

Sorry if this is an already known issue you are working on. Thanks!

jamoralp, 2020-12-01:
Hello @anaelle-sw, thank you for warning us about this issue. Indeed, this package was duplicate and this was causing errors when `rmw-microxrcedds` and `micro-ROS-Agent` were in the same build environment. I already fixed it in https://github.com/micro-ROS/micro_ros_setup/pull/242, please have a look at it and report back if you find any other problem :slightly_smiling_face: and thank you a lot!

anaelle-sw, 2020-12-01:
This PR fixed the issue for me. Thanks for support!

closed 2020-12-01 by anaelle-sw