micro-ROS/micro-ROS-Agent #42: Can micro_ros_agent be run via launch files?

anaelle-sw, 2020-12-02:
Hello! 
I open this issue as I would like to run micro_ros_agent from a launch file.

The `launch.py` file would look like this:
```c
def generate_launch_description():
    uros_agent_node = Node(
      package='micro_ros_agent',
      executable='micro_ros_agent',
      name='micro_ros_agent_node',
      output='screen',
      parameters=[
        {'transport': 'serial'},
        {'dev': '/dev/path/to/board'}]
      )

    return LaunchDescription([
        uros_agent_node
    ])
```
It would allow to give parameters to the agent (as the transport type).

I tried to launch this kind of launch file, but an error occurs: `Error: chosen transport '--ros-args' is invalid!`. If no parameter is given, the same error occurs. So I guess the problem is coming from the way I tried to pass the transport type argument. So I tried this:
```c
    uros_agent_node = Node(
      package='micro_ros_agent',
      executable='micro_ros_agent',
      output='screen',
      arguments={'serial', '--dev', '/dev/teensy_elodie'}
      )
```
But the arguments' order is changed when launching the file (I don't really know why, I am trying to solve this): the actual command executed is `/home/usr/ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent --dev /dev/path/to/board serial --ros-args`. And, as the first argument is supposed to be the transport type (I guess), the following error shows up: `Error: chosen transport '--dev' is invalid!`

For now, I can launch micro_ros_agent from a launch file using Python function `ExecuteProcess`:
```c
def generate_launch_description():
    uros_agent_process = ExecuteProcess(
             cmd=['ros2', 'run', 'micro_ros_agent', 'micro_ros_agent','serial', '--dev', '/dev/path/to/board'],
             output='screen')

    return LaunchDescription([
        uros_agent_process
    ])
```
But we would prefer to use the function `Node`, simply to follow ROS2 launch file common usage.

So, I don't know if such a feature exists. If so, how to use it and pass parameters to the agent? I am not so used to Python, may be the errors are not micro_ros_agent-related but occurs because of the Python part. May be you could provide a `launch.py` example file?

Thanks a lot for helping!

pablogs9, 2020-12-02:
This a method that we haven't tried, tomorrow we will take a look. For sure there is a way of doing this.

anaelle-sw, 2020-12-10:
Hello! I would actually have a second request related to this issue:

micro-ros-agent installed from snap release can be launched with command: `micro-ros-agent serial --dev /dev/path/to/board `
 So, using launch files, it means that I have to pass this command to the `ExecuteProcess` function. For now, it is not possible to run the agent as a node (`ros2 run ...`), when installed from snap release, as the package `micro_ros_agent` is not found. We would still rather use the `Node` function, as explained in previous post. Will it be possible to launch the agent (installed from snap) as a node ?

jamoralp, 2020-12-14:
Good morning. There were several incompatibilities regarding the way ros 2 manages parameters within launch files and our CLI parser, which were causing the launch process to fail. I have managed to find a solution, and it should be working now with changes introduced in #47 

Regarding the snap, I'm not sure you will be able to launch it as a node. The reason is that, for practical purposes, what a snap does is mounting its own virtual filesystem, with all the required libraries, headers, dependencies and environment variables set in order to run a certain executable. That means the process is self-contained and you can not "view" the ros 2 package from outside the snap, so I don't think your ROS2 Foxy installation will be able to see the snap as a ROS 2 package and manage to launch the process.

anaelle-sw, 2020-12-14:
The changes works nicely for me, I am able to launch the agent in launch files from Python function `Node(...)`. Thanks a lot !
Ok, I understand the "problem" about snap. 
I can close this issue I guess. Thanks for support !

closed 2020-12-14 by anaelle-sw

DaniGarciaLopez, 2021-08-23:
Hello! 

I'm trying to launch micro_ros_agent using a launch file and it's not working. It only works using `ExecuteProcess`:
```
ExecuteProcess(
             cmd=['ros2', 'run', 'micro_ros_agent', 'micro_ros_agent','serial', '--dev', '/dev/ttyACM0'],
             output='screen')
```
Reviewing the PR [#47](https://github.com/micro-ROS/micro-ROS-Agent/pull/47) and [main.cpp](https://github.com/micro-ROS/micro-ROS-Agent/blob/foxy/micro_ros_agent/src/main.cpp), the correct way to do it should be `arguments={'serial --dev /dev/ttyACM0'}` but [current launch file](https://github.com/micro-ROS/micro-ROS-Agent/blob/foxy/micro_ros_agent/launch/micro_ros_agent_launch.py) do it in another way.

I tried both ways of launching arguments: `arguments={'serial --dev /dev/ttyACM0'}` and `arguments={"serial", "--dev", "/dev/ttyACM0"}` with no luck.

Thanks!

Sudharsan10, 2021-12-16:
@DaniGarciaLopez It worked for me using the format PR #47. I use a PICO controller and I had the same issue when i passed it as a single argument `arguments={'serial --dev /dev/ttyACM0'}` but when I used `arguments={'serial', '--dev', '/dev/ttyACM0'}` it worked for me.

this is my launch file,
`import os
import launch
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():    
    return LaunchDescription([
        Node(
            package='micro_ros_agent',
            executable='micro_ros_agent',
            name='micro_ros_agent',
            arguments=['serial','--dev', '/dev/ttyACM0']
        )
    ])`
    
    
One problem I had was I had to unplug and plug my PICO after starting the micro ros agent node. Not sure why.

DaniGarciaLopez, 2022-01-20:
Thanks for the answer @Sudharsan10!. I checked again and I can confirm that now it works with: `arguments=['serial','--dev', '/dev/ttyACM0']`