micro-ROS/micro-ROS-Agent #44: Segmentation fault when running micro-ros-agent as snap package

anaelle-sw, 2020-12-08:
Hi! I saw the announcement that micro-ros-agent has been released as a snap package ([here](https://discourse.ros.org/t/micro-ros-agent-on-snap/17729)). I am really happy to try this release. The problem is, I got a segmentation fault error when running it.

## Setup
- Ubuntu 20.04 / ROS2 Foxy
- Teensy 3.1

## Steps to reproduce
I first removed micro-ros-agent installation from source that I already had on my computer. I also removed all other micro-ROS submodules installation from source. Then I installed micro-ros-agent as snap package following [this example](https://snapcraft.io/install/micro-ros-agent/arch):
`sudo systemctl enable --now snapd.socket`
`sudo ln -s /var/lib/snapd/snap /snap`
`sudo snap install micro-ros-agent`
And then I run it:
`micro-ros-agent serial --dev /dev/path/to/board`

I also tried this while keeping micro-ROS submodules installation from source, or having micro-xrce-dds-agent installed from snap too. The results are the same.
I can precise that running micro-ros-agent with installation from source works, with the exact same computer and board.

## Expected behavior
The agent is launched, waiting for the board or detecting it (depending on whether the board is plugged).

## Actual behavior
The agent crashes with segmentation fault error. Complete log is:
```c
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[1607423318.496277] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/path/to/board, errno: 13. Please re-run with superuser privileges.
Error while starting serial agent!
Segmentation fault
```
I tried to run the agent when the board is plugged and when it is not. Either way, the result is the same.
I also tried to follow the log recommendation (: `Please re-run with superuser privileges.`), and logged in as root with `sudo -i` command. The result is the same.

Could you tell me if some other micro-ROS packages are required to be installed from source, in order to be able to run micro-ros-agent as a snap package? Is micro-xrce-dds-agent required to be install as a snap package too?

Thanks a lot for support!

jamoralp, 2020-12-09:
Good morning @anaelle-sw,

Certainly, this should not be happening. This is probably related to the permissions given to the micro-ROS-Agent snap executable during compile time. Let me dig into this issue and I will report back ASAP.

Thank you!

jamoralp, 2020-12-10:
Good morning,

I've been digging into this and found out some interesting facts that we didn't notice, primarily because we did all our test with the snap versions of the `Micro XRCE-DDS Agent` and the `Micro-ROS Agent` using network connections:

* By default, the `snapd` daemon is not prepared to detect hotplug connections to serial ports. This is one of the reasons why snap is not properly detecting your device.
* Because of the first point, the device won't be automatically connected to the snap image.

I was able to run the `micro-ros-agent` snap using a serial device by following these steps:

* Resfresh your local installation of the snap core package
```bash
$ sudo snap refresh core --edge
```
* Enable the hotplug feature, and restart the `snapd` daemon
```bash
$ sudo snap set core experimental.hotplug=true
$ sudo systemctl restart snapd
```
* Plug in your microcontroller to the USB port, and then execute `snap interface serial-port`. You should see there a list of all the snap images in your machine that use the `serial-port` plug, and if the device is connected, a `slot` section. Mine looks like this:
```bash
name:    serial-port
summary: allows accessing a specific serial port
plugs:
  - micro-ros-agent
  - micro-xrce-dds-agent
slots:
  - snapd:cp2102cp2109uartbrid (allows accessing a specific serial port)
```
* Connect your snap image to the desired serial port (replace accordingly)
```bash
snap connect micro-ros-agent:serial-port snapd:cp2102cp2109uartbrid
```
* After this, `sudo micro-ros-agent serial --dev /dev/ttyUSB0` works for me.

The whole process seems to be a little bit tricky, but I believe this hotplug feature is currently under the final stages of its development and probably might be released in the next stable version of the snap `core` package. Maybe @kyrofa can say something about this, or even about the steps I described in order to simplify them :slightly_smiling_face:

anaelle-sw, 2020-12-10:
Hello!
Thanks for all of these explanations. This method works really well for me, the agent detects the board, and I can perfectly run our ROS2/micro-ros application with micro-ros-agent installed from the snap release!

I can just precise two things:

- In order to have the agent and ROS2 communicate, I don't need any micro-ros submodules (I mean the submodules that are installed when following the [micro-ros tutorial here](https://micro-ros.github.io/docs/tutorials/core/first_application_linux/). I used to have these installed from source, when using micro-ros-agent installed from source too). However, I do need to install micro-xrce-dds-agent as a snap package. Otherwise the agent detects the board and opens the session, but no messages are passing between ROS2 and the board.

- If micro-ros-agent is launched when logged in as root (as you suggested: `sudo micro-ros-agent serial --dev /dev/ttyUSB0`), the process is not ended when closing the terminal where it was launched, neither with `Ctrl+C` command. So when running a new micro-ros-agent, there is actually a second agent running. And having several agents running for the same board just makes a mess with our application behavior (and I guess this is normal, and not only related to our application)... Anyway, launching the agent without using `sudo` doesn't make any problem, everything works properly. 

I will let this issue opened, as you colleague might want to add something as you said. But it is all good for me, you can close the issue if you want to. Thank you!

kyrofa, 2020-12-10:
> Maybe @kyrofa can say something about this, or even about the steps I described in order to simplify them

No you're spot on, thanks for the ping. I have a meeting on Monday to discuss its experimental nature, I'll get back to you on this.

jamoralp, 2020-12-16:
> Hello!
> Thanks for all of these explanations. This method works really well for me, the agent detects the board, and I can perfectly run our ROS2/micro-ros application with micro-ros-agent installed [...]

Good morning @anaelle-sw ,
Sorry for the delay in answering. As micro-ROS is an european project, I've been very busy writing deliverable documents and wasn't able to check Github until now.

Regarding the two points you have mentioned:
* Yes, if you install the *micro-ROS Agent* as a snap package, you won't need any of the modules described in the tutorial. This is inferred from snap architecture itself: it provides you with a self-contained environment, where al the necessary dependencies required to launch a certain process are already available and properly source within the snap filesystem. This also leads to answering the next observation: I do not think it is necessary that you have installed the `micro-xrce-dds-agent` snap previously to using the `micro-ros-agent` snap. The `micro-ros-agent` snap itself should take care of downloading first the Micro XRCE-DDS Agent library, compiling it and using it as a base to compile the micro-ROS Agent. Actually, I uninstalled the `micro-xrce-dds-agent` snap, and the `micro-ros-agent` snap is still working as expected. Could you please confirm this?
* Ok with the `sudo` part. I thought that it was necessary due to the fact that, whithout the snap, you actually need to prepend `sudo` to the micro-ROS Agent executable to use serial devices. I will remove it from the instruction set. Thanks!

anaelle-sw, 2020-12-16:
Hello @jamoralp! No problem, thanks for your answer!
I removed  the `micro-xrce-dds-agent` snap installation. I am still able to launch the agent and to correctly run our application. Sorry about it, I might have missed something last time... This issue can be closed then. Thank you!

closed 2020-12-16 by anaelle-sw