micro-ROS/micro-ROS-Agent #50: build_agent.sh fails while trying to go through tutorial

agillies8, 2021-02-01:
**Describe the bug**
I'm trying to go through the "first application linux" tutorial, but am getting a build error that is preventing the micro_ros_agent from finishing its build.

**To Reproduce**
1. System Description: 
-Ubuntu 20 running in a VirtualBox with windows 10 host
-Installed ROS2 Foxy using standard install procedure, confirmed working with pub/sub example
2. Followed all steps here: https://micro-ros.github.io/docs/tutorials/core/first_application_linux/  (Not using the docer method)
3. On running this step:  --> $ros2 run micro_ros_setup build_agent.sh
Output:
Building micro-ROS Agent
Starting >>> microcdr
Finished <<< microcdr [0.30s]                     
Starting >>> microxrcedds_agent
Finished <<< microxrcedds_agent [0.64s]                     
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [0.38s]                     
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [1.54s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
/usr/bin/ld: warning: libfastrtps.so.2, needed by /home/andrewgillies/ros2_foxy/ros2-linux/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.0.2
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `spdlog::logger::log_it_(spdlog::details::log_msg const&, bool, bool)'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/andrewgillies/uagent/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.1.4.2: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:172: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [0.65s, exited with code 2]

Summary: 4 packages finished [3.80s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

jamoralp, 2021-02-01:
Good morning,
I'm afraid we would need more details about which specific Ubuntu image you mounted using VirtualBox and double check if you missed some of the commands in the tutorial. For me, it is working as expected, running on a Ubuntu 20.04 Focal *Docker* container.
That being said, this error seems to be related to some `fmt` library missing on your system. Did you try:
```
sudo add-apt-repository universe
sudo apt-get update
sudo apt-get install libfmt-dev
```
, then recompiling the agent with the steps provided? The only thing in the agent that is using some headers related to this library is the logger, to display information about the agent's status in the screen.

shic15, 2021-04-03:
Hey, 
Even I'm facing the exact same error. I just saw the issue here(after posting it myself [here](https://github.com/micro-ROS/micro-ROS.github.io/issues/303)) and tried the solution you suggested, to download fmt lib. But still, the error persists. It'll be really helpful if you can look into it, and please let me know if you need something else.

reubenstr, 2021-07-17:
Same error for me on Ubuntu 20.04.2 LTS even after executing the suggested commands to install fmt.

reubenstr, 2021-07-19:
I now able to build and execute the agent (not using the docker container).

There are several ROS2 official install tutorials and they appear to install/source ROS2 in differently thus breaking the micro-ros-agent build process.

My process:
Install Ubunutu 20.04.

Install ROS2 via Debian packages (I installed the desktop version): https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html

Install rosdep2: sudo apt install python3-rosdep2

Install colcon: sudo apt install python3-colcon-common-extensions

Install micro-ros: https://micro.ros.org/docs/tutorials/core/first_application_linux/

I am really excited to get micro-ros working with my project, thanks for the hard work!

irsyadtc, 2021-09-03:
Machine: Jetson Nano
OS: Ubuntu 18
Running Docker image: [dustynv/ros:foxy-ros-base-l4t-r32.6.1
](https://hub.docker.com/layers/dustynv/ros/foxy-ros-base-l4t-r32.6.1/images/sha256-3271b330732647c65f178980b5e6437d2b9ea03627ac0104e0d78ebb79cd55bc?context=explore)
As I'm going through the steps in [https://micro.ros.org/docs/tutorials/core/first_application_linux/](https://micro.ros.org/docs/tutorials/core/first_application_linux/), until build step of  "Creating the micro-ROS agent" `ros2 run micro_ros_setup build_agent.sh`, I've a similar error:

```
Your branch is up to date with 'origin/foxy'.
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rclc_lifecycle: No definition of [osrf_testing_tools_cpp] for OS version [bionic]
rclc: No definition of [osrf_testing_tools_cpp] for OS version [bionic]
root@caremnano2-desktop:/home/microros_ws/microros_ws# ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> builtin_interfaces
Starting >>> micro_ros_msgs
Starting >>> microxrcedds_agent                                                      
Finished <<< microxrcedds_agent [2.40s]                                
Finished <<< micro_ros_msgs [4.28s]                                    
Finished <<< builtin_interfaces [29.0s]                         
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                              
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `spdlog::logger::log_it_(spdlog::details::log_msg const&, bool, bool)'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/microros_ws/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
collect2: error: ld returned 1 exit status
make[2]: *** [micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [1.83s, exited with code 2]
```
I tried the suggestion by @jamoralp and it says `libfmt-dev is already the newest version (4.0.0+ds-2).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.`

Can someone help me?

Acuadros95, 2022-01-28:
We just added a flag to use the system version of `spdlog`: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/303 and https://github.com/micro-ROS/micro-ROS-Agent/pull/118

Agent build should succeed with this flag enabled:
`colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`

closed 2022-01-28 by Acuadros95

cyizhuo, 2024-05-25:
I solved this on my horizon rdk x3 arm board, by compile a 6.x version fmt. I'm using ubuntu 20, and I compiled fmt-6.1.2.

```sh
wget https://github.com/fmtlib/fmt/archive/refs/tags/6.1.2.zip
unzip 6.1.2.zip
cd fmt-6.1.2
mkdir build && cd build && cmake .. && make -j4
sudo make install
```

I compiled and installed fmt just before `ros2 run micro_ros_setup build_agent.sh` by the way.