micro-ROS/micro-ROS-Agent #52: Agent not sending messages

mrbhjv, 2021-02-24:
## Issue template

- Hardware description: Nucleo-F746ZG
- RTOS: FreeRTOS
- Installation type: custom lib (built as static)
- Version or commit hash: foxy

#### Steps to reproduce the issue
Start agent on a docker container using `ros2 run micro_ros_agent micro_ros_agent udp4 -p 8888 -v6`
Container is run with `docker run -p 8888:8888/udp -it --rm <image>`

Start a basic application with a subscription of type...
```
rclc_subscription_init_default(
      &subscriber_, &node_, ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
      "/twist_in");
```

From the same container running the agent, simply from another terminal, run
` ros2 topic pub /twist_in geometry_msgs/msg/Twist '{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}`

#### Expected behavior
The message is sent from the Agent, the callback is reached on the board.

#### Actual behavior
The Agent doesn't seem to send any message, even though I can see the topic, and that there's one subscriber.

#### Additional information
```
[1614158113.361169] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1614158113.361769] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1614158154.243762] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data:
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 7A 99 5A CD 81 00 FC 01
[1614158154.244349] info     | Root.cpp           | create_client            | create                 | client_key: 0x7A995ACD, session_id: 0x81
[1614158154.244615] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2056870605, address: 172.17.0.1:54161
[1614158154.244882] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 19, data:
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1614158154.246722] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 96, data:
0000: 81 80 00 00 01 05 56 00 00 0A 00 01 01 02 00 00 47 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 65 6D 6F 74 65 3C 2F 6E 61 6D 65 3E 3C
0040: 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1614158154.246838] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1614158154.249079] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1614158154.249455] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x7A995ACD, participant_id: 0x000(1)
[1614158154.249758] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1614158154.250061] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1614158154.250118] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1614158154.250132] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1614158154.255240] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1614158154.255288] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 132, data:
0000: 81 80 01 00 01 05 7C 00 00 0B 00 02 02 02 00 00 6E 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 75 72 6F 73 2F 72 65 6D 6F 74 65 5F 69 6E 3C 2F 6E 61 6D 65 3E 3C 64
0040: 61 74 61 54 79 70 65 3E 67 65 6F 6D 65 74 72 79 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F
0060: 3A 3A 54 77 69 73 74 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73
0080: 3E 00 00 01
[1614158154.255322] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1614158154.255722] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x7A995ACD, topic_id: 0x000(2), participant_id: 0x000(1)
[1614158154.255805] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1614158154.255835] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1614158154.255869] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
[... 41 lines cut ...]
[1614158154.267961] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1614158154.267971] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1614158154.268246] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 03 00 03 00 80
[1614158154.268371] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1614158154.269802] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7A995ACD, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1614158247.624339] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data:
0000: 81 80 04 00 08 01 10 00 00 0E 00 06 80 00 00 01 FF FF 00 00 00 00 00 00
```

And there seems to be a watchdog message that keeps being sent from the board
```
 81 00 00 00 0B 01 05 00 04 00 04 00 80
```

ros2 topic info
```
Type: geometry_msgs/msg/Twist
Publisher count: 0
Subscription count: 1
```

pablogs9, 2021-02-24:
Try to run the agent using this command:

```bash
docker run -it --rm -v /dev/shm:/dev/shm --privileged --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

mrbhjv, 2021-02-24:
I'm running it from Windows and can't mount /dev, neither does `--privileged` or `--net=host` do something. Just forcing the port with `-p 8888:8888/udp` does the trick. I can echo from the board and I'll see it on the agent side

pablogs9, 2021-02-24:
Ok, this is a known issue about Fast-DDS and shared memory in Dockers.

You can read a possible fix here: https://answers.ros.org/question/370595/ros2-foxy-nodes-cant-communicate-through-docker-container-border/?answer=371655#post-id-371655

If you are in windows, the best option is to load the XML profile with the shared memory transports disabled. In the link, you will find the required XML for disabling shared memory in all ROS 2 nodes.

In Linux we set the variable using something like this:
```bash
echo "export FASTRTPS_DEFAULT_PROFILES_FILE=[XML profile]" >> ~/.bashrc
. ~/.bashrc
```

Let me know if this solves the issue.

mrbhjv, 2021-02-24:
I'll try it out, and can confirm it works fine on Linux, it's purely a Docker for Windows (what a surprise) thing. Thanks!

pablogs9, 2021-02-24:
Cool, if you don't mind I'm closing. If you have further problems with Fast-DDS or ROS 2 and Docker do not hesitate to open an issue in Fast-DDS repo.

closed 2021-02-24 by pablogs9