micro-ROS/micro-ROS-Agent #54: Building Micro-XRCE-DDS Agent fails under ROS 2 Rolling

ralph-lange, 2021-03-12:
**Describe the bug**
When building the micro-ROS Agent for ROS 2 Rolling on Ubuntu 20.04, I received a build error during the build of the Micro-XRCE-DDS Agent.

**To Reproduce**
* Follow the tutorial in https://micro.ros.org/docs/tutorials/core/first_application_linux/, but with `rolling` (or rather `main`) instead of `foxy`.
* Builds fails in step `ros2 run micro_ros_setup build_agent.sh`

The problem seems to be the use of a log function with a `dds::xrce::TransportAddress` type.

```text
In file included from /home/rl/micro-ros_rolling_ws/build/microxrcedds_agent/temp_install/spdlog-1.4.2/include/spdlog/common.h:38:
In file included from /home/rl/micro-ros_rolling_ws/build/microxrcedds_agent/temp_install/spdlog-1.4.2/include/spdlog/fmt/fmt.h:22:
/home/rl/micro-ros_rolling_ws/build/microxrcedds_agent/temp_install/spdlog-1.4.2/include/spdlog/fmt/bundled/core.h:844:50: error: no matching member function for call to 'map'
    type_constant<decltype(arg_mapper<Context>().map(std::declval<T>())),
                           ~~~~~~~~~~~~~~~~~~~~~~^~~
/home/rl/micro-ros_rolling_ws/build/microxrcedds_agent/temp_install/spdlog-1.4.2/include/spdlog/fmt/bundled/core.h:1000:10: note: in instantiation of template type alias 'mapped_type_constant' requested here
  return mapped_type_constant<Arg, Context>::value |
         ^
/home/rl/micro-ros_rolling_ws/build/microxrcedds_agent/temp_install/spdlog-1.4.2/include/spdlog/fmt/bundled/core.h:1001:11: note: in instantiation of function template specialization 'fmt::v6::internal::encode_types<fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char>, dds::xrce::TransportAddress>' requested here
         (encode_types<Context, Args...>() << 4);
```

**System information:**
 - OS: Ubuntu 20.04
 - ROS 2: Rolling (as of 12 March 2021)

**Additional context**
Workaround: In Logger.hpp disable logging by `#undef UAGENT_LOGGER_PROFILE`

ralph-lange, 2021-03-12:
//Cc: @pablogs9

ralph-lange, 2021-03-16:
Just tried the tutorial under Linux and ran into the same issue.

pablogs9, 2021-03-17:
Hi Ralph, this night the CI has passed OK: https://github.com/micro-ROS/micro_ros_setup/runs/2127704213?check_suite_focus=true

I'm going to check this manually.

ralph-lange, 2021-03-17:
Okay, I've found it: On the first machine I used clang not gcc.

pablogs9, 2021-03-18:
Ok, closing since solved. Thanks Ralph.

ralph-lange, 2021-03-18:
I would not call this solved, since clang is also a widely used compiler.

pablogs9, 2021-03-18:
@Acuadros95 could you test this compilation process using clang instead of gcc?

Acuadros95, 2021-03-23:
@ralph-lange Hi Ralph, clang couldn't find the formatter for the `dds::xrce::TransportAddress` type. 

We have just merged a solution for this issue [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/233).

closed 2021-03-23 by Acuadros95

ralph-lange, 2021-03-23:
Thank you for fixing this issue, @Acuadros95