micro-ROS/micro-ROS-Agent #55: Agent behaves odd

bjv-capra, 2021-03-15:
## Issue template

- Hardware description: Jetson Nano
- Jetpack 4
- Installation type: docker container built on Jetson using micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
Build container with
```
FROM ros:foxy-ros-base-l4t-r32.4.4

RUN mkdir microros_ws
WORKDIR /microros_ws

RUN git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

RUN rosdep update && rosdep install --from-path src --ignore-src -y; exit 0

RUN . /opt/ros/${ROS_DISTRO}/install/setup.sh && colcon build
RUN . install/setup.sh \
    && ros2 run micro_ros_setup create_agent_ws.sh; exit 0
RUN . install/setup.sh \
    && ros2 run micro_ros_setup build_agent.sh; exit 0

COPY /uros_entrypoint.sh /

ENTRYPOINT ["/uros_entrypoint.sh"]
CMD ["ros2", "run", "micro_ros_agent", "micro_ros_agent", "udp4", "-p", "8888", "-v6"]
```
I had to ignore some errors when running rosdep install, as it gives me errors with no maintainers on some pluginlib.

set the container to start on boot with

```
sudo docker run --restart unless-stopped -d --network host agent-debug
```

Have a client waiting for the agent using the ping mechanism. On another docker container, have a publisher with 20hz rate wanting to publish to the client immediately when the board is up.

#### Expected behavior

Clients connect to agent and communication starts

#### Actual behavior

The client works as if the agent disconnects (currently I have no control over a possible disconnection of the agent). Then the messages seem to arrive really slow. The client runs on FreeRTOS.

If I start the agent manually, rather than let it automatically start, the communication seems to work with no problem.

#### Additional information
I've tried to debug using v6, but it gets quite cluttered. However, with v4, I got this once
```
[1615535289.626331] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615535289.627360] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534922.525715] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615534922.526631] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534921.946088] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615534921.946596] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534921.965763] info     | Root.cpp           | create_client            | create                 | client_key: 0x25DE13DB, session_id: 0x81
[1615534921.965869] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x635311067, address: 10.46.28.2:9354
micro_ros_agent: /opt/ros/foxy/src/rmw_dds_common/src/graph_cache.cpp:267: rmw_dds_common::msg::ParticipantEntitiesInfo __modify_node_info(const rmw_gid_t&, const string&, const string&, FunctorT, rmw_dds_common::GraphCache::ParticipantToNodesMap&) [with FunctorT = rmw_dds_common::GraphCache::associate_reader(const rmw_gid_t&, const rmw_gid_t&, const string&, const string&)::<lambda(rmw_dds_common::msg::NodeEntitiesInfo&)>; rmw_dds_common::msg::ParticipantEntitiesInfo = rmw_dds_common::msg::ParticipantEntitiesInfo_<std::allocator<void> >; rmw_gid_t = rmw_gid_t; std::__cxx11::string = std::__cxx11::basic_string<char>; rmw_dds_common::GraphCache::ParticipantToNodesMap = std::map<rmw_gid_t, rmw_dds_common::ParticipantInfo, rmw_dds_common::Compare_rmw_gid_t>]: Assertion `participant_info != participant_map.end()' failed.
[1615534929.106605] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615534929.107236] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

```

This is with v6
```
[1615535559.756781] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[... 1201 lines cut ...]
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80
[1615534958.975600] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5C4A6815, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80
[1615534959.176205] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5C4A6815, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80
[1615534959.376458] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5C4A6815, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80
[1615534959.576806] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5C4A6815, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80
[1615534959.777137] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5C4A6815, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0C 00 1B 00 80

```

However, on another run with v4
```
[1615535323.601357] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615535323.601833] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534922.356281] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615534922.356800] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534930.461401] info     | Root.cpp           | create_client            | create                 | client_key: 0x0556148A, session_id: 0x81
[1615534930.462369] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x89527434, address: 10.46.28.2:18170

```
Which does not provide the same error as before, yet I can see it's not working.

jamoralp, 2021-03-18:
Hello, I did not look very much to this into detail, but I saw you are launching your containers without the `--rm` option. Beware, because this could potentially lead to several agent instances being held on your system

bjv-capra, 2021-03-18:
Hi, I'm consciously doing so. Using the `--restart unless-stopped` policy. So that I don't have to manually restart them. What I see is that the agent connects, sends a few messages to the client, then the agent crashes, making the container restart (because of the policy). As I don't have anyway on my client to re-connect, it stays disconnected. I simply think this is not expected behavior.

I know there are no more agents running.

jamoralp, 2021-03-18:
Ok, I saw the trace now. That is very weird, specially considering that it seems to be related with the RMW. Which RMW vendor implementation are you using?

jamoralp, 2021-03-18:
We are currently trying to fix a big issue with the agent and an std::bad_alloc issue.

This could be related, as we identified the issue being present when using the micro-ROS Agent with the Graph manager activated, and having conflicts with the Fast-DDS library versioning.

https://github.com/micro-ROS/micro-ROS-Agent/pull/57 we are preparing this cmake flag to disable the graph manager if not being used (but also until we fix this versioning problem), could you compile the agent using this option once it is merged ?

bjv-capra, 2021-03-18:
I did see that PR, wondered why someone wouldn't want the graph manager.
Is there any difference if I run directly the microxcdr agent, instead of the micro-ros-agent? AFAIK, the only difference was the graph manager.

bjv-capra, 2021-03-18:
Regarding RMW, I'm using FastDDS, as I'm on Foxy running the default DDS vendor.

jamoralp, 2021-03-18:
Yes. That is the only difference. The graph manager can be disabled in micro-ROS applications because it consumes more memory resources, so it also makes sense to being able to disable it here. Also, consider it as a workaround for the problem I mentioned about the Fast DDS versioning. We are working on fixing it already

bjv-capra, 2021-03-22:
So, I've tried connecting using the raw agent, but after the session is made, there is no topic being created nor anything...
```
[1615534912.555541] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615534912.576765] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1615534929.905309] info     | Root.cpp           | create_client            | create                 | client_key: 0x15B6EBDB, session_id: 0x81
[1615534929.906139] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x364309467, address: 10.46.28.2:18170
```
I built the agent following https://github.com/micro-ROS/micro_ros_setup/issues/284#issuecomment-785701353
When creating the subscription, I do get a `RCL_RET_OK`, idk what could be going on now...

pablogs9, 2021-03-22:
are you using the `-v6` flag in the agent?

bjv-capra, 2021-03-22:
Sorry for the wall of logs...
When I manually start the agent, after the clinet has been pinging waiting for it to start
```
[1615535043.753321] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1615535043.755103] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1615535043.825502] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1615535043.826983] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x00000000, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1615535043.828691] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 28 12 D3 11 81 00 FC 01
[1615535043.829061] info     | Root.cpp           | create_client            | create                 | client_key: 0x2812D311, session_id: 0x81
[1615535043.829444] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x672322321, address: 10.46.28.2:32869
[1615535043.829836] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1615535043.830894] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 96, data: 
0000: 81 80 00 00 01 05 56 00 00 0A 00 01 01 02 00 00 47 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 65 6D 6F 74 65 3C 2F 6E 61 6D 65 3E 3C
0040: 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 00 00
[1615535043.832275] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615535043.834151] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615535043.838126] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615535043.846144] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1615535043.853168] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2812D311, participant_id: 0x000(1)
[1615535043.853489] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1615535043.853625] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.853751] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.853843] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.853990] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.854036] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.860110] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1615535043.860872] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2812D311, len: 136, data: 
0000: 81 80 01 00 01 05 7D 00 00 0B 00 02 02 02 00 00 6F 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 72 74 2F 72 6F 62 6F 74 2F 72 65 6D 6F 74 65 2F 69 6E 3C 2F 6E 61 6D 65 3E 3C
0040: 64 61 74 61 54 79 70 65 3E 67 65 6F 6D 65 74 72 79 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73
0060: 5F 3A 3A 54 77 69 73 74 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64
0080: 73 3E 00 00 01 00 00 00
[1615535043.862074] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2812D311, topic_id: 0x000(2), participant_id: 0x000(1)
[1615535043.862245] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1615535043.862324] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2812D311, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[... cut ...]
0000: 81 80 45 00 09 01 34 00 00 0E 00 06 D3 0E 86 D9 61 56 8E C0 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 C0
[1615535039.844145] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x15B6EBDB, len: 60, data: 
0000: 81 80 46 00 09 01 34 00 00 0E 00 06 8E 69 3A 51 92 8E 8E C0 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 C0
```

pablogs9, 2021-03-22:
Are you able to see the topic using `ros2 topic echo ...`?

bjv-capra, 2021-03-22:
Yes, from the agent-side. I'm not publishing anything from the client to the agent. I'm publishing from the agent side to the client side. I do see some messages, then it's like a burst, then it stops completely.

pablogs9, 2021-03-22:
Is possible for you to do a test? Just build and use the bare Micro XRCE-DDS Agent, we are fixing right now a bug in the micro-ROS agent and I want to discard that this is related to the bug. To do so:

```bash
git clone -b foxy https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent
mkdir build
cd build
cmake .. 
make -j$(nproc)
```

Now you have an executable in the `build` directory that can be used as the micro-ROS Agent.

Let me know if you are having the same problems with this approach.

bjv-capra, 2021-03-22:
That's actually what I'm using.

However, I've just noticed a weird behavior on our side of the code, which makes me wonder if some bug was introduced on the client-side that could be causing problems. I'll be back if nothing changes.

pablogs9, 2021-03-22:
I'm closing to clean our issue tracking system, but feel free to reopen if you found that is not your client code that is causing problems.

If so, please provide feedback.

bjv-capra, 2021-03-23:
I've continued to debug, what seems to happen is that after hitting a breakpoint, the communication with the agent starts to act weird...You can see a log from the agent (below), where I'm sending one Twist message every 1 second, then I put a breakpoint in the code and what's bellow happens. Everything works fine, before and if I continue from the breakpoint, the program seems to continue running. However, if I let the program stay in the breakpoint for too long, or if I run with a higher frequency (in the end, the same), then the uros task stops working properly and starts to run every 5 seconds (the scheduler doesn't care about that task anymore).

I have a strange feeling this could have to do with the QoS used, however, why would the agent send so many messages when the breakpoint is hit?

```
1616489112.578070] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 60, data: 
0000: 81 80 20 01 09 01 34 00 00 0E 00 06 F0 40 52 4A 63 56 8D 40 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489112.578681] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 21 01 00 00 80
[1616489113.577831] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 48, data: 
0000: B7 31 C0 11 D9 A8 8D 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489113.577993] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 60, data: 
0000: 81 80 21 01 09 01 34 00 00 0E 00 06 B7 31 C0 11 D9 A8 8D 40 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489113.578655] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 22 01 00 00 80
[1616489114.577814] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 48, data: 
0000: 04 AA 03 9F E3 F3 8D 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489114.578014] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 60, data: 
0000: 81 80 22 01 09 01 34 00 00 0E 00 06 04 AA 03 9F E3 F3 8D 40 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489114.578683] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 23 01 00 00 80
[1616489115.577791] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 48, data: 
0000: 34 C8 E7 2C 70 37 8E 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489115.578005] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 60, data: 
0000: 81 80 23 01 09 01 34 00 00 0E 00 06 34 C8 E7 2C 70 37 8E 40 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489115.761383] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 23 01 80
[1616489115.961509] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 23 01 80
[1616489116.161512] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 23 01 80
[1616489116.361690] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 23 01 80
[1616489116.561770] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 23 01 80
[1616489116.577126] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 48, data: 
0000: 5E 29 F9 D5 6D 73 8E 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489116.577199] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 60, data: 
0000: 81 80 24 01 09 01 34 00 00 0E 00 06 5E 29 F9 D5 6D 73 8E 40 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 40
[1616489116.761978] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x19CD30F5, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 23 01 24 01 80
[... cut ...]
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=3",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=2",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=<agent ip>",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
            ]
        }
    }
}
```

bjv-capra, 2021-03-23:
Wouldn't seem like I can reopen the issue

bjv-capra, 2021-03-23:
The problem could also be on my network stack, I'm using FreeRTOS+TCP. Maybe I have some configuration missing.

bjv-capra, 2021-03-23:
So, I've narrowed down the problem to, I'm running out of buffers on the network stack. This only happens when I set a breakpoint, I wait, and when I continue from the breakpoint, buffers are taken and never released. If the frequency of the publisher is high, or I wait enough time, I'll run out of buffers.

I still don't quite understand why the agent spams the client like that when the communication stops (in this case, because of the breakpoint).

Furthermore, can you point me to the implementation of the freertos+tcp port? Because I'm concerned that maybe it's there where the buffers are not being released.

bjv-capra, 2021-03-23:
Quick update,
The moment I stop the agent, all the buffers are released, meaning it definitely has something to do with how the traffic from the agent is being held on the client.

Furthermore, the interrupts on the task go up to 5s(fixed) when I run out of buffers, and go back to normal when I stop the agent. I'll guess there's some part kidnapping the buffers during interrupts when on breakpoint

Sorry for spamming like this, I'm just trying to keep this thread updated with what I find out.

pablogs9, 2021-03-23:
Don't you worry, we are interested in the evolution of this solution. The mentioned "spaming" is just the agent sending heartbeats and acknacks to the client. Maybe you can configure it with [this](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/5124991014d701626069f0f4a2b312a3233fcb94/CMakeLists.txt#L56) and [this](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/604f8dd8c785b9a60cf001b1d1b6d0cae26b1d94/CMakeLists.txt#L58). Also there is best effort modes for pub/sub entities that will not acknowledge the reception. 

In any case, I think that is important to learn how to tune the FreeRTOS+TCP buffering memory to make micro-ROS works well, please keep us updated about the progress.

bjv-capra, 2021-03-24:
The mentioned spammed, is weird, because when I put a breakpoint on the system, the agent starts shooting 10 times more messages (you can check it on the logs). The heartbeat should be a consistent period, and the agent/client comms seemed to go 1:1 (acks) until this happens. You can see that on the last log. Where it suddenly starts sending all the messages.

I'll have to dig more into the tuning of this, I guess it has to do with how it's being polled.

bjv-capra, 2021-04-07:
Hi, changing the QoS to best effort instead of reliable seems to fix the issue. However, if I'd have a topic that needs to be reliable, I'd run into the same issue when debugging. I couldn't find explicitly what is it that causes the whole problem. But it's not on the agent side, more likely on the protocol.

closed 2021-04-07 by bjv-capra