micro-ROS/micro-ROS-Agent #62: Running two agents fighting

amfern, 2021-04-25:
**Describe the bug**
When running two agents on the same PC but different MCUs, will see topics periodically of either one but not both

**To Reproduce**
Steps to reproduce the behaviour:
1. Run agent of first MCU and wait for it to connect
2. Run agent of second MCU and wait for it to connect
3. Stop second agent and start it again and wait for MCU ro reconnect
3. Run `ros2 topic list` or `ro2 node list` repeatedly
4. Every run i seed either topics from second MCU or first MCU
    Doing ros2 topic echo will sometimes fail with `Could not determine the type for the passed topic`
    Doing ros2 topic echo on topic sometimes will not continue after agent reconnection

**Expected behaviour**
When running eProsima agents, i don't exhibit the same behavior

**System information (please complete the following information):**
 - OS:  Ubuntu 18.04
 - ROS 2 Foxy
 - Version foxy

pablogs9, 2021-04-26:
How are you running the agents? Are you installing them? Using the dockerized version?

amfern, 2021-04-26:
I am using docker

amfern, 2021-04-26:
As you suggested i tried to mount /dev/shm and using --ipc=host, does solve the issue. but not completely.
If i `ros2 topic echo topic_from_mcu` and then restart the agent, it will not continue printing, i have to manually rerun `ros2 topic echo topic_from_mcu`. This is also true for any node. I have to restart the node to continue working. this doesn't happend if i compile the agent.

This is the command i use:
```
docker run --net=host --ipc=host --init -it --privileged -v /dev:/dev microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -b 230400 -v6
```

pablogs9, 2021-04-26:
This is a FastDDS problem in the usage of shared memory inside dockers, not fully related to micro-ROS. 

Is there a problem for you in building the micro-ROS Agent in your host and running from there?

amfern, 2021-04-26:
Using docker allowed me to run the newest version of the agent and save me the compilation on the target pc(arm64), but I can use the dashing version right now, its the only one i can compile on host.
Can you point me to the issue in fastDDS?

pablogs9, 2021-04-26:
https://github.com/eProsima/Fast-DDS/issues/1698#issuecomment-778039676

amfern, 2021-04-28:
I guess I can close this issue, thanks

closed 2021-04-28 by amfern