micro-ROS/micro-ROS-Agent #63: Snap agent refresh? arm64 ?

artivis, 2021-04-30:
The [micro-ros-agent snap](https://snapcraft.io/micro-ros-agent) latest release was back in December and as such it does not seem to work anymore with the latest micro-ros changes. I could confirm that my program does run fine with the Docker version tho.

Could we get a refresh of the snap?

On a side note, the snap is only release for amd64. Could we get an arm64 release in order to use it on e.g. a Raspberry Pi?

artivis, 2021-05-17:
For futur reference, note that there exists github actions to [build](https://github.com/snapcore/action-build) and [publish](https://github.com/snapcore/action-publish) the snap automatically.

Arnatious, 2021-05-28:
Taking a look at it - looks like https://github.com/micro-ROS/micro-ROS-Agent/commit/99b3d98430e612b5cf294a22415aa18f822daba5 added shared memory, which will need approval on the snapcraft forums. I opened https://github.com/micro-ROS/micro-ROS-Agent/issues/65 to investigate it.

amfern, 2021-09-26:
@artivis i also need the arm64 version, currently i compiled manually multiarch version of this image at https://hub.docker.com/layers/amfernus/micro-ros-agent/galactic/images/sha256-dd7c17125c58fddbc90a5d7fc8852d66d3af162d64d80ab12c050ab2e12d4407?context=repo

OtaviodaCruz, 2022-01-14:
Any update on this?
I'm trying to use the micro-ros agent on my raspberry pi 3 B+ with ubuntu 20.04 server along with my arduino due (micro-ros-arduino) and I'm not succeeding as reported by the users above.

when I use

> sudo micro-xrce-dds-agent serial --dev /dev/ttyACM0 -v6

I have the following output:

> [1642198472.816118] info | TermiosAgentLinux.cpp | init | running... | fd: 3
> [1642198472.821656] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6
> [1642198473.583251] info | Root.cpp | create_client | create | client_key: 0x07720432, session_id: 0x81
> [1642198473.583591] info | SessionManager.hpp | establish_session | session established | client_key: 0x124912690, address: 0
> [1642198473.584105] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x07720432, len: 19, date:
> 0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
> [1642198473.594523] debug | SerialAgentLinux.cpp | recv_message | [==>> BE <<==] | client_key: 0x07720432, len: 48, date:
> 0000: 81 80 00 00 01 07 28 00 00 0A 00 01 01 03 00 00 19 00 00 00 00 01 08 20 11 00 00 00 73 63 69 63
> 0020: 6F 62 6F 74 5F 61 72 64 75 69 6E 6F 00 00 00 00
> [1642198473.594972] debug | ProxyClient.cpp | create_participant | DDS error | client_key: 0x07720432, participant_id: 0x000(1)
> [1642198473.595388] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x07720432, len: 14, date:
> 0000: 81 80 00 00 05 01 06 00 00 0A 00 01 80 00
> [1642198473.595663] debug | SerialAgentLinux.cpp | send_message | [** <<SER>> **] | client_key: 0x07720432, len: 13, date:
> 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
> [1642198473.602488] debug | SerialAgentLinux.cpp | recv_message | [==>> BE <<==] | client_key: 0x07720432, len: 13, date:
> 0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80

There seems to be an error when trying to create the participant.

Tried to do as explained in [this post](https://answers.ros.org/question/373503/micro-ros-agent-on-raspberry-pi-3/)

OtaviodaCruz, 2022-01-15:
After a few hours... i got it

I'll leave it here for future reference. The key point (I believe) is to use the --edge flag for installation, which takes it from the development branch, where it should have solved the problem (correct me if I'm wrong), so we have:

`sudo snap install micro-xrce-dds-agent --edge`

`snap set micro-xrce-dds-agent daemon=true`

`sudo snap refresh core --edge`

`sudo snap set core experimental.hotplug=true`
 
`sudo systemctl restart snapd`

`snap serial-port interface`

`snap connect micro-xrce-dds-agent:serial-port snapd:<slots snapd>`

`sudo micro-xrce-dds-agent serial --dev /dev/ttyACM0 -v6`

Basic references:
- https://snapcraft.io/micro-xrce-dds-agent
- https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#installation-label

closed 2022-06-13 by Acuadros95

rituraj-navindgikar, 2022-07-08:
hey @OtaviodaCruz 
can i have an issue solved please
[image]
this is what i get error ..but i m trying to run on raspberry pi pico.
i guess my installation method is wrong
please could u help me thanks in advance

artivis, 2022-07-08:
@rituraj-navindgikar Looks like you are using the `micro-xrce-dds-agent`. You may want to take this issue to the appropriate GitHub repo: [eProsima/Micro-XRCE-DDS-Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent).

rituraj-navindgikar, 2022-07-08:
> @rituraj-navindgikar Looks like you are using the `micro-xrce-dds-agent`. You may want to take this issue to the appropriate GitHub repo: [eProsima/Micro-XRCE-DDS-Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent).

Okay thanks ... 1 qs does this agent work for pi 4 which has arm64

artivis, 2022-07-08:
> Okay thanks ... 1 qs does this agent work for pi 4 which has arm64

Looking at the [micro-xrce-dds-agent page on the snap store](https://snapcraft.io/micro-xrce-dds-agent), it has been release for the arm64 architecture as well.

Acuadros95, 2022-07-11:
@rituraj-navindgikar, as stated in a previous comment, snap arm64 version is not updated:

> But I am afraid that there are no plans to support arm64 at this moment.

We recommend to use the dockerized version or to build from source.