micro-ROS/micro-ROS-Agent #65: Snap access to shared memory not allowed

Arnatious, 2021-05-28:
https://github.com/micro-ROS/micro-ROS-Agent/blob/003cb38ef2fe2935027f78268121eed18fcb421b/snap/snapcraft.yaml#L153

A shared memory implementation was added recently, which requires approval on the snap store, or per https://forum.snapcraft.io/t/request-for-system-files-micro-ros-agent-arnatious/24642/5, the best option is making sure the shared memory is named `/dev/shm/snap.micro-ros-agent.xxx`. We ~might~ be able to use a layout to avoid hardcoding this.

Arnatious, 2021-06-04:
Layouts can't be applied to `/dev/` paths, so that option is out.

I tried https://forum.snapcraft.io/t/bashtop-linux-resource-monitor/17441/3 but it didn't solve the errors. For reference, the new snapcraft yaml (With some other changes for cleanup of the build):

```
name: micro-ros-agent-arnatious
base: core20
version: git
summary: Bridge between Micro ROS client applications and ROS 2
description: |
  Micro-ROS, whose default implementation is based on eProsima's
  Micro XRCE-DDS middleware, is composed of client applications
  which interact with the ROS 2 world by means of an Agent.
  This agent keeps tracks of the entities created by means of
  requests performed on the microcontroller side, and uses them
  to communicate with the ROS 2 dataspace.

  Being an extension of the Micro XRCE-DDS Agent, the micro-ROS
  agent supports being run by the user like this:

      $ micro-ros-agent --help

  In addition, the Agent supports running as a service that can be
  enabled with:

      $ snap set micro-ros-agent daemon=true

  If the service is enabled, by default it uses the `udp4` transport on
  port 8888. The following parameters can be changed (these are
  specific to the service, the `micro-ros-agent` command simply
  takes command-line arguments, but the capabilities are the same):

  * `transport`. Supported transports are `udp4`, `udp6`, `tcp4`,
    `tcp6`, `serial`, and `pseudoterminal`. Default is `udp4`. Change
    with:

        $ snap set micro-ros-agent transport="new transport"

  * `middleware`. Supported kinds of middleware are `ced`, `rtps`, and
    `dds`. Default is `dds`. Change with:

        $ snap set micro-ros-agent middleware="new middleware"

  * `verbosity`. Supported verbosity levels are 0-6, defaulting to 4.
    Change with:

        $ snap set micro-ros-agent verbosity="selected verbosity"

  * `discovery`. Enable or disable the discovery server. Defaults to
    "false". Change with:

        $ snap set micro-ros-agent discovery="true or false"

  * `discovery-port`. Port on which the discovery server (see above)
    listens. Defaults to 7400. Change with:

        $ snap set micro-ros-agent discovery-port="selected port"

  * `p2p-port`. Port to use for the P2P profile. Change with:

        $ snap set micro-ros-agent p2p-port="selected port"

  * `port`. Port on which the agent listens. Only applicable to one of
    the UDP or TCP transports (see above). Defaults to 8888. Change with:

        $ snap set micro-ros-agent port="selected port"

  * `baudrate`. Baud rate to use when accessing serial ports. Only
    applicable when using the `serial` or `pseudoterminal` transport.
    Defaults to 115200. Change with:
[... 98 lines cut ...]
    plugin: cmake
    cmake-parameters:
      - -DCMAKE_INSTALL_PREFIX=/usr
    build-packages:
      - on amd64:
        - gcc-multilib
        - g++-multilib

apps:
  micro-ros-agent:
    command: usr/bin/snapcraft-preload $SNAP/lib/micro_ros_agent/micro_ros_agent
    environment:
        LD_LIBRARY_PATH: "$LD_LIBRARY_PATH:$SNAP/lib"
        SNAPCRAFT_PRELOAD_REDIRECT_ONLY_SHM: 1
    plugs: [network, network-bind, serial-port]
    extensions: [ros2-foxy]

  daemon:
    command: usr/bin/snapcraft-preload $SNAP/usr/bin/micro-ros-agent-daemon
    environment:
        LD_LIBRARY_PATH: "$LD_LIBRARY_PATH:$SNAP/lib"
        SNAPCRAFT_PRELOAD_REDIRECT_ONLY_SHM: 1
    daemon: simple
    plugs: [network, network-bind, serial-port]
```

artivis, 2021-06-15:
For future reference, there is a [discussion on the snapcraft forum](https://forum.snapcraft.io/t/request-for-system-files-micro-ros-agent-arnatious).

artivis, 2022-06-08:
For reference, #141 disables shared memory usage in the snap.

pablogs9, 2022-06-09:
Does #141 closes this?

Acuadros95, 2022-06-09:
Yes! Latest foxy version has been released to `latest/stable`

closed 2022-06-09 by Acuadros95