micro-ROS/micro-ROS-Agent #67: Segfault starting discovery

BrettRD, 2021-06-10:
**Describe the bug**
Segfault starting micro-ros agent with discovery service

segfault in ` eprosima::uxr::UDPv4Agent::init_discovery(unsigned short)`
stack trace:
```
#0  0x00007ffff7e672fd in eprosima::uxr::UDPv4Agent::init_discovery(unsigned short) () from /home/brettrd/esp/agent_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0
#1  0x00007ffff7dd7fe5 in eprosima::uxr::agent::create_agent_thread<eprosima::uxr::UDPv4Agent>(int, char**, eprosima::uxr::agent::TransportKind, __sigset_t const*)::{lambda()#1}::operator()() const () from /home/brettrd/esp/agent_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0
#2  0x00007ffff729ed84 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff73b4609 in start_thread (arg=<optimised out>) at pthread_create.c:477
#4  0x00007ffff70dd293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
identical stack traces with `UDPv4Agent`, `UDPv6Agent`, `TCPv4Agent`, `TCPv6Agent`

**To Reproduce**
`docker run -it --rm --net=host -v /dev/shm:/dev/shm -v /dev:/dev --privileged  microros/micro-ros-agent:foxy udp4 --port 8888 -v6 -d`
```
[1623304450.075639] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
<end of output, docker exits>
```
This also applies to `dashing` and `galactic` fresh docker pulls

using micro-ros-agent built from source, gdb was attached by adding `prefix=['xterm -e gdb -ex run --args'],` to the launch file.

In all of these builds, `udp4 --port 8888 -v6 ` works normally,  `udp4 --port 8888 -v6 -d` segfaults

**Expected behaviour**
micro-ros-agent should not segfault.

micro-ros agent should run normally similar to:
`docker run -it --rm --net=host -v /dev/shm:/dev/shm -v /dev:/dev --privileged  microros/micro-ros-agent:foxy udp4 --port 8888 -v6`

**System information (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS 2 Foxy
 - Versions:
   - Foxy latest, pulled by micro-ros-setup foxy branch
   - Docker microros/micro-ros-agent:dashing
   - Docker microros/micro-ros-agent:foxy
   - Docker microros/micro-ros-agent:galactic
 

**Additional context**
Micro-XRCE-DDS-Agent at e6e2cf6e4c5ebbd0302a2ca9e90c9e263279a726
micro-ros-agent at 003cb38ef2fe2935027f78268121eed18fcb421b

pablogs9, 2021-06-10:
I have not been able to reproduce this:

- A fresh build of the `foxy` micro-ROS Agent in a fresh `ros:foxy` docker: https://asciinema.org/a/AjI2D7s1FJydSZKDOqVmSWpEr
- A fresh run of the `microros/micro-ros-agent:foxy`: https://asciinema.org/a/aFsq2CwceyH2iVCKsWMnDyd2K

Can you reproduce this issue in an isolated machine in order to discard any issue with your setup or network?

BrettRD, 2021-06-10:
Weird. Without rebuilding or rebooting, all of my agents are fine.

All of the docker containers are fine without a fresh pull, (still no topics visible from those but that's unrelated), even the source build is fine now (same binary, same launch)

I'll try another machine if it happens again.

closed 2021-06-10 by BrettRD