micro-ROS/micro-ROS-Agent #68: Is TCP fully supported on Galactic

wuhanstudio, 2021-06-12:
## Issue template

- Hardware description: STM32F103RCT6 + ESP8266
- RTOS: RT-Thread
- Installation type: similar to arduino package
- Version or commit hash: galactic

#### Steps to reproduce the issue

Hi, I tried the **micro-ros_publisher_wifi** example, but the micro-ros-agent gives no response if I started the server using TCP.

```
docker run -it -p 9999:9999/tcp --privileged microros/micro-ros-agent:galactic tcp4 -p 9999
```

I can confirm that packets are sent to the agent:

```
【Receive from 192.168.199.120 : 46495】：
80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5
63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45
01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 
10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 
80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 
63 50 81 00 FC 01 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 10 E5 63 50 81 00 FC 01 
```

#### Expected behavior

A topic is published.

#### Actual behavior

No response from the tcp agent.

#### Additional information

For the same example, everything works well using serial or udp protocol.

Thus, I'm wondering if TCP support is still working in progress.

Thanks for your help.

pablogs9, 2021-06-14:
Hello TCP is a valid transport for the micro-ROS middleware (Micro XRCE-DDS) but it is not implemented in micro-ROS RMW. Do you have any special interest in using TCP as transport for micro-ROS?

I'm closing since this is not planned to be implemented soon.

closed 2021-06-14 by pablogs9

wuhanstudio, 2021-06-14:
Hi, thanks for your help.

The serial transport is quite stable, but it requires a physical connection with Linux running micro-ros agent. Thus I would like to try wireless connection using TCP or UDP.

I tried UDP connection using the example:
https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher_wifi/micro-ros_publisher_wifi.ino

```
docker run -it -p 9999:9999/udp--privileged microros/micro-ros-agent:galactic udp4 -p 9999
```

It was quite stable initially, but after publishing the 88th INT32 message from the MCU, the micro-ros agent became unresponsive (no further messages published). And this issue happens every time on the 88th INT32 message. I can confirm that the UDP packet were still being sent from the MCU, but the agent gave no response after the 88th message. Thus I would like to try TCP transport.

Do you have any idea about the possible causes of UDP transport unresponsiveness?

pablogs9, 2021-06-14:
Can you check if after that message the MCU is still in a running state? How much stack have you configured for your tasks in the micro-ROS client?

wuhanstudio, 2021-06-14:
Thanks for your quick help. 

The MCU is still running because the serial prints out the message, and the LED is blinking. Besides, I can confirm the UDP packets is sent out by starting another UDP server so that I can see the packet.

pablogs9, 2021-06-14:
Can you share the agent output with the flag `-v6` enabled?

wuhanstudio, 2021-06-14:
Thanks for your kindly help. I'll record a video that illustrates the issue and prints out the agent output with -v6 enabled.

wuhanstudio, 2021-06-14:
After enabling the option -v6, it seems like some packets get lost after the 88th packet.

```
[1623664728.255457] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x25B9DF4A, len: 16, data:
0000: 81 01 56 00 07 01 08 00 00 64 00 05 56 00 00 00
[1623664728.255772] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data:
0000: 56 00 00 00
[1623664729.267467] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x25B9DF4A, len: 16, data:
0000: 81 01 57 00 07 01 08 00 00 65 00 05 57 00 00 00
[1623664729.267904] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data:
0000: 57 00 00 00
[1623664730.281170] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 16, data:
0000: 81 01 58 00 07 01 08 00 00 66 00 05 58 00 00 00
[1623664731.292380] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 16, data:
0000: 81 01 59 00 07 01 08 00 00 67 00 05 59 00 00 00
```

https://user-images.githubusercontent.com/15157070/121874849-ac0a1200-ccff-11eb-9c69-127b9876695f.mp4

pablogs9, 2021-06-14:
Please do one test, the system stops working after a certain number of messages or after a certain amount of time?

If you increase the publication rate to for example 10 Hz, it also stops when 88 msgs have been published?

pablogs9, 2021-06-14:
Also, I see that you are not spinning the micro-ROS node in your code. Try to [spin](https://github.com/micro-ROS/micro_ros_arduino/blob/3ffb4fbbcf724e793cb4fe9cfab57d15af555aab/examples/micro-ros_publisher/micro-ros_publisher.ino#L81) the executor or if you do not have an executor, try to spin the middleware with an `rmw_wait`. This issue was explained [here](https://github.com/micro-ROS/micro_ros_arduino/issues/183#issuecomment-805740091)

wuhanstudio, 2021-06-14:
It seems that the system is more likely to stop after a certain amount of time.

It stops at 783 if I increase the publication rate to 10Hz, and stops at 2577 under 50Hz.

wuhanstudio, 2021-06-14:
Thanks for your suggestion, I'll try to add the executor.

BTW, I use the example from Arduino as a reference which does not use an executor as well.

https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher_wifi/micro-ros_publisher_wifi.ino

https://github.com/wuhanstudio/micro_ros/blob/foxy/examples/micro_ros_pub_int32_udp.c

pablogs9, 2021-06-14:
I'm going to modify the arduino example to use an executor.

If the error is time-based, I recommend to try to spin the middleware by adding an executor, using the `rmw_wait` or using a **reliable** publisher (that internally spin the executor).

If the behavior continues, we should take a look into the roll over mechanism implemented in the [`clock_gettime`](https://github.com/micro-ROS/micro_ros_arduino/blob/3ffb4fbbcf724e793cb4fe9cfab57d15af555aab/src/default_transport.cpp#L17) function of your port.

wuhanstudio, 2021-06-14:
> Also, I see that you are not spinning the micro-ROS node in your code. Try to [...]

Hi, perhaps it's the same issue with heartbeats. But the problem is not solved by adding rmw_wait using UCP transport:

https://github.com/micro-ROS/micro_ros_arduino/issues/183#issuecomment-805760375

```
    while(1)
    {
        rt_tick_t start_loop_time = rt_tick_get();

        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
        rcl_publish(&publisher, &msg, NULL);
        msg.data++;

        // Run the XRCE session a bit
        rmw_wait(NULL, NULL, NULL, NULL, NULL, NULL, &((rmw_time_t){.sec = 0, .nsec = 1e6}));

        // If loop rate is less than required, wait necessary time
        while (rt_tick_get() - start_loop_time < (1000000.0 / LOOP_RATE))
        {
            rt_thread_mdelay(10);
        }
    }
```

Unfortunately, only two messages are published after adding rmw_wait:

```
[1623673826.068393] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data:
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 11 2F BB AE 81 00 FC 01
[1623673826.068541] info     | Root.cpp           | create_client            | create                 | client_key: 0x112FBBAE, session_id: 0x81
[1623673826.068602] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x288340910, address: 172.17.0.1:61341
[1623673826.068848] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 19, data:
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1623673826.099658] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 56, data:
0000: 81 80 00 00 01 05 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 00 20 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 72 74 74 5F 75 64 70 5F 6E 6F 64 65 00 00 00 00 00 00
[1623673826.107157] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x112FBBAE, participant_id: 0x000(1)
[1623673826.108112] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 14, data:
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1623673826.108165] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1623673826.140146] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1623673826.168695] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 100, data:
0000: 81 80 01 00 01 05 5A 00 00 0B 00 02 02 03 00 00 4C 00 00 00 24 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 5F 72 6F 73 5F 72 74 74 5F 75 64 70 5F 6E 6F 64 65 5F 70 75 62 6C 69 73 68 65 72 00 00 01 00 00
0040: 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 49 6E 74 33 32 5F 00
0060: 00 01 00 00
[1623673826.168951] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x112FBBAE, topic_id: 0x000(2), participant_id: 0x000(1)
[1623673826.169161] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1623673826.169210] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1623673826.195636] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 00 00 01 00 80
[1623673826.196061] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1623673826.367069] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1623673826.489001] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 01 00 00 01 80
[1623673826.489549] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 14, data:
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1623673826.515513] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1623673826.531193] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 24, data:
0000: 81 80 02 00 01 05 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1623673826.531540] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x112FBBAE, publisher_id: 0x000(3), participant_id: 0x000(1)
[... 12 lines cut ...]
[1623673826.579366] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x112FBBAE, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[... cut ...]
[1623673827.648511] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data:
0000: 01 00 00 00
[1623673828.651896] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x112FBBAE, len: 16, data:
0000: 81 01 02 00 07 01 08 00 00 10 00 05 02 00 00 00
[1623673828.652144] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data:
0000: 02 00 00 00
```

pablogs9, 2021-06-14:
What about using `rclc_publisher_init_default` instead of `rclc_publisher_init_best_effort`?

wuhanstudio, 2021-06-14:
After running several tests using `rclc_publisher_init_default`, the counter now stops at random position (after random number of messages being sent), and no packet are

```
static void microros_pub_int32_thread_entry(void *parameter)
{
    while(1)
    {
        rt_tick_t start_loop_time = rt_tick_get();

        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
        rcl_publish(&publisher, &msg, NULL);
        msg.data++;

        // If loop rate is less than required, wait necessary time
        while (rt_tick_get() - start_loop_time < (1000000.0 / LOOP_RATE))
        {
            rt_thread_mdelay(10);
        }
    }
}

static void microros_pub_int32_udp(int argc, char* argv[])
{
    set_microros_udp_transports("192.168.199.100", 9999);

    allocator = rcl_get_default_allocator();

    //create init_options
    if (rclc_support_init(&support, 0, NULL, &allocator) != RCL_RET_OK)
    {
        rt_kprintf("[micro_ros] failed to initialize\n");
        return;
    };

    // create node
    if (rclc_node_init_default(&node, "micro_ros_rtt_udp_node", "", &support) != RCL_RET_OK)
    {
        rt_kprintf("[micro_ros] failed to create node\n");
        return;
    }
    rt_kprintf("[micro_ros] node created\n");

    // create publisher
    rclc_publisher_init_default(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
        "micro_ros_rtt_udp_node_publisher");

    rt_kprintf("[micro_ros] publisher created\n");

    // create executor
    rclc_executor_init(&executor, &support.context, 1, &allocator);
    rt_kprintf("[micro_ros] executor created\n");

    msg.data = 0;

    rt_thread_t thread = rt_thread_create("mr_pubint32_udp", microros_pub_int32_thread_entry, RT_NULL, 2048, 25, 10);
    if(thread != RT_NULL)
    {
        rt_thread_startup(thread);
        rt_kprintf("[micro_ros] New thread mr_pubint32_udp\n");
    }
    else
    {
        rt_kprintf("[micro_ros] Failed to create thread mr_pubint32_udp\n");
    }
}
```

pablogs9, 2021-06-15:
Maybe it is a network stack problem, can you debug it?

wuhanstudio, 2021-06-15:
Thanks sincerely for your help. 

It feels like issues related to micro_ros have been fixed. The rest may come from the RTOS scheduling. I'll try to fix this by investigating  the RT-Thread documentation.

Thanks again for your kindly help.