micro-ROS/micro-ROS-Agent #70: Multiserial agent parsing arguments incorrectly in docker image

ruimscarvalho98, 2021-06-25:
## Issue template

- Hardware description: 2x Crazyflie 2.1
- RTOS: freeRTOS
- Installation type: Docker (not working) and micro-ros-setup (working)
- Version or commit hash: foxy

#### Steps to reproduce the issue
I am running the multiserial agent both in the docker container and using the micro-ros-setup helper scripts.
For the docker container I simply run:
```
sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy multiserial --devs "/dev/ttyS10 /dev/ttyS20"  -v6
```
Using the micro-ros-setup i run (as root to have permission to read/write in /dev)  after building the agent and sourcing the workspace:
```
ros2 run micro_ros_agent micro_ros_agent multiserial --devs "/dev/ttyS10 /dev/ttyS20" -v6
```
#### Expected behavior
The expected behaviour is the one i get in the micro-ros-setup building process where I receive data from *both* serial devices:
[image]

#### Actual behavior
Inside docker I only read data from one serial device - the first one to be precise, which leads me to think maybe the parsing is not working with docker. I've tried running both (without the bridges) and as you can see in the screenshot bellow the docker one only looks for the first serial port altough the command is the same:
[image]

#### Additional information
For connecting the crazyflies I am running two separate bridges with two radios and two separate socat connections: /dev/ttyS10 <-> /dev/ttyS11 and  /dev/ttyS20 <-> /dev/ttyS21

pablogs9, 2021-06-25:
@Acuadros95 can you check this?

ruimscarvalho98, 2021-06-25:
Hey again! After fiddling around with this a bit more, I found that the problem can be solved by replacing the line:

`exec ros2 run micro_ros_agent micro_ros_agent $@`

In the dockerfile's entrypoint script with:

`exec ros2 run micro_ros_agent micro_ros_agent "$@"`

more info about this can be seen [here](https://www.gnu.org/software/bash/manual/html_node/Special-Parameters.html).

It's an extremely simple change but if you want me to open a PR for it let me know

Another option, would be to replace the parsing method to not use a string such as `multiserial --devs /dev/ttyXX /dev/ttyYY /dev/ttyZZ` which might be a bit cleaner and more intuitive imo.

pablogs9, 2021-06-28:
Can you open the PR please?

ruimscarvalho98, 2021-06-28:
Done

pablogs9, 2021-06-30:
https://github.com/micro-ROS/docker/pull/62 merged, closing

closed 2021-06-30 by pablogs9