micro-ROS/micro-ROS-Agent #71: can't receive "complex messages"

protobits, 2021-06-29:
**Describe the bug**
I'm running uROS on Zephyr and have the uROS agent running on docker. I can publish an Int32 message, just like the provided sample, and receive it on the dockerized ros (testing with `ros2 topic echo`). However, I'm now trying to publish a sensor_msgs/BatteryState and `ros2 topic echo` fails with Python error.

**To Reproduce**
1. publish a sensor_msgs/BatteryState message on uROS
2. run `ros2 topic echo` on receiving side

As output I get:
<pre>make_tuple(): unable to convert arguments to Python object (compile in debug mode for details)</pre>

**Expected behaviour**
Receiving the published message.

**System information (please complete the following information):**
 - OS: uROS on Zephyr, uROS agent on docker
 - ROS 2 Galactic

**Additional context**
These two lines are the two publishers I have setup on uROS side, for context:
<pre>
	rclc_publisher_init_default(&battery_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, BatteryState), "battery_state");
	rclc_publisher_init_default(&test_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "test");
</pre>

pablogs9, 2021-06-30:
Can you share your client code? Are you [initializing the memory of the type](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/)?

protobits, 2021-06-30:
Thanks, I wasn't aware of those tutorials. I'm now initializing the message with:
<pre>
float voltages[1] = { 0.0f };
float temperatures[1] = { 0.0f };

rcl_publisher_t battery_publisher;
sensor_msgs__msg__BatteryState battery_msg =
{
  .cell_voltage = { .data = voltages, .size = 1, .capacity = 1 },
  .cell_temperature = { .data = temperatures, .size = 1, .capacity = 1 },
  .location = { .data = "", .size = 0, .capacity = 1 },
  .serial_number = { .data = "", .size = 0, .capacity = 1 },
};
</pre>

The uROS is initialized with:
<pre>
rcl_ret_t ret;
  
  /* setup uROS transport */
	
	rmw_uros_set_custom_transport(
		MICRO_ROS_FRAMING_REQUIRED,
		(void *) &default_params,
		zephyr_transport_open,
		zephyr_transport_close,
		zephyr_transport_write,
		zephyr_transport_read
	);

	/* setup uROS */
	
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
  ret = rcl_init_options_init(&init_options, allocator);
  if (ret != RCL_RET_OK)
  {
    LOG_ERR("Failed rcl_init_options_init: %i\n", ret);
    goto errout;
  }

  rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);
  ARG_UNUSED(rmw_options); // TODO: do i need this?

	rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator);

	/* setup node */
	
	rcl_node_t node;
	rclc_node_init_default(&node, "zrover", "zrover", &support);

  /* create publishers */

	rclc_publisher_init_default(&battery_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, BatteryState), "battery_state");
	rclc_publisher_init_default(&odo_publisher, &node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry), "odom");

	/* create executor */
		
	rclc_executor_init(&executor, &support.context, 1, &allocator);
</pre>

I then call `rclc_executor_spin_some(&executor, 100);` in a loop. To publish I do: `rcl_ret_t ret = rcl_publish(&battery_publisher, &battery_msg, NULL);`

However, I still get the same error.

pablogs9, 2021-06-30:
Can you copy here your whole code? I don't know if voltages[] and temperatures[] are static...

Have you tried using code snippets like the ones shown in the tutorial?

protobits, 2021-06-30:
> Can you copy here your whole code? I don't know if voltages[] and temperatures[] are static...

They are globals. I uploaded a gist here: https://gist.github.com/protobits/59c3d05ce973e50f1f62e1539e969085
 
> Have you tried using code snippets like the ones shown in the tutorial?

I'm following the examples and applying them to this datatype.
Most of the code I use is based on the provided uROS sample part of the uROS Zephyr module.

pablogs9, 2021-06-30:
I'm seeing that you are using Galactic, are you using `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`?

Remember that micro-ROS is only compatible with Fast-DDS middleware. Can you try to change it and check if it works?

pablogs9, 2021-06-30:
Another comment is that you have a bunch of variables that exist in `uros_init` stack and not in `communicator_loop` and are used in the spin executor.

Try to make `rcl_allocator_t allocator`, `rclc_support_t support`, `rcl_init_options_t init_options` and `rcl_node_t node` global variables.

protobits, 2021-06-30:
> I'm seeing that you are using Galactic, are you using `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`?
> 
> Remember that micro-ROS is only compatible with Fast-DDS middleware. Can you try to change it and check if it works?

Note that I'm using the dockerized version of the agent, using the "galactic" tag of the container.
For `ros2 topic echo` I'm exec-ing a bash on the same running container. I tried to export that variable and then
run `ros2 topic echo` again and I get:
<pre>
[ERROR] [1625062830.756422013] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library 'librmw_fastrtps_cpp.so' due to dlopen error: librmw_fastrtps_cpp.so: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-galactic-rcutils-4.0.2/src/shared_library.c:99, at /tmp/binarydeb/ros-galactic-rmw-implementation-2.4.1/src/functions.cpp:75', exiting with 1., at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/rmw_implementation_identifier_check.c:139
</pre>
Looking at the available libraries I see a `librmw_fastrtps_shared_cpp.so` so I did:
<pre>
export RMW_IMPLEMENTATION=rmw_fastrtps_shared_cpp
</pre>
but now the error is:
<pre>
[ERROR] [1625062806.789788002] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_shared_cpp'), with error message 'failed to resolve symbol 'rmw_get_implementation_identifier' in shared library '/opt/ros/galactic/lib/librmw_fastrtps_shared_cpp.so', at /tmp/binarydeb/ros-galactic-rmw-implementation-2.4.1/src/functions.cpp:104', exiting with 1., at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/rmw_implementation_identifier_check.c:139
</pre>

protobits, 2021-06-30:
> Another comment is that you have a bunch of variables that exist in `uros_init` stack and not in `communicator_loop` and are used in the spin executor.
> 
> Try to make `rcl_allocator_t allocator`, `rclc_support_t support`, `rcl_init_options_t init_options` and `rcl_node_t node` global variables.

I moved the loop inside uros_init() so everything was local and I even added the publish inside that loop, but the problem remains. Maybe it is on the agent side and there's some problem with the galactic version of the container?

pablogs9, 2021-07-01:
I have been testing your code and:
1. You are not initing string insider [`header`](https://github.com/ros2/common_interfaces/blob/42fb5cced5edc6b2fde8a0d9237e1528f93fe4dd/sensor_msgs/msg/BatteryState.msg#L34) in your BatteryState
2. You are not initing anything in `odo_msg`, it also has multiple string members: https://github.com/ros2/common_interfaces/blob/master/nav_msgs/msg/Odometry.msg

Remember that publishing a type with not initialized pointers will result in an error or a crash.

pablogs9, 2021-07-01:
You can check when the new agent docker is built here: https://hub.docker.com/repository/docker/microros/micro-ros-agent/builds

pablogs9, 2021-07-26:
Any update on that?

protobits, 2021-07-26:
Sorry, I got side tracked with other projects. I will return to this eventually. Most likely is the issue with the missing header.
If toy prefer to close this out is ok. I will update it once I test or again.

closed 2021-07-26 by pablogs9