micro-ROS/micro-ROS-Agent #87: port dosen't appear with snap interface

MuhammadUbuntu, 2021-07-11:
I am new to micro-ros and trying to communicate with Rasbperry Pi Pico micro-ros-agent but when i execute 
snap interface serial-port
I see the following output
name:    serial-port
summary: allows accessing a specific serial port
plugs:
  - arduino
  - micro-ros-agent

and It doesn't show any port. Any suggestion please!

pablogs9, 2021-07-12:
Currently, the micro-ROS snap package is not working properly, we are working on a solution. Is it possible for you to build the agent from scratch with the setup tool: [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup)?

MuhammadUbuntu, 2021-07-12:
@pablogs9  I have already tried [micro_ros_setup ](https://github.com/micro-ROS/micro_ros_setup)and when i run 
`ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407`
i have gotten some errors.

pablogs9, 2021-07-12:
Have you run it with `sudo` as the errors message says?

` Please re-run with superuser privileges.`?

MuhammadUbuntu, 2021-07-12:
I have tried it with `sudo`  and the error message was something like `colcon finished with stlerr.` . After that i have deleted all uros_ws and didn't complete ahead.

pablogs9, 2021-07-12:
Could you be more specific and paste here your output with sudo in the agent launch?

closed 2022-01-28 by pablogs9