micro-ROS/micro-ROS-Agent #90: Building agent fails on galactic

roni-kreinin, 2021-08-20:
- Hardware description: Olimex E407
- RTOS: FreeRTOS
- Installation type:  micro_ros_setup
- Version or commit hash: galactic

#### Steps to reproduce the issue
Following https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/

When I run `ros2 run micro_ros_setup build_agent.sh` I get build errors:

`micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:543:86: error: ‘ALIVE_INSTANCE_STATE’ is not a member of ‘eprosima::fastdds::dds::InstanceStateKind’`

`micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:645:20: error: ‘class eprosima::fastdds::dds::DataWriterQos’ has no member named  data_sharing’`

and a couple other errors regarding the `data_sharing` member.

Any idea why I am getting this error? I noticed that Micro-XRCE-DDS-Agent is on foxy branch and does not have a galactic branch, which may be part of the problem.

Acuadros95, 2021-08-23:
Hello @roni-kreinin,

Micro-XRCE-DDS-Agent foxy branch is intended to be used in any ROS2 environment, the build process will search or install the Fast-DDS version accordingly.

It looks like the build process failed to install the correct Fast-DDS version during the `ros2 run micro_ros_setup create_agent_ws.sh` command. You can check the installed version with the following command: `dpkg -l | grep fastrt`, the output should look like this:

```
root@username-XPS-15-9560:/microros_ws# dpkg -l | grep fastrt
ii  ros-galactic-fastrtps                               2.3.1-1focal.20210507.022611         amd64        Implementation of RTPS standard.
ii  ros-galactic-fastrtps-cmake-module                  1.2.1-2focal.20210507.022211         amd64        Provide CMake module to find eProsima FastRTPS.
ii  ros-galactic-rmw-fastrtps-shared-cpp                5.0.0-2focal.20210630.223833         amd64        Code shared on static and dynamic type support of rmw_fastrtps_cpp.
ii  ros-galactic-rosidl-typesupport-fastrtps-cpp        1.2.1-2focal.20210630.221753         amd64        Generate the C++ interfaces for eProsima FastRTPS.
```

How did you install ROS? Do you also have foxy installed in the same enviroment?
Make sure to follow all the steps in order, focusing on the initial `source /opt/ros/$ROS_DISTRO/setup.bash`.

Anyway, you can also test the tutorial and flash the board on a docker, which should work just fine:
`docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged ros:galactic`

Acuadros95, 2021-09-28:
Closing this due to inactivity, feel free to reopen if needed.

closed 2021-09-28 by Acuadros95