micro-ROS/micro-ROS-Agent #96: Is it available as a ROS2 package

wuhanstudio, 2021-08-28:
Hi,

I notice this interesting news that announces micro-ROS included in ROS 2 common packages.

https://www.eprosima.com/index.php/company-all/news/183-micro-xrce-dds-and-micro-ros-included-in-ros-2

But if I search for micro-ros:

```
$  sudo apt-cache search micro-ros
```

I only see

```
$ ros-galactic-rclc-parameter - Parameter server implementation for micro-ROS nodes
$ ros-rolling-rclc-parameter - Parameter server implementation for micro-ROS nodes
```

Is is possible to install micro-ros-agent via something like `sudo apt install ros-galactic-micro-ros-agent`?

amfern, 2021-09-26:
I would also like that, what would it take to make it available as apt package?

bjsowa, 2022-03-11:
The package seems to be correctly versioned for foxy, galactic and rolling distributions, so it's just a matter of creating a release repository and running bloom. I can take care of releasing it to ROS2 distributions. I'll just need some approval from the code maintainers. @pablogs9 ?

amilcarlucas, 2022-03-18:
@pablogs9 ping

ralph-lange, 2022-03-19:
micro_ros_setup it a second package that I think would be good to have a binary release of.

pablogs9, 2022-03-21:
We will release a new repository soon with all those packages available as precompiled. I'll report here.

bjsowa, 2022-03-21:
Why not just release the package for ROS distribution and leave all the work of building and packaging to ROS buildfarm?

pablogs9, 2022-03-22:
Working on that, we will bring news about that in a week.

v-lopez, 2022-04-12:
Any news regarding this?

bjsowa, 2022-05-27:
@pablogs9 any update?

JaimeMartin, 2022-05-29:
Hi guys,

It is ready, we will announce it this next week.

pablogs9, 2022-05-31:
Hello everyone, we have released the [Vulcanexus tool package](https://discourse.ros.org/t/vulcanexus-the-all-in-one-ros-2-tool-set/25818), so now the micro-ROS Agent can be used as a prebuilt binary package.

In order to add the Vulcanexus apt repositories, you have instructions here: https://docs.vulcanexus.org/en/latest/rst/installation/linux_binary_installation.html

Once you have our package repos, just do:

```
apt install vulcanexus-humble-micro-ros-agent
```

Once you have the installation just run it:

```
source /opt/vulcanexus/humble/setup.bash
ros2 run micro_ros_agent micro_ros_agent [ARGS]
```

closed 2022-05-31 by pablogs9