micro-ROS/micro-ROS-Agent #101: Micro-ROS-Agent / Teensy 4 subscriber: not updating Custom Message header/frame_id, and float64 (Foxy)

MatthewWalstra, 2021-09-28:
## Issue

#### Context 
Trying to write custom messages with float and header datatypes for communication between Raspberry Pi and Teensy 4, but the header.frame_id isn't updating and the float values are massive and random, which only occurs in the custom msg, not when running the header or float msgs individually.

- Hardware description: x86_64 ubuntu 20.04 desktop running micro-ros-agent:foxy, teensy 4 running micro-ros:foxy
- RTOS: teensy/arduino
- Installation type: 
1. x86_64: base micro-ros-setup from [Teensy with Micro-Ros tutorial](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/) with custom message copied into src directory and built before building micro-ros-agent. Full folder: ([microros_ws.zip](https://github.com/micro-ROS/micro-ROS-Agent/files/7240294/microros_ws.zip))
2. Teensy 4: flashing, compiling, and monitoring using PlatformIO vscode extension (Github: [micro_ros_ws](https://github.com/Calvin-CRFC/micro_ros_ws/tree/master))
3. Custom msg is copied into micro-ros-arduino:foxy extras/library_generation/extra_packages/ and compiled using docker container, then it's copied from micro-ros-arduino/src/ into the micro_ros_ws/lib/micro-ros-arduino folder:
````
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy4
````

#### Steps to reproduce the issue 
- clone above github repository and download microros_ws.zip on x86_64
- Open cloned github repository in PlatformIO and checkout branch:
1. SubInt: std_msgs/msg/Int32 subscriber
2. SubFloat: std_msgs/msg/Float32 subscriber
3. SubTest: std_msgs/msg/Header subscriber
4. SubCustomTest: drive_controller_msgs/msg/Tank subscriber
- Upload code to Teensy 4.0 using PlatformIO
- Start Serial Monitor on Serial 1 (requires external UART <-> USB Converter)
- Build and source microros_ws.zip
- Unplug Teensy 4, start micro-ros-agent, then plug-in Teensy 4
- In a new terminal tab source 'microros_ws/install/local_setup.bash' and run:
```
ros2 topic echo micro_ros_arduino_subscriber
```
- In a 3rd terminal source 'microros_ws/install/local_setup.bash' and run, depending on branch:
1. SubInt: 
```
ros2 topic pub --once /micro_ros_arduino_subscriber std_msgs/msg/Int32 'data: 0'
```
2. SubFloat:
```
ros2 topic pub --once /micro_ros_arduino_subscriber std_msgs/msg/Float64 'data: 0.54321'
```
3. SubTest: 
```
ros2 topic pub --once /micro_ros_arduino_subscriber std_msgs/msg/Header '{stamp: {sec: 2, nanosec: 520}, frame_id: base_link}'
```
4. SubCustomTest:
```
ros2 topic pub --once /micro_ros_arduino_subscriber drive_controller_msgs/msg/Tank '{header: {stamp: {sec: 2, nanosec: 520}, frame_id: base_link}, left: 0.5, right: -0.5}'
```
#### Expected Behavior
- In the Serial1 terminal monitor should see depending on branch:
1. SubInt: 
````
Data: 0
````
2. SubFloat: 
````
Data: 0.543210
````
3. SubTest: 
````
base_link: (2,520)
````
4. SubCustomTest: 
````
base_link[2,520]: (0.5, -0.5)
0.5
-0.5
````
- In ros2 topic echo terminal, should see respective message sent from the ros2 topic pub terminal

#### Actual Behavior
- In the Serial1 terminal monitor should see depending on branch:
1. SubInt (correct): 
````
Data: 0
````
2. SubFloat (correct): 
````
Data: 0.543210
````
3. SubTest (correct): 
````
base_link: (2,520)
````
4. SubCustomTest (error: frame_id isn't updated and left and right outputs are massive random numbers, numbers change when teensy and agent are restarted): 
````
Hello World[2,520]: (-1956265796568251648.000000, 5859499033423634..(~40 more digits)..4550443423956992.000000)
ovf
ovf
````

#### Additional Questions
1. Why do the values update in their individual msgs, but not when combined into a custom msg?
2. Does a general tutorial exist for creating custom msgs for micro-ros on teensy?
3. Is there an easier way to create a static library with the docker container (other than copy, build, copy), or can I use micro-ros-setup/scripts/build_firmware.sh with a Teensy and PlatformIO?
4. How do I see the msgs sent over serial to the teensy, so I can debug if it's the agent or micro-ros causing the issue?

Acuadros95, 2021-09-28:
I just tested your repository and you are using an outdated micro-ROS library.

Please pull our repository to the latest foxy version: https://github.com/micro-ROS/micro_ros_arduino/tree/foxy, get the latest builder version with `docker pull microros/micro_ros_static_library_builder:foxy` and rebuild your library.

As for your questions:
1, 2. Your approach was correct, the message is compiled correctly and is ready to be used. I think the problem is related to use a compound message with a string inside, but you are using everything correctly.
3. Currently we don't offer support for PlatformIO. I strongly recommend to use the current approach and keep up with the automatic updates of the repository, but you can always give it a try,
4. You can run the agent in debug mode with the flag `-v6`, but it is probably more useful to debug on the board using serial output as you are currently doing.

Also and just as a detail, you don't need to set an initial value to the string here:
```
strcpy(msg.header.frame_id.data, "Hello World");
msg.header.frame_id.size = strlen(msg.header.frame_id.data);
```
Ideally, you only set values on messages you want to publish.

MatthewWalstra, 2021-09-28:
If I don't set the string header, it gives a bunch of garbage characters over serial as expected.

I still need to test building using the micro-ros-setup, and once I can build custom message types, I'll test sending 2 float64s without the header.

How much of the problem would be resolved if I switched to using galactic?

Acuadros95, 2021-09-28:
> I pulled the latest docker image, foxy 2.0.1 release and cloned the latest foxy branch from the github page, but both library builds failed with this message:

Please pull and try again, should be fixed now.

MatthewWalstra, 2021-09-28:
I pulled again, and the build was successful. I'll test building custom messages with micro-ros-setup and sending 2 float64s without a header, then I'll let you know what I find out.

MatthewWalstra, 2021-09-29:
Updating to the latest version of micro_ros_arduino resolved the issue.

closed 2021-09-29 by MatthewWalstra