micro-ROS/micro-ROS-Agent #102: Agent create only client and participant

biontix, 2021-10-02:
Hello everyone! I am using microros in xilinx ultra96 with freeRTOS where I run the client. Microros Agent is running on the Host PC and the communication client/agent is via serial. I am trying to run simple demo int32_publisher found in micro-ROS-demos.

When I run the agent with `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB1 -v 6` I can see in the log that the session is established, it create client and partecipant and than stopped to create other entities. In the client side I use my own custom transport, with xilinx serial API uartps.h. The problem could be in the read Operation?

My read and write operation: 

[image]

biontix, 2021-10-03:
Problem fixed: XUartPs_Send is non-blocking function, so sometimes happened while the UART was sending, XUartPs_Recv was call causing bytes loss. We need to wait until UART finish sending

closed 2021-10-03 by biontix

sapertuz, 2023-06-08:
Hello @crivi30 

How were you able to embed MicroRos on the ZynqMP? 
Do you have any resources you can share on how to achieve this?