micro-ROS/micro-ROS-Agent #104: Questions regarding the xml generation.

xptosoft, 2021-12-13:
_uRos: During the build process, the package looks for all ROS 2 messages to generate an initial list of XML profiles._

Q1: When you build the Agent, are these profiles used to static link the existent messages to the agent OR are used to generate a configuration file ?
Q2: After building the Agent, and a new message (used by client) is added to the host do I need to rebuild the agent or can I adjust the existing xml file.

_These profiles can are referenced in the Agent-Client communication to avoid sending the full XML content. This reference mechanism can be switched on and off from the Micro XRCE-DDS middleware layer._

Q3: I'm not following... Can you please explain or point to the documentation ? 

Thank you!

pablogs9, 2021-12-13:
Hello @xptosoft:

Q1: they are supposed to be just sample XML configurations on the agent side, not related to linkage.
Q2: agent does not need to know any type information, so if you want to use a new message type in the client side, you do not need to rebuild or recofigure anything in the agent side.
Q3: XRCE-DDS clien has 3 modes for creating entities in the agent: XML, ref and binary. You can find an explanation here: https://micro-xrce-dds.docs.eprosima.com/en/latest/client.html#creation-mode-client

xptosoft, 2021-12-19:
Plabo: 
**Thanks!** 

So I did a few more tests, and it is working, although I have a few more questions below:

My test setup:

1. VM1: Agent running without any knowledge of firmware workspaces and/or custom messages.
2. ESP32 firmware publishing (UDP) a custom message.
3. VM2 without any knowledge of firmware workspaces and/or custom messages.

On VM2:
when I run:
`ros2 topic list` 
first time does not show the firmware topic. I run a second time the same command and the topic is there. 
**Q1)** Is this normal  ?  Is there a delay fetching the topics ? or is a delay obtaining the topics on first call to the ros2 topic list ?

when I run:
`ros2 topic echo /my_custom_message` 
the result is _The message type 'my_custom_message/msg/MyCustomMessage' is invalid._
**Q2)** Is this expected, if you don't have the message available on your host, you can't read the message ?

I added my_custom_message pkg, build and sourced and I run the same command and the topic is displayed.
It makes sense the definition is presented in the host, so ros2 topic echo can obtain the metadata and display the message.
**Q3)** So for any third party ros2 node to decode a custom message, requires a previous knowledge of message ? Is not possible to ask for the message definition and parse it at runtime ?

**Q4a)** I changed my_custom_message renaming the fields, compiled, sourced and ros2 topic echo shows the message content.
**Q4b)** I changed my_custom_message removing a field from the message, compiled, sourced and ros2 topic echo shows the message with content broken over a few fields.
Is this expected ? I imagined changing the message definition (field names) or adding/removing fields would break the decoding similar to ROS1 (md5) can you please comment ?

pablogs9, 2021-12-20:
> **Q1)** Is this normal ? Is there a delay fetching the topics ? or is a delay obtaining the topics on first call to the ros2 topic list ?

Usually this delay is related to discovery at DDS / ROS 2 level. The first time you launch a command with `ros2 ...` the ROS 2 daemon is enabled, probably the micro-ROS nodes are discovered later.

> when I run: `ros2 topic echo /my_custom_message` the result is _The message type 'my_custom_message/msg/MyCustomMessage' is invalid._ **Q2)** Is this expected, if you don't have the message available on your host, you can't read the message ?

If you want to use a message type in ROS 2 you need to build it in the system where you are expecting to use it. This is not about micro-ROS this is ROS 2.

> I added my_custom_message pkg, build and sourced and I run the same command and the topic is displayed. It makes sense the definition is presented in the host, so ros2 topic echo can obtain the [...]

Same as above

> **Q4a)** I changed my_custom_message renaming the fields, compiled, sourced and ros2 topic echo shows the message content. **Q4b)** I changed my_custom_message removing a field from the message, [...]

No, ROS 2 does not check type integrity along with different hosts. It just assumes the same structure. As said previously, please read ROS 2 documentation for those questions.

I'm closing, do not hesitate to reopen if you have more questions.

closed 2021-12-20 by pablogs9