micro-ROS/micro-ROS-Agent #105: Dependencies missing in manifest

norro, 2021-12-13:
**Describe the bug**
Dependencies [declared as required in cmake](https://github.com/micro-ROS/micro-ROS-Agent/blob/galactic/micro_ros_agent/CMakeLists.txt#L32-L50) are missing [in the manifest file](https://github.com/micro-ROS/micro-ROS-Agent/blob/galactic/micro_ros_agent/package.xml#L18-L22).

**To reproduce**
I encountered this issue when trying to revive an old micro-ROS agent (foxy) installation on my ubuntu PC.  Following [this tutorial](https://micro.ros.org/docs/tutorials/core/first_application_linux/), I
1. Pulled the micro_ros_setup repo
2. Executed create_agent_ws
3. Executed build_agent
---- Up to here it is expected behavior, the agent is built properly.

When I then execute colcon build in my ROS 2 workspace, the build of the micro_ros_agent package fails as it is missing microxrcedds_agent. When I place the microxrcedds_agent repo in my workspace, it will still fail on the first colcon run, since microxrcedds_agent is not specified as dependency in the micro-ROS agent package manifest, so colcon doesn't know to build it first.

**Expected behaviour**
All ROS 2 packages in my workspace successfully build

**System information (please complete the following information):**
Ubuntu 20.04, ROS 2 foxy, latest foxy branch

**Additional context**
Add any other context about the problem here.

pablogs9, 2021-12-13:
Hello @norro,

I have replicated this by creating an agent with `micro_ros_setup` utils and then building the environment with colcon. No error found: https://asciinema.org/a/o8YSI2H6duo65BwtNGYDc2QJe

Could you provide precise steps for replicating this?

norro, 2021-12-13:
@pablogs9 First of all thanks for the fantastic video 👍 that's a great way to show how you reproduced it, I like! 👍

I cleaned up everything again and managed to get rid of the issue now as well. my suspicion is that last time I forgot to *colcon build* micro_ros_setup after updating it and before calling the script to create the agent workspace.

That being said, I
a) still don't understand how this is working properly without having all dependencies declared in the manifest. Is the create workspace script taking care of all of this?
b) wonder if it still makes sense to declare all cmake dependencies in the manifest as well (although it doesn't seem to be necessary in this setup), as it somehow feels right / cleaner.

pablogs9, 2021-12-13:
> a) still don't understand how this is working properly without having all dependencies declared in the manifest. Is the create workspace script taking care of all of this?

micro-ROS agent takes care of downloading Micro XRCE-DDS Agent using a superbuild approach, check here: https://github.com/micro-ROS/micro-ROS-Agent/blob/galactic/micro_ros_agent/cmake/SuperBuild.cmake So we do not need to have an apt precompiled build of the XRCE-DDS Agent.

> b) wonder if it still makes sense to declare all cmake dependencies in the manifest as well (although it doesn't seem to be necessary in this setup), as it somehow feels right / cleaner.

Agree, we should clean the manifest. I'll keep this open to solve this.