micro-ROS/micro-ROS-Agent #117: Setting a static participant ID for micro ROS Agent

omertal88, 2022-01-26:
Hello,

I'm trying to set a static participant ID using an XML profile for the micro ROS Agent node, but when I tried to run it, it said `RTPSParticipant with the same ID already exists -> Function createParticipant`.
Then, looking at the open ports, I noticed the same process uses 4 UDP (unicast) ports (which means 2 participants).
Why are there 2 participants?
In ROS 2 documentation it states:
[image]

I still want to be able to control the participants IDs from the XML profile. Is there a way around it?

Thank you.
Omer

pablogs9, 2022-01-26:
In micro-ROS, each node has its own participant, which differs from ROS 2. 
Could you detail your use case and the requirements you have? Why do you need to have only one participant in micro-ROS Agent side?

omertal88, 2022-01-26:
Hello @pablogs9 
I have a limited communication channel between two devices. I wanted to reduce the discovery data to minimum possible.
So for nodes that only communicate locally, I'm using SPDP (since they don't add any traffic to my comm channel).
For nodes who communicate between the devices, I wanted to set the specific ports (participants) which they need to discover. For that reason, I needed each node to have a unique participant ID.

omertal88, 2022-01-27:
I just want to clarify:
It can be OK to have two participants for micro-ROS-Agent, but only if I can statically determine their IDs.

In Fast-RTPS when a given participant ID is taken, it just goes ahead and takes the next one.
In micro-ROS for some reason, it just throws an error.

pablogs9, 2022-01-27:
Could you provide XML and all the information needed in order to replicate this use case? We'll check how both FastDDS and the XRCE-Agent are behaving.

BTW: moving this to Micro XRCE-DDS Agent repo

omertal88, 2022-01-27:
@pablogs9 of course

```<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >

        <transport_descriptors>
            <transport_descriptor>
                <transport_id>DroneTransport</transport_id>
                <type>UDPv4</type>
                <maxInitialPeersRange>20</maxInitialPeersRange>
            </transport_descriptor>
        </transport_descriptors>

        <participant profile_name="global_unicast" is_default_profile="true">
            <domainId>0</domainId>
            <rtps>

                <participantID>106</participantID>

                <builtin>
                    <discovery_config>
                        <leaseDuration>
                            <sec>10</sec>
                            <nanosec>0</nanosec>
                        </leaseDuration>

                        <leaseAnnouncement>
                            <!-- DURATION -->
                            <sec>4</sec>
                            <nanosec>0</nanosec>
                        </leaseAnnouncement>
                    </discovery_config>
                    <metatrafficUnicastLocatorList>
                        <locator>
                            <udpv4>
                                <address>0.0.0.0</address>
                            </udpv4>
                        </locator>
                    </metatrafficUnicastLocatorList>
                    <initialPeersList>
                        <locator><udpv4><address>127.0.0.1</address></udpv4></locator>
                        
                        <!-- List of Remote peers -->
                        <locator><udpv4><address>192.0.1.15</address><port>7552</port></udpv4></locator>
                        <locator><udpv4><address>192.0.1.15</address><port>7554</port></udpv4></locator>
                        <locator><udpv4><address>192.0.1.15</address><port>7560</port></udpv4></locator>
                        <locator><udpv4><address>192.0.1.15</address><port>7566</port></udpv4></locator>
                        <locator><udpv4><address>192.0.1.15</address><port>7568</port></udpv4></locator>
                        <locator><udpv4><address>192.0.1.15</address><port>7572</port></udpv4></locator>
                    </initialPeersList>
                </builtin>
                
                <userTransports>
                    <transport_id>DroneTransport</transport_id>
                </userTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

pablogs9, 2022-01-27:
And your micro-ROS application? How are you using the micro-ROS client side?

omertal88, 2022-01-27:
We're running a micro-ros-arduino client which communicates with the device using a UART port. (edited)

pablogs9, 2022-01-28:
Hello @omertal88,

(CC: @EduPonz)

Let me explain a little bit what we have been investigating:
- `participantID` is used inside Fast DDS in order to generate the Participant GUID and to select the UDP ports that it will use.
- `participantID` is not the only parameter taken into account in order to generate the Participant GUID.
- UDP unicast ports are calculated using the `participantID`, but if Fast DDS found that port in use will take the contiguous one.
- ROS 2 uses one Fast DDS participant per ROS 2 context (usually one context per process) not per ROS 2 node
- micro-ROS uses one Fast DDS participant per micro-ROS node.
- micro-ROS will only use default parameters set in the XML if entities are being created using the binary entity creation mode (the default one). If XML mode is used, default parameters are not going to be taken into account.
- It is not possible to use the same UDP unicast ports in multiple Fast DDS participants. Is possible to communicate participants user data using multicast instead of unicast, in that case listening ports will be reused. Check [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/use_cases/many_subscribers/many_subscribers.html)

So, in the case of running two Talker Listener demos we have **two different processes** running different instances of Fast DDS, so we have two Fast DDS Participants because we have two different ROS 2 contexts. When both processes read the default profiles from your XML they take the same default `participantID` and:
- Each one calculates a different Participant GUID because not only `participantID` is taken into account for this calculation.
- The first takes the UDP ports that are calculated using `participantID`
- The second takes the contiguous ones because the one calculated are in used.

In the case of micro-ROS, we have only one process running Fast DDS in the micro-ROS Agent, so it will fail to create two Fast DDS Participants with the same `participantID`.

If you want to control which ports are beeing used by Fast DDS we propose this XML configuration file:

```xml
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>DroneTransport</transport_id>
                <type>UDPv4</type>
                <maxInitialPeersRange>20</maxInitialPeersRange>
            </transport_descriptor>
        </transport_descriptors>

        <participant profile_name="global_unicast" is_default_profile="true">
            <domainId>0</domainId>
        
            <rtps>
<!-- NOTE 3 -->
                <defaultUnicastLocatorList>
                    <locator>
                        <udpv4>
                            <address>localhost</address>
                            <port>7800</port>
                        </udpv4>
                    </locator>
                </defaultUnicastLocatorList>

                <builtin>
                    <discovery_config>
                        <leaseDuration>
                            <sec>10</sec>
                            <nanosec>0</nanosec>
                        </leaseDuration>

                        <leaseAnnouncement>
                            <!-- DURATION -->
                            <sec>4</sec>
                            <nanosec>0</nanosec>
                        </leaseAnnouncement>
                    </discovery_config>
                    
                    <metatrafficUnicastLocatorList>
                        <locator>
                            <udpv4>
                                <address>0.0.0.0</address>
<!-- NOTE 2 -->
                                <port>7600</port>
                            </udpv4>
                        </locator>
                    </metatrafficUnicastLocatorList>
                </builtin>
                
                <userTransports>
                    <transport_id>DroneTransport</transport_id>
                </userTransports>

<!-- NOTE 1 -->
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

_**NOTE 1**: You weren't disabling built-in transport, so in your configuration two UDP transport were being created (the default one and the one that you were defining). That why you saw double the ports._

_**NOTE 2**: You also can set the UDP port that is being involved in metatraffic unicast procedure. The second participant will try to use this default port and as it will be in use, finally it will take the contiguos (that is UDP/7602)._

_**NOTE 3**: You also can set the locator for unicast user traffic. Same as above first participant will use UDP/7800 and second will use UDP/7802._

omertal88, 2022-01-30:
OK I think that really solved my problem.
So the approach I should have take was to set the listening ports, instead of the participant IDs. I guess that makes sense.

Now that I set port 7600 (I know it doesn't match the 106's participant ID - will change later) as the listening ports, I see both metadata unicast ports opened 7600 and 7602, which is good enough for me. But I still don't quite understand why are there two ports (two nodes) running under the same process.
```
$ sudo ss -tulpn | grep micro_ros_agent
udp   UNCONN 0      0                              0.0.0.0:43792        0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=16))                 
udp   UNCONN 0      0                              0.0.0.0:7421         0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=14))                 
udp   UNCONN 0      0                              0.0.0.0:7423         0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=26))                 
udp   UNCONN 0      0                              0.0.0.0:7600         0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=13))                 
udp   UNCONN 0      0                              0.0.0.0:7602         0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=25))                 
udp   UNCONN 0      0                            127.0.0.1:53238        0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=17))                 
udp   UNCONN 0      0                            127.0.0.1:46573        0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=29))                 
udp   UNCONN 0      0                              0.0.0.0:34434        0.0.0.0:*                    users:(("micro_ros_agent",pid=8196,fd=28))                 
```

Thank you very much for your assistance so far, and for your future assistance :)

pablogs9, 2022-02-03:
Maybe if you are launching nodes with ROS 2 one participant is your own participant and the other is the ROS 2 daemon. Is this a possibility in your use case?

pablogs9, 2022-02-15:
Any update on this?

omertal88, 2022-02-17:
Hi @pablogs9 ,

Sorry I didn't realize this is still opened.
I don't think the daemon is running, but regardless, setting the ports solved my issue. Discovery is now super efficient!

Thank you very much!

closed 2022-02-17 by omertal88