micro-ROS/micro-ROS-Agent #121: Guarnatee sucessful communication from main device running the agent

Johannap1, 2022-02-26:
- Hardware description: ESP32 dev board
- RTOS: Arduino
- Version or commit hash: ROS2 foxy

Hello, 

I am currently working on implementing micro-ros on an ESP32dev board and ROS2 foxy running on my laptop. For now, I am starting the agent from a separate terminal and then subscribing to the published topics on a node implemnted with rospy. Currently, I am checking a connection of the agent only visually by the output in the terminal. However, ideally I would like to know if the agent was successfully setup from my rospy node and make sure that it discovered the MCU, to guarantee that the communication is running from the main device and not only on the MCU side (for that I was looking at the reconnection example). This would be similar to a handshake functionality at startup in a serial communication setting. It should furthermore open the possibility to detect from my rospy node if connection to the device was lost during runtime.
I also thought that in the end it should be possible to just check that messages are coming in or implementing a service upon startup, but that would already require a successful communication setup and no possibility to automatically reconnect from the main device side if the communication is not working for some reason.
Could you please help me out and tell me if I am simply missing something here or if this functionality is not available yet. Also, if this has not been implemented yet I would appreciate some hints where to look at to solve this issue.

Thanks

pablogs9, 2022-02-28:
Hello @Johannap1,

In the micro-ROS client-side you can implement application logic for checking if the agent is available using the ping functionality. You have the example code in the reconnection example.

On your ROS 2 application side, the only way to check if micro-ROS node is correctly enabled is by checking the ROS 2 graph, for example, using [`get_node_names()`](https://docs.ros2.org/latest/api/rclpy/api/node.html?highlight=graph#rclpy.node.Node.get_node_names) in order to check if your micro-ROS node is alive.

Johannap1, 2022-02-28:
Hello @pablogs9, 

thank you for your quick answer. Indeed my question concerns more the ROS 2 application side. I tried your suggestion using the get_node_names(). This should work to check whether the agent is running on startup to see whether the node is there, but what I discovered is that when connection is lost to the microcontroller (either unplugging it or killing the agent), the node remains for up to half a minute before it cannot be detected by ROS anymore. So this would not be a solution to detecting if connection is lost. Do you see the same behavior with your system?

pablogs9, 2022-03-01:
Hello @Johannap1,

On the ROS 2 side you have two situations:
- If the agent is killed, micro-ROS entities will appear for at least 20 seconds, that is the default [lease time](https://fast-dds.docs.eprosima.com/en/latest/fastdds/discovery/general_disc_settings.html#discovery-lease-dur) of the underlaying DDS participant.
- If the MCU is unplugged, the entities should appear forever as the micro-ROS agent is not allowed to kill entities without the client's request. This is the default behavior of XRCE-DDS, the underlying middleware for micro-ROS.

If you want to check the liveliness of your MCU system, my recommendation would be to send a topic in a certain rate and check this rate at the ROS 2 side. For this deadline QoS can be handy, check [this](https://answers.ros.org/question/352472/ros2-qos-deadline-usage/)

pablogs9, 2022-03-08:
Any update on this?

pablogs9, 2022-03-17:
Some comments here: https://discourse.ros.org/t/our-conclusions-on-trying-to-implement-micro-ros-in-our-robot/24690/2?u=pablogs

pablogs9, 2022-06-14:
Closing due to inactivity

closed 2022-06-14 by pablogs9