micro-ROS/micro-ROS-Agent #122: ssize_t not declared in this scope

rosterloh, 2022-03-01:
**Describe the bug**
When trying to build micro-ROS-Agent in a ROS 2 source workspace on a Raspberry Pi I get the following errors.
```
In file included from /home/pi/ros2_galactic/build/micro_ros_agent/agent/src/xrceagent/src/cpp/transport/stream_framing/StreamFramingProtocol.cpp:15:
/home/pi/ros2_galactic/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/transport/stream_framing/StreamFramingProtocol.hpp:98:41: error: ‘ssize_t’ was not declared in this scope; did you mean ‘size_t’?
   98 |     using WriteCallback = std::function<ssize_t (
      |                                         ^~~~~~~
      |                                         size_t
```
**To Reproduce**
Steps to reproduce the behaviour:
1. Setup build environment as [here](https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html) 
2. git clone https://github.com/micro-ROS/micro_ros_msgs.git -b galactic src/uros/micro_ros_msgs
3. git clone https://github.com/micro-ROS/micro-ROS-Agent.git -b galactic src/uros/micro_ros_agent
4. colcon build --install-base /opt/ros/galactic --merge-install --packages-up-to micro_ros_agent --cmake-args "-DUAGENT_BUILD_EXECUTABLE=OFF -DUAGENT_P2P_PROFILE=OFF --no-warn-unused-cli"

**Expected behaviour**
A successful build.

**System information:**
 - OS: Debian Bullseye 64 bit
 - ROS 2 Galactic
 - Version galactic

rosterloh, 2022-03-01:
[image]

rosterloh, 2022-03-01:
I think this is related to https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/297. SuperBuild is using ros2 branch and this fix is in develop

rosterloh, 2022-03-01:
Confirmed working when changing https://github.com/micro-ROS/micro-ROS-Agent/blob/galactic/micro_ros_agent/cmake/SuperBuild.cmake#L29 to `develop`

AntumArk, 2022-03-01:
Are you running on RPI3b+ or 4?

rosterloh, 2022-03-01:
@AntumArk on a CM 4

Acuadros95, 2022-03-01:
Just updated `ros2` branch with develop changes, should fix this problem.

pablogs9, 2022-03-01:
Please confirm the fix and we will close

rosterloh, 2022-03-01:
This will definitely fix it. Happy to close

closed 2022-03-01 by rosterloh