micro-ROS/micro-ROS-Agent #123: Build micro-ROS agent failed

Jimsing-Wong, 2022-03-09:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I was trying to install micro-ROS following the [ESP32 guide](https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df) recommended on the micro-ROS website. All the outputs are in line with the tutorial until the last step where need to build my agent using:
`ros2 run micro_ros_setup build_agent.sh`

#### Expected behavior

#### Actual behavior
```
/home/hans/ros2_foxy/my_uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In member function ‘void uros::agent::graph_manager::GraphManager::update_node_entities_info()’:
/home/hans/ros2_foxy/my_uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:579:86: error: ‘ALIVE’ is not a member of ‘eprosima::fastdds::dds::InstanceStateKind’
  579 |         if (sample_info.instance_state == eprosima::fastdds::dds::InstanceStateKind::ALIVE)
      |                                                                                      ^~~~~
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:89: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [Makefile:141: all] Error 2
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:113: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
---
Failed   <<< micro_ros_agent [2min 23s, exited with code 2]

Summary: 1 package finished [2min 46s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
```

#### Additional information
After searching for similar issues I found [this](https://github.com/micro-ROS/micro_ros_arduino/issues/219), but it seems like it has been fixed. Are there any steps I've been missing?

Jimsing-Wong, 2022-03-09:
And after searching for the enumerators of `eprosima::fastdds::dds::InstanceStateKind`, I found an [older version](https://fast-dds.docs.eprosima.com/en/v2.0.1/fastdds/api_reference/dds_pim/subscriber/instancestatekind.html) which has `ALIVE` but the [latest version](https://fast-dds.docs.eprosima.com/en/latest/fastdds/api_reference/dds_pim/subscriber/instancestatekind.html) only has `ALIVE_INSTANCE_STATE`. I edited the enumerator to the newer version and now it gives me a stderr output:
```
--- stderr: micro_ros_agent                              
/usr/bin/ld: warning: libfastrtps.so.2, needed by /home/hans/ros2_foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.4.1
/usr/bin/ld: warning: libfastcdr.so.1, needed by /home/hans/ros2_foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastcdr.so.1.0.22
---
Finished <<< micro_ros_agent [5.49s]
```
It seems that it has built but, a) is this a bug or my modification is wrong, and b) will this stderr output affect my future work?

pablogs9, 2022-03-09:
Do you have your ROS 2 Foxy packet up to date? This was an old issue but we solved it. Make sure that all your ROS 2 installed packages are up to date.

Jimsing-Wong, 2022-03-09:
I followed the process [here](https://github.com/micro-ROS/micro_ros_setup) today, and this error still comes up. I set my versions to foxy, and it should offer me the newest version available. I'm still not sure which part went wrong.

pablogs9, 2022-03-09:
Hello @Jimsing-Wong, I have replicated the agent installation here: https://asciinema.org/a/JzmqYAOStptH2UF4WENKBq5Yx

It seems that there is something weird with your ROS 2 installation. Can you test it with a clean ROS 2 Foxy environment?

Jimsing-Wong, 2022-03-10:
Hi, I reinstalled ROS2 and microROS following your steps, and the output is the same as yours now. However, when I run the example codes(int32_publisher), my board stops at `set_verbose_level` and never creates a client or establishes connection. The output is: 
```
hans@hans-VJS131:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
[1646879058.425936] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1646879058.427001] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```
And when I run `ros2 topic list`, the output is: 
```
/parameter_events
/rosout
```
It seems that my board isn't running the program normally. Could there be some steps I missed?

Jimsing-Wong, 2022-03-10:
Furthermore, I found out that no matter what code I choose to configure my firmware, the file size that flashed to my board is always the same. 
```
Compressed 352080 bytes to 157840...
Writing at 0x00010000... (10 %)
Writing at 0x00014000... (20 %)
Writing at 0x00018000... (30 %)
Writing at 0x0001c000... (40 %)
Writing at 0x00020000... (50 %)
Writing at 0x00024000... (60 %)
Writing at 0x00028000... (70 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (90 %)
Writing at 0x00034000... (100 %)
Wrote 352080 bytes (157840 compressed) at 0x00010000 in 14.0 seconds (effective 201.9 kbit/s)...
Hash of data verified.
```
As for int32_publisher, I saw some examples online that showed the size of the firmware should be much larger. Is there any steps I missed?

pablogs9, 2022-03-10:
How are you flashing your ESP32? Which transport are you using UDP or Serial? Provide steps for replicating.

Jimsing-Wong, 2022-03-10:
Sorry for not describing the problem thoroughly. After the instruction you provided, which creates and builds the agent withour errors, I then create, configure and build firmware using:
`ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`
which didn't show up any errors, then I configure the firmware using serial transport and the example program: 
`ros2 run micro_ros_setup configure_firmware.sh int32_publisher --transport serial`
which also didn't show up any errors, then I build the firmware: 
ros2 run micro_ros_setup build_firmware.sh
during this process, some stderr output occured but it still finished building. But in the end it built the target with output: 
```
[100%] Generating binary image from built executable
python /home/hans/microros_ws/firmware/toolchain/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB --elf-sha256-offset 0xb0 -o /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/int32_publisher.bin int32_publisher.elf
esptool.py v2.9-dev
/usr/bin/cmake -E echo "Generated /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/int32_publisher.bin"
Generated /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/int32_publisher.bin
/usr/bin/cmake -E md5sum /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/int32_publisher.bin > /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/.bin_timestamp
make[2]: Leaving directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
[100%] Built target gen_project_binary
make -f CMakeFiles/app.dir/build.make CMakeFiles/app.dir/depend
make[2]: Entering directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
cd /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build /home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles/app.dir/DependInfo.cmake --color=
Scanning dependencies of target app
make[2]: Leaving directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
make -f CMakeFiles/app.dir/build.make CMakeFiles/app.dir/build
make[2]: Entering directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
make[2]: Nothing to be done for 'CMakeFiles/app.dir/build'.
make[2]: Leaving directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
[100%] Built target app
make[1]: Leaving directory '/home/hans/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
/usr/bin/cmake -E cmake_progress_start /home/hans
/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build/CMakeFiles 0
```
then I flash my firmware with
`ros2 run micro_ros_setup flash_firmware.sh`
which flashed firmwares including the one I posted above. Then I run it with 
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0`
I get the output which says the board stopped at `set-verbose-level`.

pablogs9, 2022-03-10:
Test a couple of things:
1. Find your device in `/dev/serial/by-id/...` and use this path with the agent.
2. Reset the board after opening the agent.

Jimsing-Wong, 2022-03-10:
2) worked for me, I get the output as expected. Thank you so much for your patience!

closed 2022-03-10 by Jimsing-Wong

RioKJ21, 2022-11-24:
> 2. worked for me, I get the output as expected. Thank you so much for your patience!

I followed your steps, and get an issue. After I run and pressed the EN button, {ros2run}: Segmentation fault.
Can you help me? Thank youu

veeran07, 2024-05-01:
getting this error 
---
Finished <<< statistics_msgs [1.42s]
--- stderr: rmw_implementation         
CMake Error at /home/veeran/microros_ws/firmware/mcu_ws/install/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:60 (message):
  Could not find ROS middleware implementation 'rmw_cyclonedds_cpp'.  Choose
  one of the following: rmw_microxrcedds
Call Stack (most recent call first):
  CMakeLists.txt:22 (get_default_rmw_implementation)

---
Failed   <<< rmw_implementation [0.24s, exited with code 1]
Aborted  <<< rcl_interfaces [1.62s]
Aborted  <<< rosidl_typesupport_microxrcedds_test_msg [0.31s]
Aborted  <<< std_msgs [1.73s]                     
Aborted  <<< test_msgs [1.16s]                    
Aborted  <<< example_interfaces [1.18s]

Summary: 39 packages finished [9.04s]
  1 package failed: rmw_implementation
  5 packages aborted: example_interfaces rcl_interfaces rosidl_typesupport_microxrcedds_test_msg std_msgs test_msgs
  23 packages had stderr output: action_msgs builtin_interfaces example_interfaces lifecycle_msgs micro_ros_msgs micro_ros_utilities microxrcedds_client rcl_interfaces rcl_logging_interface rcl_logging_noop rcutils rmw rmw_implementation rmw_microxrcedds rosgraph_msgs rosidl_runtime_c rosidl_typesupport_c rosidl_typesupport_microxrcedds_c statistics_msgs std_msgs std_srvs test_msgs unique_identifier_msgs
  22 packages not processed
make[3]: *** [libmicroros.mk:33: colcon_compile] Error 1
make[3]: Leaving directory '/home/veeran/microros_ws/firmware/freertos_apps/microros_esp32_extensions'
make[2]: *** [esp-idf/main/CMakeFiles/libmicroros.dir/build.make:89: libmicroros-prefix/src/libmicroros-stamp/libmicroros-build] Error 2
make[2]: Leaving directory '/home/veeran/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
make[1]: *** [CMakeFiles/Makefile2:3862: esp-idf/main/CMakeFiles/libmicroros.dir/all] Error 2
make[1]: Leaving directory '/home/veeran/microros_ws/firmware/freertos_apps/microros_esp32_extensions/build'
make: *** [Makefile:139: all] Error 2
[ros2run]: Process exited with failure 2

flux0-0, 2024-07-23:
i can't use ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888, It seems that the data is not sent through this port  how can I fix it but not use serial ?