micro-ROS/micro-ROS-Agent #126: A way to set a ROS namespace in the Agent

bjsowa, 2022-03-21:
I would like to have the ability to set an arbitrary ROS namespace of the node from the computer running the Agent. The way of doing this in ROSSerial is to just run the `serial_node` under a namespace. In micro-ROS it's a little more problematic because the namespace has to be passed to `rclc_node_init_default` function, so it has to be retrieved before the ROS communication is initialized. 

The way I see it could work is when the Agent could be run with some parameters and the client had functions to retrieve these parameters. Then, I could just define the `namespace` parameter and retrieve it in the firmware before initializing the node.

pablogs9, 2022-03-22:
The name handling of the micro-ROS nodes is completely handled on the client-side. In fact, the micro-ROS agent does not do any modification on any property of the entities created by the client. I'll take a look at that.

It would be important in your use case that the client knows in which namespace it is being exposed or it is ok if the agent just overrides the client name prepending a namespace?

pablogs9, 2022-05-17:
ping @bjsowa

bjsowa, 2022-05-27:
It's not important that the client knows in which namespace it is running. As long as all names are mapped correctly on the agent side it's ok for me.

pablogs9, 2022-05-30:
@bjsowa here we have a PoC, could you test it? https://github.com/micro-ROS/micro-ROS-Agent/pull/145

Just do:
```bash
ros2 run micro_ros_agent micro_ros_agent serial --dev [DEVICE] -v6 --namespace-remapping /mynamespace
```

And let us know if this is what you needed.

pablogs9, 2022-06-14:
Amy update on this?

bjsowa, 2022-08-26:
@pablogs9 Any plan on continuing the PoC? The current implementation doesn't seem to work.

Also, do you think it would be possible to implement Domain ID remapping? This is even more important for me than namespace remapping.

pablogs9, 2022-09-05:
No bandwidth to work on this right now. I'll keep this open for the future. If you make some advances do not hesitate to PR your progress.

woutertollenaar, 2023-05-09:
I would also like to set up the ROS namespace in the agent instead of in the microcontroller. Has a possible solution been found for this yet?
(I want to pass an id from the microros_agent launche file to the microcontroller)

#145 has also not been resolved so far.

elgarbe, 2023-12-20:
> ```shell
> --namespace-remapping /mynamespace
> ```

This is not working for me. 
I run:

`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 --namespace-remapping /chori`

I also tried:

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='micro_ros_agent',
            namespace='chori',
            output='screen',
            executable='micro_ros_agent',
            name='micro_ros_agent',
            arguments=["serial", "--dev", "/dev/ttyACM0"]
        )
    ])

```

and, nothing, the topics comming from my board are located under / namespace.

tombaby97, 2025-03-28:
Could someone provide an update on this issue, Is adding a namespace possible with micro-ROS,Tried passing ns arguments, pushRosNamespace functionality etc but still not working.  If yes then how? If No, Are there any workarounds to achieve this

TannerGilbert, 2025-05-24:
I would also be interested in this. I just tried the [feature/namespace_override branch](https://github.com/micro-ROS/micro-ROS-Agent/tree/feature/namespace_override) with ROS 2 Humble but it only remapped my node `/ap -> /test/ap` but all my topics still had the `/ap` namespace.