micro-ROS/micro-ROS-Agent #130: Snap not connecting, Docker works though.

madgrizzle, 2022-04-13:
## Issue template

- Hardware description: Badger2040 (pico with e-ink display) and ROS2 galactic on host
- RTOS: ?none?
- Installation type: latest/stable from snap
- Version or commit hash: galactic, snapID: Rfmoo8EkRPvm6YoW8SyIzchrKCttYfND

#### Steps to reproduce the issue
1) build and upload uf2 from micro_ros_raspberrypi_pico_sdk to pico.
2) install micro-ros-agent snap following directions
2) execute: `micro-ros-agent serial --dev /dev/ttyACM0 baudrate=115200 -v6`

#### Expected behavior
pico connects to micro-ros-agent

#### Actual behavior
pico hangs on rmw_uros_ping_agent and times-out

#### Additional information
I can get a connection using docker
`sudo docker run -it --rm --net=host -v /dev:/dev -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:galactic serial --dev /dev/ttyACM0 -v6`

but when I run the snap, it sits hanging when the pico reaches the ping and eventually times out.

Is docker the only solution for galactic?  If so, how do I change the ROS_DOMAIN_ID of the docker as it seems to run on 0 by default.  I tried -e ROS_DOMAIN_ID=1 (which is what I use) and it didn't work.

Acuadros95, 2022-04-18:
Hi @madgrizzle,
There are known issues with snap serial port connections, so the dockerized Agent is the way to go.

> Is docker the only solution for galactic? If so, how do I change the ROS_DOMAIN_ID of the docker as it seems to run on 0 by default. I tried -e ROS_DOMAIN_ID=1 (which is what I use) and it didn't work.

This is configured on the client code, you can use multiple clients in different domain ids within the same agent.
Check this example [link](https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/configuration_example/configured_publisher/main.c#L46) or this tutorial [link](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/node/#initialization)

madgrizzle, 2022-04-18:
Thanks.  So the problem is isolated to the snap and building from source doesn't experience the same problem?  

Also thanks for the info on the domain.  This is my first foray into ROS2.

Acuadros95, 2022-04-19:
> So the problem is isolated to the snap and building from source doesn't experience the same problem?

Exactly, have fun!

closed 2022-04-19 by Acuadros95

madgrizzle, 2022-04-19:
Thanks.. Why does the README say there's issues with using docker and not snap?

Acuadros95, 2022-04-19:
> There is an unknown issue when dealing with serial ports **shared** with the micro-ROS agent running inside a Docker. Sometimes it works with a remarkable packet loss.

Note that it says ***shared*** serial ports. Avoid opening the used serial port with other programs such as minicom, as it will end with missing data on the Agent side.

But yes, I am going to add the snap issue.

Acuadros95, 2022-04-19:
Done https://github.com/micro-ROS/micro-ROS-Agent/pull/131, thanks for the catch!