micro-ROS/micro-ROS-Agent #137: Multiple micro-ROS agents causes data loss

Benblob688, 2022-05-17:
## Issue template

- Hardware description: Nvidia Jetson running ubuntu 20.04
- RTOS: Teensy 4.0
- Installation type: docker agent
- Version or commit hash: foxy

#### Steps to reproduce the issue

I use the following commands to start up a micro-ROS agent (never gone deeper than this):
`sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/<device-id> -v6 && export ROS_DOMAIN_ID=20`

One device plugged in, one agent running, one device's data received normally
Two devices plugged in, one agent running, one device's data received normally
Two devices plugged in, two agents running, two device's data received intermittently & issue arises:

#### Expected behavior

Should be able to run multiple agents for multiple devices plugged in.

#### Actual behavior

There are occasional freezes to the incoming data, and these freezes are synchronised between the devices. We then also get an initial value which is far from the expected value (temperature, pressure etc.) see attached screenshots.

#### Additional information

I get the impression that multiple devices should be able to run off one micro-ROS agent, but can't find any guidance or examples on this. Can two agents running interfere with each other? Can the Jetson be CPU-overwhelmed by the overhead of running two agents? Is setting the `DOMAIN_ID` the same on two simultaneous agents a problem? Is there a simple way to modify the agent docker command above to listen to multiple devices?

[image]

Acuadros95, 2022-05-17:
Hi @Benblob688,
You can use the multiserial agent directly, avoiding multiple agent instances with unnecessary overhead:
`sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy multiserial --devs "/dev/1 /dev/2 ..." -v6`

Also, you dont need to set `ROS_DOMAIN_ID` on the Agent, this is configured on the micro-ROS client side.
In fact, a single Agent can handle communications on multiple domain IDs: Check this example [link](https://github.com/micro-ROS/micro-ROS-demos/blob/galactic/rclc/configuration_example/configured_publisher/main.c#L46) or this tutorial [link](https://micro.ros.org/docs/tutorials/programming_rcl_rclc/node/#initialization)

Benblob688, 2022-05-17:
Hi @Acuadros95,
I used the command as you suggest, and the screenshot is below. It does find the devices but I don't get the rapid printout of a single connection, nor do the ros topics `list`. The agent stops at the initialisation stage. Interestingly, it did not care when I unplugged the devices. Nothing changed. I tried with and without the `export ROS_DOMAIN_ID` with no change, as you explained.
[image]

Acuadros95, 2022-05-18:
> Interestingly, it did not care when I unplugged the devices
What do you mean with this? You can start the agent with the serial devices disconnected (The file is not present on `/dev/...`) and the Agent should wait for the connection.

Can you try that? Start the multiserial agent with both ports disconnected, then connect them one by one while checking the log.

Benblob688, 2022-05-18:
I mean that on the`serial` version of the docker command, unplugging the devices would result in a `server stopped` and then repeated `serial port not found` messages every second. 
In this case with `multiserial` version of the docker command, I can unplug the devices and no error codes occur, no print statements come out. For example, in the screenshot attached, I followed your request to start with none plugged in, then one, then both. I then unplugged both. The screenshot is taken with no devices attached, but it has not reverted back to the `serial port not found` behaviour that I expect to see.
[image]

in terms of the micro-ROS app, we haven't implemented it in this way, we followed this style: https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher/micro-ros_publisher.ino
do you think this could be the cause of the `multiserial` command getting stuck on the `Root.cpp` step?

Acuadros95, 2022-05-19:
I could not replicate on our side using [micro-ros_publisher.ino](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher/micro-ros_publisher.ino) code and 2 Teensy 4.1.

- Can you try with this exact example code?
- Are you using [v2.0.4-foxy](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.4-foxy) release?
- Also make sure you have the latest agent image: `docker pull microros/micro-ros-agent:foxy`
- Can you run the image without `sudo`?

Acuadros95, 2022-05-30:
Any updates on this?

Acuadros95, 2022-06-07:
Closing, feel free to reopen if the problem is not solved.

closed 2022-06-07 by Acuadros95

Acuadros95, 2022-06-16:
It looks like `select` implementation differs between architectures: `select()` returned `EINVAL` errno for `tv_usec >= 1 second` values on a RPi, but this does not happen on my AMD laptop.

Regular `serial` agent works because it uses `poll()` instead of `select()`, avoiding this bug.

This PR should fix the problem: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/311
Will update when the fixed ARM dockers are ready.

Acuadros95, 2022-06-16:
Dockers updated!

Run `docker pull microros/micro-ros-agent:foxy` and try again

Hubery-wl, 2023-10-27:
Hello, I'm a novice, and I've been using multiple micro-ros recently. Can you share the following how you realize multiple micro-ros? I'd be very grateful. Thank you.

Hubery-wl, 2023-10-27:
Hello, I'm a novice, and I've been using multiple micro-ros recently. Can you share the following how you realized multiple micro-ros, or provide some information? I'd be very grateful. Thank you.