micro-ROS/micro-ROS-Agent #155: Run micro_ros_agent only shows two lines of info

Genozen, 2022-06-25:
## Issue template

- Hardware description: ESP32 DevKit micro-ROS
- RTOS: freeRTOS where you are using micro-ROS
- Installation type: ROS2 Foxy on VMware workstation with Serial Port enabled

#### Steps to reproduce the issue
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
ros2 topics list (on another terminal)

#### Expected behavior:
Longer list after pressing reset on ESP32?

#### Actual behavior: 
[1656141640.108886] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1656141640.109234] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'topics' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'launch', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'topic', 'wtf')

I saw another thread with a similar issue, but I'm still not quite sure how to solve this.

Genozen, 2022-06-25:
Same result using the docker method:

genozen@ubuntu:~/Desktop/microros_ws$ docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
Unable to find image 'microros/micro-ros-agent:foxy' locally
foxy: Pulling from microros/micro-ros-agent
d5fd17ec1767: Pull complete 
ca43c78a9071: Pull complete 
695af473b1cd: Pull complete 
20ecbdd5d86a: Pull complete 
184a755bf529: Pull complete 
9c4352d7e259: Pull complete 
ea0e9bbed1d0: Pull complete 
65bf8fe27777: Pull complete 
a7e4758ada80: Pull complete 
34a8c9291c75: Pull complete 
7686189d3c32: Pull complete 
220958e97c88: Pull complete 
4f4fb700ef54: Pull complete 
f1dfaa08f5aa: Pull complete 
99d8818fd412: Pull complete 
bc1aae8b424b: Pull complete 
4afcc20c1708: Pull complete 
c5a804c68b15: Pull complete 
7b5ed05a6860: Pull complete 
ec57159ad4fb: Pull complete 
Digest: sha256:6ed0649ced29f775594f2cb9d5af89c2d9f06041634ca78bfae8f019dc39bac8
Status: Downloaded newer image for microros/micro-ros-agent:foxy
S[1656142838.718245] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1656142838.718727] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

pablogs9, 2022-06-27:
The micro-ROS client will try to connect to the agent in rclc_support_init() for a [certain amount of times](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L57) with a [certain timeout](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L58).

If this attempt fails, in this example, the app will fail. You can check the return value of this function and retry until the agent is reached. You can [ping the agent](https://github.com/micro-ROS/micro_ros_arduino/blob/e85b9b47142601a72c742e1af71a0ff399b03c07/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L93) in order to wait until it is available. But in general, you should handle the client-agent connection.

In your use case, if you open the agent and later the client, the client might lose all the opportunities to connect. If you open the agent and then the client, it will succeed.

Genozen, 2022-06-27:
Hmm...so there's something wrong with the client-side (ESP32) connection?
When I follow the tutorial like this one: https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df

Is the client already running after flashing onto the ESP32? (step 6).
If so, I should be able to build and run the agent and establish a connection right?

pablogs9, 2022-06-27:
Yes, if the micro-ROS Agent does not show any logging information (use the flag `-v6`) there is a communication problem between both entities. I recommend checking the network status of the ESP32 and the communication between your ESP32 and you computer. Once everything is working correctly, micro-ROS should be able to communicate.

Genozen, 2022-06-27:
Sry, could you guide me a little bit more on how to check the network status of the ESP32 in microROS? I know we can use the WiFi library in the Arduino framework, not sure about microROS...(very new to this).
Another thing that might be causing the network connection issue is that I'm running these in a Virtual Machine, where I enable serial devices from my USB port. Hence the network might be different in the VM?

pablogs9, 2022-06-27:
The networking status is not related to micro-ROS, just check a networking example on the ESP32. In any case, I recommend you use the official ESP-IDF module for micro-ROS: https://github.com/micro-ROS/micro_ros_espidf_component.

BTW, are you trying to communicate the client and the agent by serial or by the network?

Genozen, 2022-06-28:
I'm trying to communicate via the network (WiFi).
Also, I tested it on another computer with native Linux installed instead of the virtual machine and it works! So for now I'll use that one to save me some trouble. Thank you so much! Excited to dive deeper into microROS.

Also, I have some common issues while installing microROS and have some suggested installations to make the user experience smoother, is there a way to post it somewhere?

pablogs9, 2022-06-28:
You can share your comments just here so other users can see when they have the same issues as you

closed 2022-06-28 by pablogs9