micro-ROS/micro-ROS-Agent #156: How to compile microros agent with source code?

chrisjaxy, 2022-06-30:
## Issue template

- Hardware description: <Ubuntu 20.04>
- RTOS: 
- Installation type: 
- Version or commit hash: 

I want to use `ros2 run micro_ros_setup create_agent_ws.sh` to get the source code of the agent, and add some logs to view some information. But i found `Micro-XRCE-DDS-Agent` is in another repository, and it seems that it is not easy to add log to the source code of `Micro-XRCE-DDS-Agent` to compile. I want to ask, how to easily add log to the source code of `Micro-XRCE-DDS-Agent`, and then compile the agent.

pablogs9, 2022-06-30:
Just create a workspace with the micro-ROS Agent packet and use colcon to build it:

```
mkdir -p ws/src
cd ws/src
git clone -b [your required branch here] https://github.com/micro-ROS/micro-ROS-Agent
cd ..
apt update && rosdep update && rosdep install --from-paths src --ignore-src -y
colcon build
```

What do you mean by "add log to the source code"?

chrisjaxy, 2022-06-30:
> Just create a workspace with the micro-ROS Agent packet and use colcon to build it:
>
> ```
> mkdir -p ws/src
> cd ws/src
> git clone -b [your required branch here] [...]

I want to try to modify the code of `Micro-XRCE-DDS-Agent`.Is it possible to use the above method? And compiling with the above method fails.
#
CMake Error at CMakeLists.txt:51 (find_package):
  By not providing "Findmicro_ros_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "micro_ros_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "micro_ros_msgs"
  with any of the following names:

  micro_ros_msgsConfig.cmake
  micro_ros_msgs-config.cmake

pablogs9, 2022-06-30:
You are not installing `micro_ros_msgs`, make sure that `apt update && rosdep update && rosdep install --from-paths src --ignore-src -y` is running correctly.

If you want to modify Micro XRCE-DDS Agent, just [clone the repo](https://github.com/eProsima/Micro-XRCE-DDS-Agent) and [follow the instructions](https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#installing-the-agent-standalone).

chrisjaxy, 2022-06-30:
> If you want to modify Micro XRCE-DDS Agent, just [clone the repo](https://github.com/eProsima/Micro-XRCE-DDS-Agent) and [follow the instructions](https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#installing-the-agent-standalone).

After following the instructions and running the script below. Compiling the agent will report an error, which was normal before.
```
git clone -b ros2 https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
cd Micro-XRCE-DDS-Agent
mkdir build && cd build
cmake ..
make
sudo make install
```
#
--- stderr: micro_ros_agent                             
/usr/bin/ld: warning: libfastrtps.so.2, needed by /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so, may conflict with libfastrtps.so.2.4.1
/usr/bin/ld: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o: undefined reference to symbol '_ZN8eprosima8fastrtps5types12ReturnCode_t10RETCODE_OKE'
/usr/bin/ld: /opt/ros/foxy/lib/libfastrtps.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

pablogs9, 2022-06-30:
Try to not source your ROS 2 installation when building the Micro XRCE-DDS Agent

chrisjaxy, 2022-06-30:
> Try to not source your ROS 2 installation when building the Micro XRCE-DDS Agent

How to do it? I ran the script above, did I make a mistake when compiling the Micro-XRCE-DDS-Agent, or did I make a mistake when compiling the microros agent?

pablogs9, 2022-06-30:
In order to uninstall just remove the files that has been copied to your filesystem, they appears when you run `sudo make install`

chrisjaxy, 2022-06-30:
I didn't have the source /opt/ros/..../setup.sh before executing the above script. Is there a way to compile the source code of the cloned `Micro XRCE-DDS Agent` when compiling microros_agent?

chrisjaxy, 2022-06-30:
> After following the instructions and running the script below. Compiling the agent will report an error, which was normal before.
>
> ```
> git clone -b ros2 [...]

Will this happen when you execute it here? How did you solve it in the end?

pablogs9, 2022-06-30:
The easiest way to build the micro-ROS Agent with a modified version of the inner Micro XRCE-DDS Agent is to fork the former, make your modifications and set this file before building the micro-ROS Agent: https://github.com/micro-ROS/micro-ROS-Agent/blob/da7fb682e9594294302bce5ef404f5dd9618f7b1/micro_ros_agent/cmake/SuperBuild.cmake#L26-L29

chrisjaxy, 2022-06-30:
> The easiest way to build the micro-ROS Agent with a modified version of the inner Micro XRCE-DDS Agent is to fork the former, make your modifications and set this file before building the micro-ROS [...]

Thank you for your help. The problem that `make install` affects the compilation of microros agent. I hope there will be a better solution in the future besides deleting files.

closed 2022-06-30 by pablogs9