micro-ROS/micro-ROS-Agent #158: How to solve the sticky bag problem?

vd112, 2022-07-11:
If I publish  10 packets, I can subscribe to the 10 packets directly.However, after forwarding by Micro ROS Agent, the number of packets received by MCU is less than 10. So I did the following debugging.
  I sent 10 packets of data and data content is "aaaaaaaaaa+number". Through Wireshark, I found that there were only two packets(RTPS packets) after DDS for the 10 packets I sent.Here are two pictures.The contents of the serializedData packet are correct.
[image]
[image]
  By checking agent log, I find that "[==>> DDS <<==]" is two packets of data.So I guess micro Ros Agent didn't disassemble the second RTPS packet.
[image]
  I would appreciate it if you could solve my question!

pablogs9, 2022-07-11:
Please provide code for replicating this issue.

vd112, 2022-07-12:
Thank you for your reply.
[test.zip](https://github.com/micro-ROS/micro-ROS-Agent/files/9088789/test.zip)

pablogs9, 2022-07-12:
I cannot open this file (seems to be corrupted), could you paste here your code, details of your platform, your colcon.meta files and instructions for replicating?

vd112, 2022-07-12:
My platform is Ubuntu 20.04 system installed ros2 Foxy version.
The following figure shows the directory structure.
[image]
"spubTest.hpp" content is as follows.
`#ifndef TEST_HPP__
#define TEST_HPP__

#include <std_msgs/msg/string.hpp>
#include <rclcpp/rclcpp.hpp>
#include <glog/logging.h>
#include <iostream>
#include <memory>
#include <thread>
#include <string>

class SpubTest
{
public:
    explicit SpubTest(rclcpp::Node::SharedPtr pub_node);
    virtual ~SpubTest();

private:
    int num = 0;
    void subCallback(const std_msgs::msg::String::SharedPtr msg);
    void sub_func_thread();

    std::thread sub_thread_;
    rclcpp::Node::SharedPtr node_;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr test_pub_;
    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr test_sub_;
};
#endif`

"main.cpp" content is as follows.

`#include "spubTest.hpp"
#include <signal.h>
#include <iostream>
#include <rclcpp/rclcpp.hpp>

void sigintHandler(int sig)
{
  std::cout << "stain_detect Ctr-C signal: " << sig << std::endl;
  rclcpp::shutdown();
}

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);

  auto pub_node = std::make_shared<rclcpp::Node>("test_pub_node");
  signal(SIGINT, sigintHandler);

  SpubTest may_testor(pub_node);

  rclcpp::executors::MultiThreadedExecutor exec;
  exec.add_node(pub_node);
  exec.spin();
  exec.remove_node(pub_node);
  rclcpp::shutdown();

  return 0;
}`

"spubTest.cpp" content is as follows.

`#include "spubTest.hpp"

using std::placeholders::_1;

SpubTest::SpubTest(rclcpp::Node::SharedPtr pub_node) : node_(pub_node)
{
    test_pub_ = node_->create_publisher<std_msgs::msg::String>("/test_spub", rclcpp::QoS(100).best_effort());
    auto message = std_msgs::msg::String();
    sleep(5);
    int num = 0;
    
    for (int i = 0; i < 10; i++)
    {
        num++;
	    message.data = std::string("aaaaaaaaaa" + std::to_string(num));
        test_pub_->publish(message);
        std::cout << "pub num : " << num << std::endl;
    }
    sleep(1);
}

SpubTest::~SpubTest()
{
}
`

"CMakeLists.txt" content is as follows.

`cmake_minimum_required(VERSION 3.5)
project(spubTest)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)

# find_package
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)

# uncomment the following section in order to fill in
# further dependencies manually.
include_directories(
  include
)

add_executable(spubTest
  src/main.cpp
  src/spubTest.cpp
)

ament_target_dependencies(spubTest
  rclcpp
  std_msgs
)

install(TARGETS
  spubTest
  DESTINATION lib/${PROJECT_NAME})

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)

  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  # set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  # set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
`
"package.xml" content is as follows.

`<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>spubTest</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="cheyulu@cvte.com">cheyulu</maintainer>
  <license>TODO: License declaration</license>
  <build_depend>rclcpp</build_depend>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <exec_depend>rclcpp</exec_depend>
  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
`

vd112, 2022-07-12:
My platform is Ubuntu 20.04 installed ros2 Foxy version.
The following figure shows the code directory structure.
[image]
[spubTest_hpp.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090536/spubTest_hpp.txt)

vd112, 2022-07-12:
My platform is Ubuntu 20.04 installed ros2 Foxy version.
The following figure shows the code directory structure.
[image]
[spubTest_hpp.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090572/spubTest_hpp.txt)
[main_cpp.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090573/main_cpp.txt)
[spubTest_cpp.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090574/spubTest_cpp.txt)
[CMakeLists.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090579/CMakeLists.txt)
[package_xml.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9090584/package_xml.txt)
The suffix of the above files has been changed to "TXT".You should restore the file name.
"spubTest.hpp", "main.cpp", "spubTest.cpp", "CMakeLists.txt", "package.xml".

Acuadros95, 2022-07-13:
Hi @vd112, 

I don't think this problem is related to micro-ROS, I am missing messages both on micro-ROS side and with `ros2 topic echo /test_spub`:

- Publication:
  ```bash
  root@mydevice:/home/fast_test# ./build/spubTest/spubTest pub num : 1
  pub num : 2
  pub num : 3
  pub num : 4
  pub num : 5
  pub num : 6
  pub num : 7
  pub num : 8
  pub num : 9
  pub num : 10
  ```

- micro-ROS Subscription:
  ```bash
  root@mydevice:/home/microros_ws# ros2 run micro_ros_demos_rclc string_subscriber 
  I have heard: "aaaaaaaaaa3"
  I have heard: "aaaaaaaaaa5"
  I have heard: "aaaaaaaaaa6"
  I have heard: "aaaaaaaaaa7"
  I have heard: "aaaaaaaaaa8"
  I have heard: "aaaaaaaaaa9"
  I have heard: "aaaaaaaaaa10"
  ```

- ROS2 subscription:
  ```bash
  root@mydevice:/home/microros_ws# ros2 topic echo /test_spub 
  data: aaaaaaaaaa6
  ---
  data: aaaaaaaaaa7
  ---
  data: aaaaaaaaaa8
  ---
  data: aaaaaaaaaa9
  ---
  data: aaaaaaaaaa10
  ---
  ```

How are you running this on micro-ROS side and what is your use case? 
Is this solved using reliable entities?

vd112, 2022-07-14:
Hi @Acuadros95 ,
   Thanks for your response.
   The result of your experiment is the same as that of mine.But I think "ros2 topic echo /test_spub" is not a good way to verify.So I created a Subscription project code similar to Publication.Running the Subscription project will see that it can receive all data sent by the Publication.To solve these problems, I used Wireshark to capture and analyze packets.The captured data discovery Publication sent all the data.So the Subscription project code I created accepts all data.So I think "ROS2 topic echo/test_spub" is not a good way to verify.
    Here is my Subscription project file.Please let me know if you can't open this file.
[sub_test.zip](https://github.com/micro-ROS/micro-ROS-Agent/files/9108816/sub_test.zip)
    I'm sorry I don't agree with that”Also, micro-ROS Agent handles incoming DDS data sub messages one by one, it does not matter if they come in a single RTPS message“.I did a lot of experiments and it turns out that if an RTPS packet has multiple data it can only parse one data.I can guess that the DDS in your reply to my example sent only 7 RTPS packets, but the 7 packets contained all the data.“aaaaaaaaaa1”,“aaaaaaaaaa2” and "aaaaaaaaaa3" in the first RTPS packet."aaaaaaaaaa4" and "aaaaaaaaaa5" in the second RTPS packet.You can use wireshark to verify my guess.
     So my question now is if "aaaaaaaaaa1", "aaaaaaaaaa2" and "aaaaaaaaaa3" are in the same RTPS packet, why can micro-ros Agent only receive "aaaaaaaaaa3".
     I'm going to do an experiment today using Reliable.I anticipate your reply sincerely.

Acuadros95, 2022-07-14:
Reliability should solve this, but you are right, this behavior does not make much sense.
I am going to take a good look into this.

vd112, 2022-07-14:
Thank you very much.I look forward to your research results.

vd112, 2022-07-14:
Hi, @Acuadros95,
I'm sorry that I modified the "Subscriber_qos.depth = 10" and the result is the same as before. I tried to solve this problem in a reliable way but the result was the same as before.Therefore, I think the main problem is how to deal with the agent when it receives a RTPS package containing multiple data.

Acuadros95, 2022-07-14:
Could you share your micro-ROS code along with your setup? (Board, build system used, ROS Distro, ...)

Also, please share your Agent debug log (-v6) for this test.

vd112, 2022-07-15:
Hi, @Acuadros95,
I'm sorry that I modified the "Subscriber_qos.depth = 10" and the result is the same as before. I tried to solve this problem in a reliable way but the result was the same as before.Therefore, I think the main problem is how to deal with the agent when it receives a RTPS package containing multiple data.
The MCU I use is STM32F429 connected to RK3399 through serial port.Rk3399 soc runs ubuntu 20.04.The ros version is ROS2 Foxy.Agent configuration is the following files "microros_lib.cmake" and "microros_lib.meta".
[microros_lib_cmake.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9118984/microros_lib_cmake.txt)
[microros_lib_meta.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9118987/microros_lib_meta.txt)

vd112, 2022-07-15:
Hi, @Acuadros95,
I'm sorry that I modified the "Subscriber_qos.depth = 10" and the result is the same as before. I tried to solve this problem in a reliable way but the result was the same as before.Therefore, I think the main problem is how to deal with the agent when it receives a RTPS package containing multiple data.
The MCU I use is STM32F429 connected to RK3399 through serial port.Rk3399 soc runs ubuntu 20.04.The ros version is ROS2 Foxy.Agent configuration is the following files "microros_lib.cmake" and "microros_lib.meta".
[agent_log.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9119084/agent_log.txt)
[microros_lib_cmake.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9119100/microros_lib_cmake.txt)
[microros_lib_meta.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9119101/microros_lib_meta.txt)

Acuadros95, 2022-07-18:
There are a lot of things going on on that log. Please try to replicate the error with a minimum replicable micro-ROS example. 

This way we can replicate, debug and find a fix for the problem on our side.

vd112, 2022-07-18:
> replicate
Hi, @Acuadros95,
I want to know if changing QOS subscriber configuration Agent can forward all publisher data in this experiment？I'm shrinking the log.

vd112, 2022-07-18:
Hi, @Acuadros95,
I want to know if changing QOS subscriber configuration Agent can forward all publisher data in this experiment？I'm shrinking the log.
https://github.com/micro-ROS/micro-ROS-Agent/issues/158#issuecomment-1183079272

Acuadros95, 2022-07-18:
> QOS subscriber configuration Agent

What do you mean exactly? The Agent subscribers uses the same QoS as the micro-ROS subscriber, that is why you should get the data after the depth or reliability modifications.

There may be other factors here, but the Agent should get all the data.

vd112, 2022-07-18:
Sorry, my engineering environment is complicated.It is not easy to extract the minimum micro-ros code.
I think you can solve this problem on the platform you've already established(This platform: https://github.com/micro-ROS/micro-ROS-Agent/issues/158#issuecomment-1183079272 ).Maybe my QOS configuration is ok.The main problem may be Agent parsing RTPS packets.
Thank you!

Acuadros95, 2022-07-18:
Could you share your exact micro-ROS subscriber initialization code?

We already checked the Agent with a simple test and it looks OK after modifying the depth as explained here: https://github.com/micro-ROS/micro-ROS-Agent/issues/158#issuecomment-1184165298

vd112, 2022-07-18:
I'm cleaning up the initialization process. Wait reply.

vd112, 2022-07-18:
I initialized subscriber using the function “rclc_subscription_init_best_effort”.The details are in the following file.Please start at the bottom.
[subscription_init_c.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/9130390/subscription_init_c.txt)

Acuadros95, 2022-07-19:
I dont understand your attached file, are you re implementing the rclc methods? 
Why is the implementation of `rcl_subscription_init` there?

vd112, 2022-07-19:
Yes,it is.The MCU I use is STM32F429 connected to RK3399 through serial port.MCU code is written in C language.MCU and Linux platform RK3399 use Micro ROS Agent to publish and subscribe messages.

closed 2022-07-25 by vd112