micro-ROS/micro-ROS-Agent #161: Micro Ros Agent build without internet.

coalman321, 2022-08-10:


pablogs9, 2022-08-10:
1. Add Micro XRCE-DDS Agent to your workspace, branch `ros2`: https://github.com/eProsima/Micro-XRCE-DDS-Agent
2. Build Micro XRCE-DDS Agent with this flags: https://github.com/micro-ROS/micro-ROS-Agent/blob/da7fb682e9594294302bce5ef404f5dd9618f7b1/micro_ros_agent/cmake/SuperBuild.cmake#L35-L52
3. Build micro-ROS Agent with [`MICROROSAGENT_SUPERBUILD`](https://github.com/micro-ROS/micro-ROS-Agent/blob/da7fb682e9594294302bce5ef404f5dd9618f7b1/micro_ros_agent/CMakeLists.txt#L25) set to off.
4. Make sure that Micro XRCE-DDS Agent is built first (manually with a `colcon build --packages-up-to ...`) because this package is not in the `package.xml` of micro-ROS Agent so colcon won't resolve the dependency.

If you have any problem I will create a replicable command set for solving this.

coalman321, 2022-08-10:
I have implemented your suggestions, but I did notice something of concern. It looks like the XRCE agent has a similar cloning behavior built in. Can I forcibly include those packages into my build as well? 
is doing something similar as well.

Output from colcon during a build
```
Starting >>> microxrcedds_agent
[Processing: microxrcedds_agent]                               
--- stderr: microxrcedds_agent                                       
Cloning into 'microxrcedds_client'...
HEAD is now at bdaddba Release v2.1.0
Cloning into 'microcdr'...
HEAD is now at 1c712a0 Release v2.0.0
```

pablogs9, 2022-08-11:
Those packages are extra functionality for the agent, I guess that you can get rid of them setting [`UAGENT_P2P_PROFILE`](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/851bbbf5faa45f1d757fe90d1a5f2749d28d8cfa/CMakeLists.txt#L36) and [`UAGENT_CED_PROFILE`](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/851bbbf5faa45f1d757fe90d1a5f2749d28d8cfa/CMakeLists.txt#L34) to `OFF`. 

I have been able to build micro-ROS Agent without downloading any extra package with these commands:

```bash
mkdir -p ws/src && cd ws/src
git clone https://github.com/micro-ROS/micro_ros_msgs
git clone https://github.com/micro-ROS/micro-ROS-Agent
git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd ..

# Enter into a network-less environment
docker run -it --rm --net=none -v $(pwd):/ws ros:humble

cd ws
colcon build --packages-select microxrcedds_agent --event-handler console_direct+ --cmake-args -DUAGENT_P2P_PROFILE=OFF -DUAGENT_CED_PROFILE=OFF -DUAGENT_USE_SYSTEM_FASTDDS:BOOL=ON -DUAGENT_USE_SYSTEM_FASTCDR:BOOL=ON -DUAGENT_BUILD_EXECUTABLE=OFF -DUAGENT_ISOLATED_INSTALL:BOOL=OFF

source install/local_setup.bash

colcon build --packages-up-to micro_ros_agent --event-handler console_direct+ --cmake-args -DMICROROSAGENT_SUPERBUILD=OFF

source install/local_setup.bash

ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6

```

closed 2024-02-27 by coalman321