micro-ROS/micro-ROS-Agent #173: Agent crashes with Not enough memory in the buffer stream

chrisjaxy, 2022-08-31:
**Describe the bug**
The agent crashes when I send a small amount of incorrect data to the agent port.

**To Reproduce**
Steps to reproduce the behaviour:
1. ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
**Expected behaviour**
A clear and concise description of what you expected to happen.

**System information (please complete the following information):**
 - OS: [e.g. WIndows 10, Ubuntu 16.04, ...]
 - ROS 2 [e.g. Dashing, Foxy, ...]
 - Version [e.g. commit hash, tag, ...]

**Additional context**
Add any other context about the problem here.

chrisjaxy, 2022-09-05:
I tested on humble and found the same problem.

pablogs9, 2022-09-05:
Working on it

pablogs9, 2022-09-05:
Working on it here: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/314

chrisjaxy, 2022-09-05:
I did a simple test and the problem didn't happen again.

closed 2022-09-05 by chrisjaxy