micro-ROS/micro-ROS-Agent #174: The agent node did not recognize the disconnection to the MCU.

Interactics, 2022-09-06:
## Issue template

- Hardware description: STM32F429ZI Nucleo Board
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash:  Humble

#### Steps to reproduce the issue
Can't reproduce the issue. sometimes it occrued.

When micro-ROS subscribing to ```geometry_msg/msg/point```,  it sometimes lost connection.

#### Expected behavior
When the MCU board is disconnected, the agent node must also be disconnected.

#### Actual behavior

However, Agent node is mistaken Nucleo Board for being connecting.
So, it does not attempt the reconnect process.

When I tested it with STM IDE debugger, 
Agent Node is still connecting but Nucleo board's connection condition is in WAITING_AGENT.

 
#### Additional information

MicroROS
- Number of Node - 1
- Number of Publishers - 10
- Number of Subscribers - 4
- Number of Executors - 1

What I want to know is, how do agent knows that connection is lost?

pablogs9, 2022-09-06:
The agent won't remove the created entities by default. In fact, the agent won't assume that the client connection is lost because maybe the client is just in a deep sleep mode for example. 

If you need to force this behavior, you need to build the client enabling [this flag](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/6f4edd25c0984fd30bf3bf84e2d1d33af03f24a1/CMakeLists.txt#L73-L74).

Interactics, 2022-09-07:
Thank you for your quick reply. I'll apply it and let you know the result.

Interactics, 2022-09-07:
> The agent won't remove the created entities by default. In fact, the agent won't assume that the client connection is lost because maybe the client is just in a deep sleep mode for example.
>
>
>
> If [...]

Thank you for your quick reply. I'll apply it and let you know the result.

Interactics, 2022-09-16:
Sorry for the late reply.

It works for me. 
So I leave the STM32 resolution procedure  for STM32 users who have the same problem as me.

In my case, I builded a Nucleo Board using [micro_ros_stm32cubemx_utils](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils)

To change ```UCLIENT_HARD_LIVELINESS_CHECK``` in Micro-XRCE-DDS-Client CMakeLists.txt,
I added ``` "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON" ``` in `micro_ros_stm32cubemx_utils/microros_static_library/library_generation/colcon.meta`

```
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_HARD_LIVELINESS_CHECK=ON" // I added!
            ]
        },
```

This is my result.

```
[1663298134.744854] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x2C45D82E, subscriber_id: 0x004(4), participant_id: 0x000(1)
[1663298134.751199] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x2C45D82E, datareader_id: 0x004(6), subscriber_id: 0x004(4)
[1663298147.161007] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x2C45D82E
[1663298147.161063] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x2C45D82E, address: 0
[1663298147.161070] info     | Processor.cpp      | check_heartbeats         | Session destroyed due to liveliness timeout | client_key: 0x2C45D82E, address: 0  
# Above message is created when connection is lost 

```

And the reconnection works very well.

Thank you

closed 2022-09-16 by Interactics