micro-ROS/micro-ROS-Agent #176: Foxy debian package

roni-kreinin, 2022-09-28:
I saw in #96 that debian packages are available on Vulcanexus, but they are only released for Galactic and Humble. Can you release a Foxy debian for `micro_ros_agent`? Either to Vulcanexus or the ROS2 package server. Thanks.

pablogs9, 2022-09-29:
Hello @roni-kreinin, AFAIK Vulcanexus will not have a Foxy version (right @eduPonz?). Have you considered moving to Humble?

tonybaltovski, 2023-03-22:
@pablogs9 can the micro-ros-agent be added the Open Robotics package server for Humble please?  Having the Vulcanexus package server will allow the debian package to be installed but `rosdep` will not find the dependency.

tonynajjar, 2024-01-17:
> @pablogs9 can the micro-ros-agent be added the Open Robotics package server for Humble please? Having the Vulcanexus package server will allow the debian package to be installed but `rosdep` will not find the dependency.

Any update on this?

pulak-gautam, 2024-03-05:
Hi, I am facing the same issue. We are not able to release debian packages for our repository due to its dependency on ``micro_ros_agent``. @pablogs9 Could you please clarify?
Thank you for your time.