micro-ROS/micro-ROS-Agent #178: Cannot build agent on arm64 (RPi 3B+)

UVk, 2022-12-12:
## Issue template

- Hardware description: RPi 3B+
- RTOS: Ubuntu Mate 22.04
- Installation type: ROS2 - Humble, micro_ros_setup
- Version or commit hash: latest

#### Steps to reproduce the issue
Following the instructions on Repo

#### Expected behavior
micro_ros_msgs - Build successfully 
micro_ros_setup - Build successfully
micro_ros_agent - Build successfully
#### Actual behavior
micro_ros_msgs - Build successfully 
micro_ros_setup - Build successfully
micro_ros_agent - Freeze at micro_ros_agent:build 9%
#### Additional information
I did a fresh install of the OS, ROS2, and micro-ros few times with the same results
The same procedure on PC with Ubuntu works perfectly.

UVk, 2022-12-12:
Here is the log:
HEAD is now at 021828f Fix requester and replier reuse behaviour (#318)
[ 12%] [34m[1mNo patch step for 'xrceagent'[0m
[ 18%] [34m[1mPerforming configure step for 'xrceagent'[0m
loading initial cache file /home/ros/uros/build/micro_ros_agent/agent/tmp/xrceagent-cache-.cmake
-- Setting build type to 'Release' as none was specified.
[33mCMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
[0m
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ros/uros/build/micro_ros_agent/agent/src/xrceagent-build
[ 25%] [34m[1mPerforming build step for 'xrceagent'[0m
[  2%] [32mBuilding CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/Agent.cpp.o[0m
[  4%] [32mBuilding CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/AgentInstance.cpp.o[0m
[  7%] [32mBuilding CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o[0m
[  9%] [32mBuilding CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/processor/Processor.cpp.o[0m

pablogs9, 2022-12-12:
There is no error on the log you provide, can you provide:
- The full log
- Instructions for replicating

UVk, 2022-12-12:
This is logall.log
[4.072s] DEBUG:colcon:Command line arguments: ['/usr/bin/colcon', 'build']
[4.073s] DEBUG:colcon:Parsed command line arguments: Namespace(log_base=None, log_level=None, verb_name='build', build_base='build', install_base='install', merge_install=False, symlink_install=False, test_result_base=None, continue_on_error=False, executor='parallel', parallel_workers=4, event_handlers=None, ignore_user_meta=False, metas=['./colcon.meta'], base_paths=['.'], packages_ignore=None, packages_ignore_regex=None, paths=None, packages_up_to=None, packages_up_to_regex=None, packages_above=None, packages_above_and_dependencies=None, packages_above_depth=None, packages_select_by_dep=None, packages_skip_by_dep=None, packages_skip_up_to=None, packages_select_build_failed=False, packages_skip_build_finished=False, packages_select_test_failures=False, packages_skip_test_passed=False, packages_select=None, packages_skip=None, packages_select_regex=None, packages_skip_regex=None, packages_start=None, packages_end=None, allow_overriding=[], cmake_args=None, cmake_target=None, cmake_target_skip_unavailable=False, cmake_clean_cache=False, cmake_clean_first=False, cmake_force_configure=False, ament_cmake_args=None, catkin_cmake_args=None, catkin_skip_building_tests=False, verb_parser=<colcon_defaults.argument_parser.defaults.DefaultArgumentsDecorator object at 0xffff80cb60e0>, verb_extension=<colcon_core.verb.build.BuildVerb object at 0xffff8105bd60>, main=<bound method BuildVerb.main of <colcon_core.verb.build.BuildVerb object at 0xffff8105bd60>>)
[4.422s] Level 1:colcon.colcon_core.package_discovery:discover_packages(colcon_meta) check parameters
[4.423s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) check parameters
[4.423s] Level 1:colcon.colcon_core.package_discovery:discover_packages(ignore) check parameters
[4.423s] Level 1:colcon.colcon_core.package_discovery:discover_packages(path) check parameters
[4.424s] Level 1:colcon.colcon_core.package_discovery:discover_packages(colcon_meta) discover
[4.424s] Level 1:colcon.colcon_core.package_discovery:discover_packages(recursive) discover
[4.424s] INFO:colcon.colcon_core.package_discovery:Crawling recursively for packages in '/home/ros/uros'
[4.425s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['ignore', 'ignore_ament_install']
[4.426s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'ignore'
[4.426s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'ignore_ament_install'
[4.427s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['colcon_pkg']
[4.427s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'colcon_pkg'
[4.428s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['colcon_meta']
[4.428s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'colcon_meta'
[4.428s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['ros']
[4.429s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'ros'
[4.648s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['cmake', 'python']
[4.648s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'cmake'
[4.649s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'python'
[4.649s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extensions ['python_setup_py']
[4.649s] Level 1:colcon.colcon_core.package_identification:_identify(.) by extension 'python_setup_py'
[4.650s] Level 1:colcon.colcon_core.package_identification:_identify(build) by extensions ['ignore', 'ignore_ament_install']
[4.651s] Level 1:colcon.colcon_core.package_identification:_identify(build) by extension 'ignore'
[4.652s] Level 1:colcon.colcon_core.package_identification:_identify(build) ignored
[4.653s] Level 1:colcon.colcon_core.package_identification:_identify(install) by extensions ['ignore', 'ignore_ament_install']
[4.654s] Level 1:colcon.colcon_core.package_identification:_identify(install) by extension 'ignore'
[4.654s] Level 1:colcon.colcon_core.package_identification:_identify(install) ignored
[4.655s] Level 1:colcon.colcon_core.package_identification:_identify(log) by extensions ['ignore', 'ignore_ament_install']
[4.656s] Level 1:colcon.colcon_core.package_identification:_identify(log) by extension 'ignore'
[4.656s] Level 1:colcon.colcon_core.package_identification:_identify(log) ignored
[4.666s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extensions ['ignore', 'ignore_ament_install']
[4.667s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extension 'ignore'
[4.668s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extension 'ignore_ament_install'
[4.669s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extensions ['colcon_pkg']
[4.669s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extension 'colcon_pkg'
[4.670s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extensions ['colcon_meta']
[4.670s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extension 'colcon_meta'
[4.671s] Level 1:colcon.colcon_core.package_identification:_identify(src) by extensions ['ros']
[... cut ...]
[13.009s] DEBUG:colcon.colcon_core.event_handler.log_command:Invoking command in '/home/ros/uros/build/micro_ros_agent': AMENT_PREFIX_PATH=/home/ros/uros/install/micro_ros_msgs:${AMENT_PREFIX_PATH} CMAKE_PREFIX_PATH=/home/ros/uros/install/micro_ros_msgs:/home/ros/ros_ws/install/micro_ros_msgs:/opt/ros/humble LD_LIBRARY_PATH=/home/ros/uros/install/micro_ros_msgs/lib:${LD_LIBRARY_PATH} PYTHONPATH=/home/ros/uros/install/micro_ros_msgs/local/lib/python3.10/dist-packages:${PYTHONPATH} /usr/bin/cmake --build /home/ros/uros/build/micro_ros_agent -- -j4 -l4

pablogs9, 2022-12-13:
Well, you're almost out of RAM memory and your swap is full. Your device is freezing due to a lack of memory for building.

[image]

UVk, 2022-12-17:
Thanks to point it out. As far as you know, can I build it on x86 processor (PC) and copy it to the RPi 3?

pablogs9, 2022-12-18:
You should cross compile on the x86 machine for an ARM target

closed 2022-12-18 by pablogs9

AhmadSameh, 2023-01-27:
is there a way to cross compile other than using the ros_cross_compile pip package?