micro-ROS/micro-ROS-Agent #179: Client is deleted with mRo Control Zero H7

gorghino, 2022-12-28:
Hi,
I flashed the latest PX4 repo in my mRo Control Zero H7. 
I'm actually synced with this PR (https://github.com/PX4/PX4-Autopilot/pull/20812) that fixes a problem with nuttx, otherwise the latest working PX4 version is the commit before 21 Nov (https://github.com/PX4/PX4-Autopilot/issues/20793).

In my SD card I put ```microdds_client start -t serial -d /dev/ttyS1 -b 115200``` in extras.txt.

On my Nvidia NX (Jetpack 5.0.2 - Ubuntu 20.04) I installed ROS2 humble by sources and I cloned micro-ROS-Agent in my workspace. No errors with compiling everything.

When I run ```./build/micro_ros_agent/micro_ros_agent serial -D /dev/ttyTHS1 -b 115200``` I can see it starting and subscribing the topics from PX4 but after 2/3sec the client is deleted and the session is closed. Then the session is either established or re-established.

```nvidia@nvidia-desktop:~/drone-racing$ ./build/micro_ros_agent/micro_ros_agent serial -D /dev/ttyTHS1 -b 115200
[1672233093.421832] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1672233093.422874] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1672233104.307837] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1672233104.308323] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1672233104.395390] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1672233104.414770] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1672233104.415100] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1672233104.416100] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1672233104.439207] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1672233104.439447] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1672233104.440339] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1672233104.463996] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1672233104.464295] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[1672233104.465774] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EA(6), subscriber_id: 0x3EA(4)
[1672233104.487957] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EB(2), participant_id: 0x001(1)
[1672233104.488238] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EB(4), participant_id: 0x001(1)
[1672233104.489680] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EB(6), subscriber_id: 0x3EB(4)
[1672233104.517728] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0A1(2), participant_id: 0x001(1)
[1672233104.518132] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0A1(3), participant_id: 0x001(1)
[1672233104.519725] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0A1(5), publisher_id: 0x0A1(3)
[1672233104.544851] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0A8(2), participant_id: 0x001(1)
[1672233104.545183] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0A8(3), participant_id: 0x001(1)
[1672233104.546852] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0A8(5), publisher_id: 0x0A8(3)
[1672233104.572646] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x009(2), participant_id: 0x001(1)
[1672233104.573022] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x009(3), participant_id: 0x001(1)
[1672233104.574964] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x009(5), publisher_id: 0x009(3)
[1672233104.866546] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0AC(2), participant_id: 0x001(1)
[1672233104.867773] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0AC(3), participant_id: 0x001(1)
[1672233104.873510] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0AC(5), publisher_id: 0x0AC(3)
[1672233106.198530] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0DE(2), participant_id: 0x001(1)
[1672233106.199584] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0DE(3), participant_id: 0x001(1)
[1672233106.204974] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0DE(5), publisher_id: 0x0DE(3)
[1672233106.438109] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x00000001
[1672233106.438547] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x00000001, address: 1
[1672233106.439388] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1672233106.439504] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1672233106.491187] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1672233106.510310] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1672233106.510564] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1672233106.511468] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1672233106.536030] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1672233106.537162] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1672233106.539827] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1672233106.561140] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1672233106.561386] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[... cut ...]
```./build/micro_ros_agent/micro_ros_agent serial -D /dev/ttyTHS1 -b 115200```

**Expected behaviour**
micro-ros-agent shall start, subcribe the topics and never lose/delete the session

**System information (please complete the following information):**
 - OS: Ubuntu 20.04 - NVIDIA NX (Carrier board) - Jetpack 5.0.2
 - ROS 2 Humble (Tested also with Foxy)
 - Version: 0e41ea652238bd0123f6249d69b533d282882dd6

LOG with v6 flag: 
[micro-ros-agent-v6-mro.log](https://github.com/micro-ROS/micro-ROS-Agent/files/10314511/micro-ros-agent-v6-mro.log)

gorghino, 2022-12-29:
I chcked the logging module on the mro and apparently micro-ros-agent is complaining about a missing ping:

[image]

pablogs9, 2022-12-30:
I guess that this shall be moved to PX4 client.

closed 2022-12-30 by pablogs9