micro-ROS/micro-ROS-Agent #182: Specifying domain-ID through agent reference

iluetkeb, 2023-03-03:
We would like to configure the domain ID in the agent, so that we can use the same firmware image for all devices. This is, in principle, possible using a reference-based connection configuration.

However, the agent always takes the domain ID from the client, even if the connection is created by reference and the reference specification contains an ID (which is just ignored).

This could be solved by either always taking the ID from the reference instead, or by having a special client ID which tells the agent to use the reference ID.

What are your thoughts on this?

btw, we also noticed some discrepancy in the documentation related to this and the code.  The function signature in the documentation does not contain a domain_id parameter (see https://micro-xrce-dds.docs.eprosima.com/en/latest/client_api.html#create-entities-by-reference), but the code actually does (see https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/include/uxr/client/core/session/create_entities_ref.h#L50)

pablogs9, 2023-03-03:
Hello @iluetkeb,

I will fix the documentation ASAP: https://github.com/eProsima/Micro-XRCE-DDS-docs/pull/84

Regarding the feature, as you mention, is it not supported. I like the idea of having a special number (maybe the max DOMAIN_ID) that triggers the agent to determine the Domain ID from other sources. We are open to PR if you implement this feature.

mmigliavacca, 2023-03-10:
Hello @pablogs9, we also would need to configure (or override) the DOMAIN_ID from the agent, so that we don't have to update the firmware on each robot.

By the way, in the past on the clint (firmware) side we were calling `rclc_node_init_with_options()` setting the domain_id into `rcl_node_options_t` (see https://github.com/micro-ROS/rcl/blob/e6307be0a017a8e9787b0c291ef881765212298b/rcl/include/rcl/node_options.h#L55), but this field disappeared starting from galactic (see https://github.com/micro-ROS/rcl/blob/ba7d67d770d8539665fff7a16994ccbbccd0e81d/rcl/include/rcl/node_options.h#L37).

How can we change the DOMAIN_ID in galactic / humble?

Thanks!

pablogs9, 2023-03-10:
AFAIK, this is the humble way of doing so: https://github.com/micro-ROS/micro-ROS-demos/blob/45f342d4ef887f6a2e9f2dfa45d1d3388d7f6dbd/rclc/configuration_example/configured_publisher/main.c#L49

mmigliavacca, 2023-03-10:
Thank you, we missed that change.

iluetkeb, 2023-03-23:
Just as a heads-up: We have a patch where, when initializing the client by reference, the domain id is passed from the agent. It is very minimal and works for us, but we need to some initial clearing before releasing it. Should be soon now.

delihus, 2023-05-19:
Hi @iluetkeb!
Is  the micro-ROS-agent with Micro-XRCE-DDS-agent with this change eProsima/Micro-XRCE-DDS-Agent#331should change `ROS_DOMAIN_ID` in Micro-ROS nodes? Here i created workspace to build `develop` version https://github.com/delihus/micro_ros_setup.

DominikN, 2023-07-12:
Hi, there is a [Micro-XRCE-DDS-Agent v2.4.1 release](https://github.com/eProsima/Micro-XRCE-DDS-Agent/releases/tag/v2.4.1) already. I have tested it and it works great. We have created even a Docker image:

Can you tell me when the new release of the Micro-ROS-Agent will be published based on `v2.4.1`?

pablogs9, 2023-07-12:
CC @Acuadros95

pablogs9, 2023-08-10:
Hello @DominikN It should be ready to be used in the micro-ROS Agent, the latest version is newer than https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/331.

BTW can we close this @iluetkeb ?

iluetkeb, 2023-09-05:
Done :-)

closed 2023-09-05 by iluetkeb