micro-ROS/micro-ROS-Agent #186: Micro-ros-agent does not seem to compile with fmt v9

PetitPrince6, 2023-03-31:
On Ubuntu 22.04 LTS clean install
Repo clone in a ros2 workspace. Building through colcon build.

The compilation fails, lots of errors related to fmt and inability to format. It seems that something is not backward compatible with fmt. If I force the fmt version 8, it compiles. Unfortunately, the v8 is no available in apt packages, requiring a specific install just for micro ros agent.

pablogs9, 2023-04-03:
Not able to reproduce on a clean ROS 2 Humble Ubuntu 22.04 docker: https://asciinema.org/a/ScEwg9hddtPFHyc3Pb9FtAL4S

Could you provide more details?

PetitPrince6, 2023-04-03:
Ok, sorry, I found out, I did it on two different machines but I had the installation of a custom library messing with fmt, it was intalling the fmt v9 version with which the micro-ros-agent doesn't compile. So not micro-ros-agent issue. Sorry for that.

closed 2023-04-03 by PetitPrince6

Acuadros95, 2023-04-03:
You can force the Agent to use the installed version of the `spdlog` and `fmt`, check this comment: https://github.com/micro-ROS/micro_ros_setup/issues/455#issuecomment-1042635375

PetitPrince6, 2023-04-03:
The option -DUAGENT_USE_SYSTEM_LOGGER=ON didn't work for me, but maybe I was not using it properly, I got lost at a certain point ^^. Anyway, I found two alternative solutions: a) switch to the snap version. b) Compile the agent before installing my custom library. :)