micro-ROS/micro-ROS-Agent #188: Request official ROS release

qw8z5fUB, 2023-04-13:
I have a client package which is dependent on `micro_ros_agent`.
I'd like to add to my `package.xml` the dependency `<depend>micro_ros_agent</depend>`, then have `micro_ros_agent` installed automatically via `rosdep install --from-paths my_package`. However this does not work.

(I'm migrating from a `rosserial` client, where declaring a dependency on `rosserial_arduino` was enough to have all dependencies installed and get the client running, without any additional steps).

Looking at the [ROS package index](https://index.ros.org/p/micro_ros_agent/github-micro-ROS-micro-ROS-Agent/#humble) I see that `micro_ros_agent` is not officially released (it shows release:UNRELEASED). I assume this is why the packages are not available through `rosdep`.

I see from related issues (#96, #163) that you have other distribution channels via snap and third-party apt repos, is there a particular reason the official ROS channels are not supported?

My understanding is that to release through official channels you need to create a bloom release repository - is that something you could look at? This would make it much easier for end-users to adopt this package.

gavanderhoorn, 2023-04-13:
A bit of a *me too*, but I just wanted to add that we've encountered the same situation.

As the agent is a ROS package, we can `exec_depend` on it, but we can't tell users to just `rosdep` it, as without the additional repository setup and the addon to the `rosdep` DB, that won't work.

pablogs9, 2023-04-13:
Hi @qw8z5fUB , 
We're sorry if this solution is not the one you were looking for. However, we do take note of your feedback and we'll take it into account for future decision-making.

qw8z5fUB, 2023-04-13:
Thanks for the comment. Yes I was mostly curious if there was a particular reason for not releasing the package, or if an official release just hasn't been considered yet. If the later I'd definitely suggest considering it as it would greatly facilitate use/adoption of micro ROS, for the reasons mentioned above.

gavanderhoorn, 2023-04-13:
@pablogs9: I'm not sure what the requirements / criteria are these days, but OSRF has infrastructure to import specific `.deb`s into the main package repositories. `catkin_lint` and a couple other packages are imported into `repositories.ros.org` that way -- same thing for some ROS 2 dependencies.

That would not need any more work/maintenance on your side, except a couple of PRs every now and then to import a new version.

Not sure whether packages that depend on other ROS 2 packages are handled though. That might make OSRF hesitant to do this with `micro_ros_agent`.

tonybaltovski, 2023-04-20:
+1 for this. We would like to have `micro_ros_agent` available from the Open Robotics buildfarm as well.

closed 2023-04-20 by pablogs9

reinzor, 2023-07-10:
+1 here

rgariepy, 2023-08-30:
@pablogs9 (cc @JaimeMartin) Any update on the plans here? Can we help at all? We have the same desire as @qw8z5fUB , though possibly with a bit more scale :)