micro-ROS/micro-ROS-Agent #202: Run micro-ros-agent under rmw_fastrtps_dynamic_cpp, using introspection

henriksod, 2023-08-01:
## Issue template

- Hardware description: Linux aarch64
- Installation type: https://github.com/mvukov/rules_ros2
- Version or commit hash: humble

I am getting segmentation fault at 
https://github.com/micro-ROS/micro-ROS-Agent/blob/8fdc5060c7601a5d7d9e12254a1681e51da02d2c/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp#L33

due to the `type_support_` object being erased from memory somewhere in 
https://github.com/micro-ROS/micro-ROS-Agent/blob/8fdc5060c7601a5d7d9e12254a1681e51da02d2c/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp#L31

`type_support_->typesupport_identifier` is `rosidl_typesupport_introspection_cpp` since I am using `rmw_fastrtps_dynamic_cpp` as middleware. I see that L33 in graph_typesupport.cpp compares with `rosidl_typesupport_fastrtps_cpp`, which is why I assume micro-ros-agent requires rmw_fastrtps_cpp as middleware.

My question is, can micro-ros-agent support introspection, i.e. rmw_fastrtps_dynamic_cpp?
What are the necessary changes if not?

pablogs9, 2023-08-02:
Hello, not sure why you need introspection in the micro-ROS Agent.

In any case, if there is a solution that fits your requirements, please feel free to contribute with a PR. Maybe we can make those type support to be selected by means of an environment variable.

henriksod, 2023-08-02:
I am using this repo https://github.com/mvukov/rules_ros2 to build ROS2 nodes using the Bazel build system. They currently have support for CycloneDDS which uses `rosidl_typesupport_introspection`. I have made a PR to add FastDDS and since there is already generator rules implemented for `rosidl_typesupport_introspection`, it makes sense to use `rmw_fastrtps_dynamic`.

Maybe I am too new to ROS2 to understand, why would I not want to use introspection? It reduces complexity in the projects and makes switching between DDS middlewares more seamless.