micro-ROS/micro-ROS-Agent #203: Run micro-ros-agent but no node and msg

Junming-Liang, 2023-08-10:
## Issue template

- Hardware description: STM32F407ZG
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
Following the official example instructions, I'm trying to run the pingpong example. When I run the command `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0`, the terminal output seems to be normal (full output in the Additional Information section below), the serial port is blinking with data outputs, yet I cannot see any nodes or messages in ROS.

#### Expected behavior
Expected to see nodes and messages in ROS when running the pingpong example.

#### Actual behavior
No nodes or messages appear in ROS even though the terminal output seems to be normal and the serial port is blinking with data outputs.

#### Additional information
Terminal output:
```
$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 
[1691655948.255286] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1691655948.255633] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1691655957.357919] info     | Root.cpp           | create_client            | create                 | client_key: 0x37218F85, session_id: 0x81
[1691655957.358069] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x37218F85, address: 0
[1691655957.381921] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x37218F85, participant_id: 0x000(1)
[1691655957.396572] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x37218F85, topic_id: 0x000(2), participant_id: 0x000(1)
[1691655957.405984] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x37218F85, publisher_id: 0x000(3), participant_id: 0x000(1)
[1691655957.416498] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x37218F85, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1691655957.431244] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x37218F85, topic_id: 0x001(2), participant_id: 0x000(1)
[1691655957.440660] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x37218F85, publisher_id: 0x001(3), participant_id: 0x000(1)
[1691655957.450924] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x37218F85, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1691655957.465664] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x37218F85, topic_id: 0x002(2), participant_id: 0x000(1)
[1691655957.474996] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x37218F85, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1691655957.485772] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x37218F85, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1691655957.502091] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x37218F85, topic_id: 0x003(2), participant_id: 0x000(1)
[1691655957.511964] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x37218F85, subscriber_id: 0x001(4), participant_id: 0x000(1)
[1691655957.523983] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x37218F85, datareader_id: 0x001(6), subscriber_id: 0x001(4)

```

pablogs9, 2023-08-10:
Paste here the micro-ROS Agent log running with the flag `-v6`

Have you tried to use ROS 2 Iron? Since ROS 2 Foxy is deprecated.

Are you using any specific `ROS_DOMAIN`?

Which is the output of `ros2 topic list`?

Junming-Liang, 2023-08-10:
> Paste here the micro-ROS Agent log running with the flag `-v6`
> 
> Have you tried to use ROS 2 Iron? Since ROS 2 Foxy is deprecated.
> 
> Are you using any specific `ROS_DOMAIN`?
> 
> Which is the output of `ros2 topic list`?

Thank you very much for your prompt reply. Here is the output using -v6 as a flag:
```
$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6
[1691657384.718806] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1691657384.719092] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1691657394.224720] info     | Root.cpp           | create_client            | create                 | client_key: 0x37218F85, session_id: 0x81
[1691657394.224866] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x37218F85, address: 0
[1691657394.225117] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1691657394.235004] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 44, data: 
0000: 81 80 00 00 01 07 24 00 00 0A 00 01 01 03 00 00 16 00 00 00 00 01 02 08 0E 00 00 00 70 69 6E 67
0020: 70 6F 6E 67 5F 6E 6F 64 65 00 00 00
[1691657394.248009] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x37218F85, participant_id: 0x000(1)
[1691657394.248077] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1691657394.248089] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1691657394.254114] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1691657394.262931] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 80, data: 
0000: 81 80 01 00 01 07 47 00 00 0B 00 02 02 03 00 00 39 00 00 00 11 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 52 4F 53 2F 70 69 6E 67 00 00 01 20 1D 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64
0040: 64 73 5F 3A 3A 48 65 61 64 65 72 5F 00 00 01 00
[1691657394.263050] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x37218F85, topic_id: 0x000(2), participant_id: 0x000(1)
[1691657394.263113] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1691657394.263127] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1691657394.268760] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1691657394.272451] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1691657394.272542] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x37218F85, publisher_id: 0x000(3), participant_id: 0x000(1)
[1691657394.272586] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1691657394.272607] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1691657394.277725] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1691657394.282789] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 36, data: 
0000: 81 80 03 00 01 07 1C 00 00 0D 00 05 05 03 00 00 0E 00 00 00 00 02 01 20 03 00 01 20 0A 00 00 00
0020: 00 00 00 03
[1691657394.283471] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x37218F85, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1691657394.283607] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1691657394.283664] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1691657394.289436] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x37218F85, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
```
I am using Ubuntu 20.04 and my existing project is built on the Foxy version, so it is not feasible for me to switch to the Iron version in a short time. I indeed set a special ROS_DOMAIN_ID, here is the content  [... cut ...] y ~/.zshrc:
```
export ROS_DOMAIN_ID=117
```
Here are my rostopic and rosnode:
```
$ ros2 topic list               
/clicked_point
/control/vehicle_control_command
/ego_pose
/goal_pose
/initialpose
/parameter_events
/planning/target_path_marker
/poses_record
/rosout
/rviz/car_marker
/rviz/car_marker_array
/tf
/tf_static
$ ros2 node list
/rviz
/transform_listener_impl_55f853cf96e0
```
I am very much looking forward to your reply. This problem has troubled me for a long time. Thank you once again.

Junming-Liang, 2023-08-10:
Something I forgot, I changed the app-colcon.meta from:
```
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=2",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
...
```
to:
```
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
```

pablogs9, 2023-08-10:
You see to set the domain ID in the micro-ROS code, just like this: https://github.com/micro-ROS/micro-ROS-demos/blob/ea56acfc45f1cc971a15d4a50e3e59b3dbe1f749/rclc/configuration_example/configured_publisher/main.c#L46-L50

pablogs9, 2023-08-10:
Also it is possible to use the Domain ID 255 on the micro-ROS client side and then use the environment variable `UXR_CLIENT_DOMAIN_ID_TO_OVERRIDE_WITH_ENV` in the agent side, just check this: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/333/files

Junming-Liang, 2023-08-10:
I am extremely grateful to you. With your guidance, I have successfully received the topic in ROS. The specific problem is consistent with your analysis - the Agent class did not successfully obtain the domain_id from the environment variable and defaulted to 0. After changing the domain_id to 0 in my ~/.zshrc：
```
export ROS_DOMAIN_ID=0
```
I was able to receive the corresponding topic normally. 
```
ros2 node list 
/pingpong_node
```
While it seems there might be a more permanent solution, I can't think of a better one at the moment. I'll use 0 as my domain_id for now. Thank you once again.

closed 2023-08-10 by Junming-Liang

russkel, 2023-10-23:
Is there a way we can override this on the Agent side? We have agents running in containers that may have any number of ROS_DOMAIN_IDs and we have no good way of setting this on the client side.

russkel, 2023-10-23:
@KyleJewiss