micro-ROS/micro-ROS-Agent #208: Keep getting build error for fmt

kyuhyong, 2023-10-18:
## Issue template

- Hardware description: voxl
- RTOS: 
- Installation type: microros setup
- Version or commit hash: foxy

#### Steps to reproduce the issue

Foxy is installed
I followed below method to install micro_ros_agent
```
mkdir -p ~/ros2_ws/src
cd ros2_ws/src
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git
git clone  https://github.com/PX4/px4_ros_com.git
git clone https://github.com/PX4/px4_msgs.git
cd ~/ros2_ws
colcon build
source install/local_setup.sh
sudo rosdep init
rosdep update
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
```
Initially had issue with libfmt so I apt installed below
`$ sudo apt install libfmt-dev libspdlog-dev`

#### Expected behavior

#### Actual behavior

I am keep getting below error
```
~/ros2_ws$ ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [1.39s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `vtable for fmt::v6::format_error'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `char fmt::v6::internal::decimal_point_impl<char>(fmt::v6::internal::locale_ref)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_32'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `char fmt::v6::internal::thousands_sep_impl<char>(fmt::v6::internal::locale_ref)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `std::locale fmt::v6::internal::locale_ref::get<std::locale>() const'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `typeinfo for fmt::v6::format_error'
/home/nearth/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v6::basic_string_view<char>, spdlog::level::level_enum, fmt::v6::basic_string_view<char>)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::format_error::~format_error()'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::basic_data<void>::hex_digits'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::basic_data<void>::digits'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/home/xxx/ros2_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.4.1: undefined reference to `fmt::v6::internal::basic_data<void>::zero_or_powers_of_10_64'
collect2: error: ld returned 1 exit status
make[2]: *** [micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< micro_ros_agent [0.80s, exited with code 2]

Summary: 1 package finished [2.33s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
```

How to solve this issue?

#### Additional information

pablogs9, 2023-10-19:
Does this branch fixes the issue https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/345 ?

kyuhyong, 2023-10-19:
> Does this branch fixes the issue [eProsima/Micro-XRCE-DDS-Agent#345](https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/345) ?

This issue resolved actually by setting

closed 2023-10-20 by pablogs9

kyuhyong, 2023-11-06:
I revisited this package and still get error regard to fmt
either UAGENT_USE_SYSTEM_LOGGER is ON or OFF 

I checked the version of spdlog as below.
```
$ sudo apt show libspdlog-dev
Package: libspdlog-dev
Version: 1:0.16.3-1
Priority: optional
Section: universe/libdevel
Source: spdlog
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Med Packaging Team <debian-med-packaging@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 501 kB
Homepage: https://github.com/gabime/spdlog
Download-Size: 143 kB
APT-Manual-Installed: yes
APT-Sources: http://ports.ubuntu.com/ubuntu-ports bionic/universe arm64 Packages
Description: Very fast, header only, C++ logging library
 Features: Very fast - performance is the primary goal; Headers only; No
 dependencies - just copy and use; call style using the excellent cppformat
 library; ostream call style is supported too; Optional extremely fast
 asynchronous mode - using lockfree queues and other tricks to reach millions
 of calls/sec; Custom formatting; Multi/Single threaded loggers; Easily
 extendable with custom log targets (just implement a single function in the
 sink interface). Severity based filtering - threshold levels can be modified
 in runtime as well as in compile time.
 .
 Various log targets:
 Rotating log files, Daily log files, Console logging, Linux syslog.
```

How can I solve the issue?

kyuhyong, 2023-11-06:
This is issue resolved by source install spdlog from the git repository.
Latest spdlog is v1.15 and after install, it builds without error.